autorope/donkeycar #1020: Car cannot run in a straight line

mseld, 2022-06-15:
The car cannot run in a straight line and I'm looking for something like `STEERING_MID` to adjust it

```py
DRIVE_TRAIN_TYPE = "PWM_STEERING_THROTTLE"

PWM_STEERING_THROTTLE = {
    "PWM_STEERING_PIN": "PCA9685.1:40.1",   # PWM output pin for steering servo
    "PWM_STEERING_SCALE": 1.0,              # used to compensate for PWM frequency differents from 60hz; NOT for adjusting steering range
    "PWM_STEERING_INVERTED": False,         # True if hardware requires an inverted PWM pulse
    "PWM_THROTTLE_PIN": "PCA9685.1:40.0",   # PWM output pin for ESC
    "PWM_THROTTLE_SCALE": 1.0,              # used to compensate for PWM frequence differences from 60hz; NOT for increasing/limiting speed
    "PWM_THROTTLE_INVERTED": False,         # True if hardware requires an inverted PWM pulse
    "STEERING_LEFT_PWM": 460,               #pwm value for full left steering
    # STEERING_MID_PWM 400
    "STEERING_RIGHT_PWM": 290,              #pwm value for full right steering
    "THROTTLE_FORWARD_PWM": 420,            #pwm value for max forward throttle
    "THROTTLE_STOPPED_PWM": 370,            #pwm value for no movement
    "THROTTLE_REVERSE_PWM": 320,            #pwm value for max reverse throttle
}
```

sctse999, 2022-06-15:
https://docs.donkeycar.com/guide/calibrate/

sctse999, 2022-06-15:
Follow the doc. This is probably not a bug therefore I am closing the issue now. Reopen if I misunderstand your question.

closed 2022-06-15 by sctse999

Ezward, 2022-06-15:
The midpoint is assumed to be the average of the two steering values.  So that means you need to be calibrated such that the left and right wheels are at the same angles (relative to straight ahead) at their maximum steering.  Calibration is a little tricky; make sure you are picking the value that just barely makes the servo whine, then back off by 1 until it stops whining.  Do this for both left and right.  Generally you will find that if you add the left and right PWM values and divide by two you should get something very close to 370.  If you do not you may not be evenly calibrated between left and right.  So in our case (460 + 290) / 2 = 375, so it is close, but I would apply the 'just until it whines' methodology I suggested.