autorope/donkeycar #1027: Donkey UI

duttasantanuGH, 2022-06-29:
We are trying to connect Ubuntu 18.04 to Jetson Nano donkey car. Have followed the instructions available in https://docs.donkeycar.com/utility/ui/. 
Not being able to configure Username and Hostname and hence getting the attached error message.
Rest all the screens of donkey UI are working, only car sync related functionality are not working.
Any help will be highly appreciated
[image]
.

Ezward, 2022-06-29:
Make sure that you can ssh into the Jetson Nano without the UI first.  I have had the experience that the ssh daemon does not start until you login with a physical keyboard and screen attached to the Jetson.  So try this sequence so we can figure out what is happening
1. Power on a Jetson Nano without a keyboard or screen attached.
2. From another machine try to ssh into that Jetson.  Could you ssh?  If you can then my theory is incorrect and I'm not sure what is going on.  If you cannot, then proceed to 3.
3. Restart the Jetson Nano, but with a keybaord and monitor attached.  Log into the Jetson using the keyboard and monitor.  Now again try to ssh into the Jetson from another machine.  Could you ssh?
- If you could ssh after logging in with the keyboard, then my theory is correct.  To make the ssh daemon start when the Jetson boots, run these commands
```
sudo systemctl enable ssh
```
-  If you could not ssh at all, the the ssh daemon is not running.  To get it running and make it run on boot, run these commands;
```
sudo ufw allow ssh
sudo systemctl start ssh
sudo systemctl enable ssh
```
4. Now you can try 1. and 2. again; you should be able to ssh into the Jetson without having to login with a keyboard.
5. Once you can ssh into the Jetson, try the donkey ui again, it should work.

Ezward, 2022-06-29:
Please let us know how this turns out by adding a comment to this issue.

duttasantanuGH, 2022-07-02:
Thanks for your response. But ssh is not the problem. ssh works fine - as suggested "The login-free setup is required for the screen to work." was also done. All these were done before logging this issue. 
One more point to add - "donkey findcar" also can not find the car.
So the issue is unresolved. Will appreciate support regarding resolution.

duttasantanuGH, 2022-07-04:
Thanks for your kind support. It works.

closed 2022-07-04 by duttasantanuGH

Ezward, 2022-07-04:
I've update the documentation so that it says to set PI_USERNAME and PI_HOSTNAME.  https://docs.donkeycar.com/utility/ui/#the-car-connection