autorope/donkeycar #1030: What's the latest Raspberry Pi OS this supports?

dlarue, 2022-07-13:
The documentation is saying to install Raspbian Legacy( Buster ) but that's been long superseded by Raspberry PiOS and Buster was what was used over two years ago... Is that still what donkeycar is based on and does it even support rPi4?

https://docs.donkeycar.com/guide/robot_sbc/setup_raspberry_pi/#step-1-flash-operating-system

sctse999, 2022-07-13:
@dlarue , buster supports rPi4. I think we want to move to the latest Pi OS but i think it is not on the highest priority.

dlarue, 2022-07-13:
I guess as long as rPi.org is still supporting Buster on rPi4 then probably ok to not update. I had a new user tell me he was getting a 404 on step 10 of the install and when I saw Buster I immediately figured the repo was archived. But if you're saying it's an actively supported OS then makes sense to stick with it. Thx.

closed 2022-07-13 by dlarue