autorope/donkeycar #1032: On Mac donkey car error with "donkey/lib/libbcm_host.so' (no such file)" when I run python manage.py drive

unnatis07, 2022-07-26:
Logs 
python manage.py drive
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v4.3.6.3 ...
WARNING:donkeycar.parts.pins:RPi.GPIO was not imported.
WARNING:donkeycar.parts.pins:pigpio was not imported.
INFO:numexpr.utils:NumExpr defaulting to 8 threads.
loading config file: /Users/vense03/mysim/config.py
loading personal config over-rides from myconfig.py
INFO:__main__:PID: 36298
cfg.CAMERA_TYPE PICAM
INFO:__main__:cfg.CAMERA_TYPE PICAM
Traceback (most recent call last):
  File "manage.py", line 853, in <module>
    meta=args['--meta'])
  File "manage.py", line 142, in drive
    cam = PiCamera(image_w=cfg.IMAGE_W, image_h=cfg.IMAGE_H, image_d=cfg.IMAGE_DEPTH, framerate=cfg.CAMERA_FRAMERATE, vflip=cfg.CAMERA_VFLIP, hflip=cfg.CAMERA_HFLIP)
  File "/Users/vense03/Unnati/donkeycar/donkeycar/parts/camera.py", line 22, in __init__
    from picamera.array import PiRGBArray
  File "/Users/vense03/opt/miniconda3/envs/donkey/lib/python3.7/site-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/Users/vense03/opt/miniconda3/envs/donkey/lib/python3.7/site-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/Users/vense03/opt/miniconda3/envs/donkey/lib/python3.7/site-packages/picamera/mmal.py", line 47, in <module>
    from .bcm_host import VCOS_UNSIGNED
  File "/Users/vense03/opt/miniconda3/envs/donkey/lib/python3.7/site-packages/picamera/bcm_host.py", line 47, in <module>
    _lib = ct.CDLL('libbcm_host.so')
  File "/Users/vense03/opt/miniconda3/envs/donkey/lib/python3.7/ctypes/__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: dlopen(libbcm_host.so, 0x0006): tried: '/Users/vense03/opt/miniconda3/envs/donkey/lib/libbcm_host.so' (no such file), '/Users/vense03/opt/miniconda3/envs/donkey/lib/libbcm_host.so' (no such file), '/Users/vense03/opt/miniconda3/envs/donkey/lib/python3.7/lib-dynload/../../libbcm_host.so' (no such file), '/Users/vense03/opt/miniconda3/envs/donkey/lib/libbcm_host.so' (no such file), '/Users/vense03/opt/miniconda3/envs/donkey/bin/../lib/libbcm_host.so' (no such file), 'libbcm_host.so' (no such file), '/usr/local/lib/libbcm_host.so' (no such file), '/usr/lib/libbcm_host.so' (no such file), '/Users/vense03/mysim/libbcm_host.so' (no such file)

cfox570, 2022-07-26:
Are you running on a Raspberry Pi, Windows, Linux, Mac?   Miniconda install is recommended for Windows, Linux, Mac.  And you have specified the picamera to run.

unnatis07, 2022-07-27:
I am running it on Mac. 

I am following steps listed here : https://docs.donkeycar.com/guide/simulator/

Earlier, since I noticed that it was looking for picamera, I did this step (although its not listed in the steps)
export READTHEDOCS=True
pip install picamera

post this, I executed 
conda activate donkey

and then 
python manage.py drive  : and This step is where I get the error posted above.

cfox570, 2022-07-27:
Did you also follow these instructions: https://docs.donkeycar.com/guide/host_pc/setup_mac/ ?
Did you set DONKEY_GYM = True in myconfig.py?  This replaces the Picamera setting. You do not need picamera installed for the Mac.  If you are still having a problem, I suggest you attach the myconfig.py file with your settings.

I also suggest you join the Discord server for help instead of posting here.  You will likely receive alot more responses.
https://www.donkeycar.com/community.html

Ezward, 2022-10-29:
@unnatis07 It seems you have the RaspberryPi camera selected in your myconfig.py file; that will not work on a Mac.  You want to choose either 'MOCK' for no camera or 'CVCAM' to use opencv to access a webcam or laptop camera (you must have installed the optional opencv dependencies to do this).

Ezward, 2022-10-29:
@unnatis07 Is this still a problem for you?  It sounds like a configuration issue.  I would like to close this issue; please give us an update.

unnatis07, 2022-10-29:
Thanks Ed. Its not an issue anymore. I solved it but dont recall specifics. Thanks again.

closed 2023-03-01 by Ezward