autorope/donkeycar #1040: Evolve the GPS path follow template

Ezward, 2022-09-10:
We have implemented a template that used GPS to record a path and follow it.  This is the issue [#991 Add gps part and integrate into path_follow template](https://github.com/autorope/donkeycar/issues/991) describing the work and what is left to do.  This is the branch that implements it [921-gps-logger](https://github.com/autorope/donkeycar/tree/921-gps-logger).  As noted in [issue #991](https://github.com/autorope/donkeycar/issues/991), there are three steps for getting this merged into the main branch.  The first of these steps is included below.  As these are merged the list will be updated with testing for the subsequent PRs.

1. Test refactor of complete.py template (The DNN template)
    - Success criteria; The refactored complete.py should work the same as  the current main branch (no new functionality was added, this is just a refactor). 
    - Any instructions in docs.donkeycar.com should work.  If there are discrepancies then we should open an issue to either change the code or update the docs at the time we merge this branch into main.
    - Report any bugs (add issues in github)
    - Work with developer to fix bugs and close issues.
    - do this for the other 2 stages are described in [issue #991](https://github.com/autorope/donkeycar/issues/991) as PRs become available.

2.  Experiment with various GPS receiver technologies
    - compare RTK, RTK with dead reckoning and non-RTK, non-RTK with dead reckoning gps receivers.  See this [article](https://ezward.github.io/gps/) for how to setup RTK for this purpose 
    - how many laps (how much time) can you go with non-RTK gps before drift inteferes?
    - Which technologies perform better in a time trial?
    - what is the highest performing setup?
    - what is the most cost effective setup?
  
  
3.  Improve path follow 1
    - The current algorithm uses cross-track error and a PID to turn towards line to follow; no heading is involved.
    - How do optimal PID values vary with velocity?
    - What recommendations can we makes for starting PID values and for a process to identify optimal values?
    - Observer the behavior of the vehicle if it moves too far way from line to follow; the vehicle will get stuck in a tight turn an never achieve the line to follow.  Why does this happen?  What changes could be made to recover from this state?
    - What happens if the path to follow crosses itself?  What changes can be made to reliably drive on such paths?
    
  
4. Improve path follow 2
    - The current algorithm uses cross-track error and a PID to turn towards line to follow; no heading is involved.
    - How could we add in heading? 
      - software only (delta of current and previous GPS location)
      - hardware 6 axis acc/gyro like MPU6500
      - hardware 9 axis acc/gyro/mag like MPU9250 (see https://quadmeup.com/mpu6000-vs-mpu6050-vs-mpu6500/ for comparison)
      - wheel encoders and a kinematic model.
    - How do these complement GPS?  How would data from these sources be use in conjunction with GPS positions?
    - Implement one or more of these sources of heading and add this to the vehicles current pose estimate. (x, y) -> (x, y, heading).

5. Improve path follow 3
    - Once we have heading, we can use more sophisticated path follow algorithms, like pure pursuit.
    - implement pure pursuit using one or more of the techologies identified in #2 using a constant velocity.
    - how does pure pursuit compare to cross-track error; what are benefits and how are failure modes different?

6. Pure Pursuit with waypoint velocity
   - Rather than each waypoint being a simple (x,y) coordinate, add a target velocity to achieve at that waypoint (x, y, velocity).
   - If wheel encoders are available, then closed loop velocity control can be added to achieve these target velocities.
   - In the absence of wheel encoders, a software only approach where velocity is calculated from the current and previous pose estimate.
   - develop a control algorithm to accelerate/decelerate to achieve the target velocity at each waypoint.
   - how precise does the velocity estimate and control need to be to deliver benefits?
   - compare this to cross-track/pid and constant velocity pure pursuit;  
       - which can achieve lower (faster) lap times? 

7. Pre-calculate an optimal trajectory 
    - given a map that indicates the driveable area (perhaps as paths for the left and right borders of the track), calculate an optimal path for the vehicle to follow.
    - try minimizing distance traveled in one lap
    - try minimizing curvature of path in one lap
    - try maximizing single-point velocity that can be achieved.

TCIII, 2022-09-10:
@Ezward,

An excellent path forward to be able to integrate both heading and IMU functions and possibly wheel encoders with GPS be it just GPS or GPS + RTK.
I presently have a Traxxas E-Maxx Truck chassis running a Rpi 3B+/Robo HAT MM1 M4 configuration. The Robo HAT has a built in MPU9250 at hex address 0x69. Unfortunately it will be difficult to add wheel encoders to my E-Maxx.
However I do have a large 6WD chassis with wheel encoders and a small 6WD chassis with motor wheel encoders. The 6WD chassis are not fast, but have used both wheel encoders and GPS to validate the `path_follow.py` template from the DC `921-gps-logger` branch.

Here is a [link](https://youtu.be/_PI20CWPxes) to a video of my Traxxas E-Max running a Sparkfun NEO-M9N GPS with an active antenna on a ground plane. At the time the video was shot the E-Maxx batteries were running low and a cloud cover was coming in as the E-Maxx had performed better during previous runs.

Regards,
TCIII

Ezward, 2023-02-06:
IMU pose estimation and IMU/GPS fusion resources:

- https://pypi.org/project/droneposelib/. dronepose lib is a python library that will make pose estimates based on IMU data.
- https://pypi.org/project/imusensor/ library to make reading the IMU easy.  Look at the acknowledgements at the bottom of the page.
- https://github.com/micropython-IMU/micropython-fusion micro-python library implements Madgwick
- https://github.com/xioTechnologies/Fusion AHRS fusion in C with Python bindings.  Includes some error rejection.
- https://github.com/smahajan07/sensor-fusion Python fuse GPS with IMU for position and orientation estimation
- https://ahrs.readthedocs.io/en/latest/index.html  Collection of Python AHRS (Attitude and Heading Reference Systems) methods for combining IMU and GPS etc.