autorope/donkeycar #1041: Setup a car with DC_STEER_THROTTLE using DC Adafruit Motor Hat (PCA9685)

AlcorDust, 2022-09-14:
Hi, i bought a cheap RC car which uses 2 DC motors, one for steering and one for throttle.
I installed the donkeycar software on Raspberry 3, and i also iunstalled the motor hat requirements. I managed to make the motor hat work with a simple python script.

Now i'm trying to configure mycar editing the myconfig.py file. I selected DRIVE_TRAIN_TYPE = "DC_STEER_THROTTLE", but when i run the drive command i can't make the car move, but i can see the live camera output.

I took a look at the manage.py file, and seems like the DRIVE_TRAIN_TYPE = "DC_STEER_THROTTLE" only uses the L298N controller:

`
elif cfg.DRIVE_TRAIN_TYPE == "DC_STEER_THROTTLE":
        dt = cfg.DC_STEER_THROTTLE
        steering = actuator.L298N_HBridge_2pin(
            pins.pwm_pin_by_id(dt['LEFT_DUTY_PIN']),
            pins.pwm_pin_by_id(dt['RIGHT_DUTY_PIN']))
        throttle = actuator.L298N_HBridge_2pin(
            pins.pwm_pin_by_id(dt['FWD_DUTY_PIN']),
            pins.pwm_pin_by_id(dt['BWD_DUTY_PIN']))
`

How can i link the drive train type with the Adafruit Motor Hat? 

Thanks in advance

Ezward, 2022-09-15:
The Adafruit motor driver uses a TB6612 hbridge rather than an L298 hbridge.  We do have a driver for this; https://github.com/autorope/donkeycar/blob/c61de767b2af4bcd673534f9d28054f098a52f16/donkeycar/parts/actuator.py#L501, but it's not exposed in the configuration and not used in the complete.py template.  You could add it to your manage.py yourself.

AlcorDust, 2022-09-19:
> 

Thank you! 
I edited the file manage.py and then i added and selected the option in myconfig.py
Do you know where i can find the indexes for `STEERING_MOTOR` and `THROTTLE_MOTOR`?

Ezward, 2022-10-26:
@AlcorDust did that work?  Can we close this issue?

closed 2023-02-18 by Ezward