autorope/donkeycar #1044: Modify the path_follow algorithm to head back to the path if it gets too far away

Ezward, 2022-10-09:
The current path_follow algorithm searches for the nearest waypoint to it's current position, then chooses the next waypoint in the path and uses those two points to create a line; it then determines which side of the line it is on and how far from the line it is and uses that as the input to a PID that decides which direction to turn in and how much to turn to get back to the line.

If the car moves far from the path for some reason (it is going very fast and overshoots a turn or another vehicle bumps it off it's path) then the algorithm can get confused.  If it is far enough way from the nearest way point that the PID algorithm indicates a maximum turn, but that turn does not bring the car close enough to a waypoint, then the car will stay in an infinite turn.

You can demonstrate this by recording a path and then placing the car far from the path.  When you put the car in autopilot mode it will start an infinite turn and spin in place.

Modify the algorithm to get back to the path if it detects that is to too far away from the path.

Mux199, 2022-10-21:
Hello,
I want to contribute in this project.
Can I take this issue?

Ezward, 2022-10-24:
@Mux199 Yes of course.  Fork the repo, then create a branch from main.  When you are confident that you have code that works then open a PR to merge back into the donkeycar main branch.  We will see that and can then add comments to your PR.  Also make sure to join the discord; an invite can be found at https://www.donkeycar.com/community.html

TCIII, 2022-10-28:
@Ezward,

I am TCIII and I have a Traxxas E-Maxx Truck Chassis test vehicle running the DC `main branch` and using the `path_follow.py` template that employs input from a Sparkfun NEO-M9N GPS module with an actve 32db gain antenna.

I will be available to test and validate any code branches that add velocity, heading, and IMU input to the GPS CTE input mix.

Regards,
TCIII