autorope/donkeycar #1047: Update path_follow to support velocity target on a waypoint

Ezward, 2022-10-11:
The current path_follow algorithm searches for the nearest waypoint to it's current position, then chooses the next waypoint in the path and uses those two points to create a line; it then determines which side of the line it is on and how far from the line it is and uses that as the input to a PID that decides which direction to turn in and how much to turn.  Throttle is held constant at some value > 0 and <= 1.0 specified in configuration.

Because throttle is constant, in practice we set it to limit the speed to what we can handle driving around the tightest corner.  That means we are much slower on a long straight then we could be, or conversely we are too fast heading into a corner and so may overshoot.

Modify the waypoint data structure to include a target throttle to be achieved at that waypoint.  Modify the vehicle loop in the path_follow.py template to set the throttle based on the waypoint that we are headed to (so the next waypoint from the closest one, as described above).  Allow this to be overridden by a constant throttle if it is specified in the configuration; if no constant throttle is specified then use the throttle values in the waypoints (path) file.

zlite, 2022-10-11:
Nice! I've been hoping variable throttle would come and here it is! It's the only way to be competitive

TCIII, 2022-10-12:
@Ezward,

My 4WD Traxxas E-Maxx can easily do 12-15 m/sec on the straightaways, but would tip over on the curves at that velocity.
Presently I run the E-Maxx at around 3.2 m/sec to avoid tipping over on the curves.
Reducing the vehicle velocity on the curves to a sustainable level and increasing the velocity to a predefined maximum on the straightaways would make for some very interesting racing competition.

TCIII

TCIII, 2022-10-12:
@zlite,

Doesn't ArduRover already have a variable throttle for straightaways and curves (turns)?

TCIII

evanphilipsmith, 2022-10-28:
@Ezward I'm with a team of students at UCSD using donkeycar for a class project. We're interested in GPS path following and would like to try and implement a solution to address this issue. Just wanted to put it out there so you know we are working on it! 😄

TCIII, 2022-10-28:
@evanphilipsmith,

I am TCIII, a DC Maintainer, and I have a Traxxas E-Maxx Truck Chassis test vehicle running the DC `main branch` and using the `path_follow.py` template that employs input from a Sparkfun NEO-M9N GPS module with an actve 32db gain antenna.

I will be available to test and validate any code branches that add velocity, heading, and IMU input to the GPS CTE input mix.

Regards,
TCIII

TCIII, 2022-11-12:
@evanphilipsmith,

[Improve](https://github.com/autorope/donkeycar/issues/1060) DC imu.py part. 

TCIII

PR #1068 (2023-02-03): Support velocity target on a waypoint in `path_follow`
Added support for variable throttle in `path_follow` by modifying the waypoint data structure to record the throttle at each waypoint. The feature may be turned off by setting `USE_CONSTANT_THROTTLE = True` in your `myconfig.py` (the flag is set to `False` by default).

Closes #1047.
closed 2023-02-03 by DocGarbanzo