autorope/donkeycar #1051: Support picamera2 python library for libcamera on RasperryPi OS Bullseye

Ezward, 2022-10-16:
The legacy camera library no longer work on bullseye unless it is configured using raspi-config.  That is a short term solution, but eventually most software will support the newer libcamera/picamera2 apis.
- implement a picamera2 based camera part
- if possible, make this choice transparent to the donkeycar user, so that choose `PICAM` as the camera type will work in picamera2 is available or fall back to picamera if it is not.

Ezward, 2022-10-16:
On Bullseye there is an error attempting to install the optional OpenCV dependencies:
```
$ sudo apt-get install libilmbase-dev libopenexr-dev libgstreamer1.0-dev libjasper-dev libwebp-dev libatlas-base-dev libavcodec-dev libavformat-dev libswscale-dev libqtgui4 libqt4-test
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libqtgui4 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'libqtgui4' has no installation candidate
E: Unable to locate package libqt4-test
```

Ezward, 2022-10-16:
There were a couple of warnings doing the `install -e .[pi]`  
```Installing collected packages: pyusb, sysv-ipc, rpi-ws281x, pyftdi, Adafruit-PureIO, Adafruit-PlatformDetect, typing-extensions, adafruit-circuitpython-requests, Adafruit-Blinka, adafruit-circuitpython-typing, adafruit-circuitpython-busdevice, pillow, wcwidth, tifffile, PyWavelets, networkx, imageio, utm, tornado, simple-pid, Shapely, scikit-image, pynmea2, pyfiglet, progress, PrettyTable, paho-mqtt, opencv-python, h5py, docopt, Adafruit-GPIO, adafruit-circuitpython-register, adafruit-circuitpython-framebuf, imgaug, donkeycar, Adafruit-PCA9685, adafruit-circuitpython-ssd1306, adafruit-circuitpython-rplidar, adafruit-circuitpython-lis3dh
  Attempting uninstall: typing-extensions
    Found existing installation: typing-extensions 3.7.4.3
    Not uninstalling typing-extensions at /usr/lib/python3/dist-packages, outside environment /home/ed/env
    Can't uninstall 'typing-extensions'. No files were found to uninstall.
  Attempting uninstall: pillow
    Found existing installation: Pillow 8.1.2
    Not uninstalling pillow at /usr/lib/python3/dist-packages, outside environment /home/ed/env
    Can't uninstall 'Pillow'. No files were found to uninstall.
  Running setup.py develop for donkeycar
Successfully installed Adafruit-Blinka-8.5.0 Adafruit-GPIO-1.0.3 Adafruit-PCA9685-1.0.1 Adafruit-PlatformDetect-3.32.0 Adafruit-PureIO-1.1.9 PrettyTable-3.4.1 PyWavelets-1.4.1 Shapely-1.8.5.post1 adafruit-circuitpython-busdevice-5.2.3 adafruit-circuitpython-framebuf-1.4.14 adafruit-circuitpython-lis3dh-5.1.18 adafruit-circuitpython-register-1.9.14 adafruit-circuitpython-requests-1.12.8 adafruit-circuitpython-rplidar-1.2.10 adafruit-circuitpython-ssd1306-2.12.11 adafruit-circuitpython-typing-1.8.2 docopt-0.6.2 donkeycar h5py-3.7.0 imageio-2.22.2 imgaug-0.4.0 networkx-2.8.7 opencv-python-4.6.0.66 paho-mqtt-1.6.1 pillow-9.2.0 progress-1.6 pyfiglet-0.8.post1 pyftdi-0.54.0 pynmea2-1.18.0 pyusb-1.2.1 rpi-ws281x-4.3.4 scikit-image-0.19.3 simple-pid-1.0.1 sysv-ipc-1.1.0 tifffile-2022.10.10 tornado-6.2 typing-extensions-4.4.0 utm-0.7.0 wcwidth-0.2.5
```
they are probably ok.

Ezward, 2022-10-16:
A more serious issue is that the version of tensorflow we try to install will not install on bullseye
```$ pip install https://github.com/lhelontra/tensorflow-on-arm/releases/download/v2.2.0/tensorflow-2.2.0-cp37-none-linux_armv7l.whl
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tensorflow-2.2.0-cp37-none-linux_armv7l.whl is not a supported wheel on this platform.
```

Ezward, 2022-10-16:
So I tried using these instructions; https://qengineering.eu/install-tensorflow-2.2.0-on-raspberry-pi-4.html
However, they failed; can't install cython or liblapack because no package candidates.  Then I say this warning at the top of the page:
```TensorFlow 2.2.0 is not supported by Bullseye due to Python 3.9.
Only use the 32-bit Buster operating system with Python 3.7.
Bullseye supports TensorFlow 2.5 and above on a 64-bit RPi.
```
So unless we are moving to tensorflow 2.5, then we cannot support bulleye.

Ezward, 2022-10-16:
just attempting to install using pip does not work either
```$ pip install tensorflow
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement tensorflow
ERROR: No matching distribution found for tensorflow
```

Ezward, 2022-10-16:
For now I think I could ignore tensorflow for the purposes of creating the new camera driver, but I also don't see a need for the new camera driver if we cannot actually run on bullseye because of tensoflow.

TCIII, 2022-10-19:
@Ezward,

I have successfully installed tf 2.7 and OpenCV 4.5.3 on a Rpi 3B+ running the Bullseye 64 bit OS.

Tensorflow was installed using this [tutorial](https://qengineering.eu/install-tensorflow-2.7-on-raspberry-64-os.html).

OpenCV was installed using this [tutorial](https://qengineering.eu/install-opencv-4.5-on-raspberry-64-os.html). 
I recommend using a 64GB or larger SD card and make the swap file at least 4GB or larger.
Expect the OpenCV build and installation to take around six and 1/2 hours.

TCIIII