autorope/donkeycar #1053: fix auto-record on throttle for robothat MM1 controller

Ezward, 2022-10-22:
Reported by user `fooOO` in [discord]( https://discord.com/channels/662098530411741184/671604048218292233/1032996525497929748)

"Hi, Has anyone using RoboHAT MM1 and donkeycar 4.3.23? As I have configed the myconfig to record with throttle, but seems like it's not triggering the recording.
...
I did follow the instructions to give the right permission and disabled the nvgetty and I could use RC to control the car and in the Jetson nano serial monitor also output correct pwm info, I think it might be caused by something updated in the donkeycar 4.x, as it used to work with donkeycar 3.x. 

BTW I have tried by using the start record button on the web console it seems working properly, so I might just stick to this method."

Ezward, 2022-10-22:
I reviewed the code and it looks like we did not address the robothat controller when we added the code to make the game controllers and web controller cooperate with setting drive mode (user/autosteering/autopilot) and recording mode.  So I ported the code from the joystick controller into the robothat controller.  That is in branch [robothat-auto-record-on-throttle](https://github.com/autorope/donkeycar/tree/robothat-auto-record-on-throttle)

wallarug, 2022-10-22:
Hey @Ezward ,

It's been a few years now since I added this functionality.   So what exactly is still broken that may need my review?

It looks like you have already got a patch/fix in place.

CC: @peterpanstechland (foo00)

Ezward, 2022-10-22:
@wallarug Thanks for commenting.  We added code a while back to make the game controllers and web controller cooperate when changing the drive mode and recording mode.  We did not address the robothat at that time.  This PR https://github.com/autorope/donkeycar/pull/1055 fixes that by porting the code from the game controller to the robothat controller.  Basically we need to latch any  change that is made by the user so it can overrwrite the incoming state in run_threaded() sometime later.  This branch has been tested and shown to fix the issue.

Ezward, 2022-10-26:
Original poster reports that is is fixed https://discord.com/channels/662098530411741184/671604048218292233/1034481825202049157

closed 2022-10-26 by Ezward