autorope/donkeycar #1061: conda creating environment endlessly

arch-user-france1, 2022-11-15:
Conda won't finish doing that:
[image]
I left it running for more than five hours and it still was "examining ..." something

I did:

1. To get a stable release
2. If this is not your first install, update Conda and remove old donkey
3. Create the Python anaconda environment (recommended faster install time)

https://docs.donkeycar.com/guide/host_pc/setup_ubuntu/

What could I do?

arch-user-france1, 2022-11-15:
removing -n base -c conda-forge worked, however now it says "mamba" command not found and mamba is not a python module either
I then tried `pip install mamba` but now I end up with
`/usr/bin/python: No module named mamba.__main__; 'mamba' is a package and cannot be directly executed`

TCIII, 2022-11-15:
@arch-user-france1,

SBC/PC: ?
OS: ?

Was this your first install of conda, if so why did you perform #2?
I just installed mamba on my Xavier NX (JP502) running Ubuntu 20.04 without issue.

TCIII

Ezward, 2022-11-16:
The ubuntu install documentation that you linked already has the answer I think; 
```
conda install mamba -n base -c conda-forge
mamba env create -f install/envs/ubuntu.yml
conda activate donkey
pip install -e .[pc]
```

arch-user-france1, 2022-11-16:
That's what I did but it won't install mamba that way

TCIII, 2022-11-16:
@Ezward,

I believe that arch-user-france1's command line should say (base) when installing mamba, if conda has been installed correctly, and it doesn't so has conda been installed correctly?

TCIII

arch-user-france1, 2022-11-16:
> @Ezward,
> 
> I believe that arch-user-france1's command line should say (base) when installing mamba, if conda has been installed correctly, and it doesn't so has conda been installed correctly?
> 
> TCIII

Conda has been installed correctly. There is no wrong way.

sctse999, 2022-11-22:
I support this finding. Conda takes a lot of time to resolve dependencies and sometimes it is taking forever. I am suspecting that this is due to pytorch or other depdencies. 

@arch-user-france1 Could you please comment out the following and get back?

In `ubuntu.yml` (https://github.com/autorope/donkeycar/blob/main/install/envs/ubuntu.yml)

delete the following lines:

- pytorch=1.7.1
- torchvision
- torchaudio
- pytorch-lightning
- fastai

Delete the donkey environment and retry the installation.

TCIII, 2022-11-22:
@sctse999,

I just recently installed Conda on a Xavier NX 8GB running JP502 and I selected the Python 3.8 version of conda, for ARM processors, which installed without issues.
When I went to install mamba, the CLI indicated (base), unlike arch-user-france's CLI which says "donkey", which is the correct env prior to creating the donkey env.
Mamba installed without issue though we had to adjust the ubuntu.yml and [pc] for a successful install of DC.

TCIII

sctse999, 2022-11-22:
@arch-user-france1 try not to use mamba and skip this line. In the next line, use this:

`conda env create -f install/envs/ubuntu.yml`

sctse999, 2022-11-22:
> @sctse999,
> 
> I just recently installed Conda on a Xavier NX 8GB running JP502 and I selected the Python 3.8 version of conda, for ARM processors, which installed without issues. When I went to install mamba, the CLI indicated (base), which is the correct env prior to creating the (donkey) env, unlike arch-user-france's CLI which displays "donkey", Mamba installed without issue though we had to adjust the ubuntu.yml and [pc] for a successful install of DC.
> 
> TCIII

Unfortunate it doesn't work for me today on both Windows 11 and MacOS (Apple M1/M2). All of them are stuck in somewhere and cannot proceed.

TCIII, 2022-11-22:
@sctse999,

Interesting to say the least.

It took a lot of flag waving to get the `ubuntu.yml` and `[pc}` to run. I suggest that you get on the DC Discord channel and send DGarbanzo a private message explaining your issues and I believe he can help you.

TCIII

arch-user-france1, 2022-11-22:
> @arch-user-france1 try not to use mamba and skip this line. In the next line, use this:
> 
> `conda env create -f install/envs/ubuntu.yml`

so then we're doing

_Alternative (slower install time):_

which would be basically never ending

arch-user-france1, 2022-11-22:
> I support this finding. Conda takes a lot of time to resolve dependencies and sometimes it is taking forever. I am suspecting that this is due to pytorch or other depdencies.
>
> @arch-user-france1 [...]

trying this one right now; still solving environment, but we'll see if it's faster. Will respond once either complete or "hanging".

arch-user-france1, 2022-11-22:
still solving environment, so fastai might not be the problem

TCIII, 2022-11-22:
@sctse999,

When we installed ubuntu.yml we actually used jetson.yml which is [here](https://github.com/autorope/donkeycar/blob/tf_2_9/install/envs/jetson.yml) which removes the pytorch dependencies and modified kivy.

TCIII

sctse999, 2022-11-23:
> still solving environment, so fastai might not be the problem

Have you tried deleting pytorch and its friends too?

arch-user-france1, 2022-11-23:
> 

I did what you have asked for and nothing more

sctse999, 2022-11-23:
@arch-user-france1 so you have removed pytorch as well?

sctse999, 2022-11-23:
Can you use main instead of stable release tag?

arch-user-france1, 2022-11-25:
> Can you use main instead of stable release tag?

I didn't use stable last time

tylerjheslop, 2022-11-30:
I'm having this same issue on an M1 Mac

tylerjheslop, 2022-11-30:
I was able to get the environment running. 
I removed the packages mentioned above by @sctse999 
I ran [this command](https://github.com/datitran/object_detector_app/issues/41#issuecomment-588469805) `conda config --set restore_free_channel true`
Created the env using [Rosetta](https://github.com/Haydnspass/miniforge#rosetta-on-mac-with-apple-silicon-hardware): `CONDA_SUBDIR=osx-64 conda create -f install/envs/mac.yml`

sctse999, 2022-11-30:
> I was able to get the environment running. I removed the packages mentioned above by @sctse999 I ran [this command](https://github.com/datitran/object_detector_app/issues/41#issuecomment-588469805) [...]

@tylerjheslop Thanks for the information. 

what does `conda config --set restore_free_channel true` do? What's its effect?

I am not using Mac. Is it a must to use Rosetta?