autorope/donkeycar #1062: Donkey UI cannot launch due to protobuf versionn

sctse999, 2022-11-22:
I did a fresh donkey install on Windows 11 with miniconda. For some reason I remove the dependency of pytorch and fastai. As a result, my `pip list` shows that my protobuf version is 4.x.x

When I launch donkey ui, it fails and shows the stacktrace below: 
```
WARNING:kivy:stderr:     _message.Message._CheckCalledFromGeneratedFile()
 TypeError: Descriptors cannot not be created directly.
WARNING:kivy:stderr: TypeError: Descriptors cannot not be created directly.
 If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
WARNING:kivy:stderr: If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
 If you cannot immediately regenerate your protos, some other possible workarounds are:
WARNING:kivy:stderr: If you cannot immediately regenerate your protos, some other possible workarounds are:
  1. Downgrade the protobuf package to 3.20.x or lower.
WARNING:kivy:stderr:  1. Downgrade the protobuf package to 3.20.x or lower.
  2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).
WARNING:kivy:stderr:  2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

WARNING:kivy:stderr:
 More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
WARNING:kivy:stderr: More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
```

I then pinned the protobuf version to `3.20.3` and tried again. Donkey UI launched successfully. 

I think we should pin the protobuf version to 3.20.x. Any thoughts?

DocGarbanzo, 2022-11-23:
I think we should be doing what you suggest in the windows yaml file.

sctse999, 2022-11-23:
How about ubuntu and mac yaml? Should we pin them as well?

Let me send a PR afterwards.

closed 2022-12-02 by DocGarbanzo