autorope/donkeycar #1071: Improve calibration steps for IMU

Ezward, 2022-12-12:
We call the calibrateMPU6500 method, but we do no calibration of the magnetometer on the MPU9250.  We could do much better and we need to if we want to use the IMU to improve path following.  See this repo for a set of python scripts that do a more thorough calibration https://github.com/makerportal/mpu92-calibration

TCIII, 2022-12-12:
@Ezward,

The makerportal mpu92 calibration documentation is very impressive and through.

TCIII

Ezward, 2023-02-06:
Here are some links for integrating IMU and for fusing IMU with GPS

- https://pypi.org/project/droneposelib/. dronepose lib is a python library that will make pose estimates based on IMU data.
- https://pypi.org/project/imusensor/ library to make reading the IMU easy.  Look at the acknowledgements at the bottom of the page.
- https://github.com/micropython-IMU/micropython-fusion micro-python library implements Madgwick
- https://github.com/xioTechnologies/Fusion AHRS fusion in C with Python bindings.  Includes some error rejection.
- https://github.com/smahajan07/sensor-fusion Python fuse GPS with IMU for position and orientation estimation

Ezward, 2023-02-15:
The driver that we currently use also has calibration methods; https://pypi.org/project/mpu9250-jmdev/