autorope/donkeycar #1074: Error into  Dockerfile

salamisodikiolawale, 2022-12-17:
Hello, if I have to build a docker image with the current dockerfile project, I get this error: 
> [C 08:34:18.925 NotebookApp] Bad config encountered during initialization: 
> [C 08:34:18.925 NotebookApp] No such notebook dir: ''/app/notebooks''.

So I propose to solve this error with my current contribution #1073 with  commmand 
> **RUN mkdir notebooks** 
into Dockerfile if you don't mind.

Ezward, 2023-02-11:
Thank you for contributing to the Donkeycar project.  Here are some guidelines that can help you be successful in getting your contributions merged into Donkeycar.

**What makes for a good Pull Request?**
- The code implements a bug fix or optimization/refactor to a preexisting feature or adds a new feature.
- If the code is adding a new feature, it would be advisable to open an Issue in the Donkeycar repo prior to opening the PR.  This would allow discussion on how the feature may be best implemented and even if the feature is something that would get accepted into Donkeycar.  The new feature should have fairly broad applicability; it should be a feature that would be useful to a lot of users.
- The code must work on our currently supported platforms; Raspberry Pi OS and the Jetson Nano for cars and Linux, Mac and WSL on Windows.
- The code should have unit tests.
- The code should work on the version of Python specified in the installation documentation at docs.donkeycar.com
- The PR should include instructions/steps telling how the feature/fix can be tested by a person.  In some cases you may want to create a video and link it from Youtube or Vimeo to show the process; for instance if it includes mounting hardware.
- For new features or changes to preexisting feature you should also open a PR in the documentation repo https://github.com/autorope/donkeydocs with updated docs.  A human tester will want to refer to that when testing.

**What makes for a Pull Request that is not likely to get accepted?**
- It adds a feature that is not useful to a broad audience or is too complex/complicated for the Donkeycar audience.  For instance, it adds a driver for a custom piece of hardware or a piece of hardware that is not generally obtainable.  In this case we will encourage you to maintain the feature in your own fork on Donkeycar and keep that fork up to date with the main branch.
- It does not have unit tests.
- The PR is opened and it receives comments and/or requests for change, but no response is made by the owner of the PR.  In this case we will eventually close the PR as unresponsive.

**Pull Request Process**
- Fork the Donkeycar repository into your own github account.  See [About Forks](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/working-with-forks/about-forks) in the github docs.
- Create a branch in your repository and make the changes.  
- Open a pull request to the Donkeycar main repository.  If it is associated with a github issue then reference the issue in the pull request.  See [Creating a Pull Request from a Fork](https://docs.github.com/en/pull-requests/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request-from-a-fork) in the github docs.
  - If the PR is associated with an Issue then link the issue in the PR description.
  - If the PR is associated with a documentation change then link to the associated PR in the documentation repo.
- The maintainers will be automatically notified of the pull request.  
- Maintainers will provide comments and/or request for changes in the PR.  This could take a little while; we have a small volunteer team that is working on a number of initiatives.  You can get more visibility by announcing the PR in the software channel on the Discord.
- The owner of the PR should be checking it for comments and/or request for changes and responding.  In particular, if there are requests for changes but you cannot get to them reasonably quickly then add a comment in the PR that helps us understand your timeframe so that we don't close that PR as unresponsive.
- There is a possibility that we choose to not move forward with the PR and it will be closed.  You can minimize that chance by discussing the feature or fix in an Issue prior to opening a PR (see above).
- If once all requested changes are made then the PR can be accepted.  At this point one of the maintainers will merge the PR and the PR will be closed as completed.  Congratulations, you just made the world better.

Ezward, 2023-02-27:
@salamisodikiolawale what is the status here?  Do you plan to open a PR with your proposed fix?

Ezward, 2023-03-09:
Closing as unresponsive.

closed 2023-03-09 by Ezward