autorope/donkeycar #1080: Is there a donkey APP which is same name with donkey car？

hsy75, 2023-01-01:
root@DESKTOP-M4INDEG:/home/donkeycar/proj01/donkeycar# donkey createcar --path /mycar
donkey: unrecognized option '--path'
usage:  donkey [-n num] [-f func] sequence seed
        donkey -i
        donkey -h | --help | -v | --version | --features

I try to Create a Car in WSL enviroment ,and found that thie donkey command is not support createcar ,and it has it own usage, is that same one ,where is the right one?

DocGarbanzo, 2023-01-01:
@zlite - maybe you can help w/ WSL?

dlarue, 2023-02-18:
@hsy75  are you sure you followed all the steps to install donkey? I just finished a clean install and when I run either 'donkey -h' or 'donkey createcar -h' I get quite different results from what you are showing. See below:

> Successfully installed Kivy-Garden-0.1.5 donkeycar-4.4.0
> (donkey) maker@Maker-MacBookAir:~/Projects/donkeycar$ donkey -h
> ________             ______                   _________              
> ___  __ \_______________  /___________  __    __  ____/_____ ________
> __  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
> _  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
> /_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
>                                  /____/                              
> 
> using donkey v4.4.0 ...
> Usage: The available commands are:
> ['createcar', 'findcar', 'calibrate', 'tubclean', 'tubplot', 'tubhist', 'makemovie', 'createjs', 'cnnactivations', 'update', 'train', 'models', 'ui']
> (donkey) maker@Maker-MacBookAir:~/Projects/donkeycar$ donkey createcar -h
> ________             ______                   _________              
> ___  __ \_______________  /___________  __    __  ____/_____ ________
> __  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
> _  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
> /_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
>                                  /____/                              
> 
> using donkey v4.4.0 ...
> usage: createcar [options]
> 
> optional arguments:
>   -h, --help           show this help message and exit
>   --path PATH          path where to create car folder
>   --template TEMPLATE  name of car template to use
>   --overwrite          should replace existing files

Ezward, 2023-02-27:
Closing as not responsive.

closed 2023-02-27 by Ezward