autorope/donkeycar #1081: Anyone success in WSL enviroments, seems the guider from Donkeycar.com IS ERROR?

hsy75, 2023-01-01:
Install Donkeycar on Windows (WSL)
The Windows Subsystem for Linux (WSL) lets developers run a GNU/Linux environment -- including most command-line tools, utilities, and applications -- directly on Windows, unmodified, without the overhead of a traditional virtual machine or dualboot setup.

Install [Windows Subsystem for Linux](https://docs.microsoft.com/en-us/windows/wsl/install-win10).
If using Windows 10 (this is not necessary for Windows 11), turn on Windows 10 "Windows Subsystem for Linux" Feature (Settings > Apps > Programs and Features > Turn Windows features on or off)
Download a Linux Distribution from the Microsoft Store (recommend [Ubuntu](https://www.microsoft.com/en-us/p/ubuntu/9nblggh4msv6?activetab=pivot:overviewtab) Latest)
Open the Ubuntu App and configure.

Open the Ubuntu App to get a prompt window via Start Menu | Ubuntu

Install pip using sudo apt install python3-pip

Install the following two libraries for the UI to work: sudo apt-get install libmtdev1 xclip

Change to a directory that you would like to use as the head of all your projects.

mkdir projects
cd projects
Get the latest donkey from Github.
git clone https://github.com/autorope/donkeycar
cd donkeycar
git checkout main
NOTE: The main branch has the latest (unstable) version of donkeycar with experimental features.

Get a stable release from Github:
git clone https://github.com/autorope/donkeycar
cd donkeycar
git fetch --all --tags
git checkout tags/4.3.6.1
Install Donkeycar into Python
pip3 install -e .[pc]
Once you're done with this, close the Ubuntu terminal and then reopen it so the path changes take effect.

Experimental Support - GPU Users: Install Tensorflow GPU - only for NVIDIA Graphics cards
If you have an NVIDIA card, you should update to the lastest drivers and [install Cuda SDK](https://www.tensorflow.org/install/gpu#windows_setup).

pip3 install tensorflow
Create your local working dir:
donkey createcar --path /path/to/projects/mycar --template complete

hsy75, 2023-01-01:
I just following the guide above ,but it sees the guide has some problem ,it is incomplete, and maybe WSL is same as Linux guide right?

hsy75, 2023-01-01:
donkeycar@DESKTOP-M4INDEG:~/projects/donkeycar$ donkey
Command 'donkey' not found, but can be installed with:
sudo apt install donkey

Ezward, 2023-02-27:
@hsy75 can you ask this question in discord?  You are much more likely to find someone who is using WSL.