autorope/donkeycar #1084: Add support for 4 wheel differential drive trains

Ezward, 2023-01-09:
There are a number of cars that use 4 motors in a differential drive configuration.  It is a simple change to support such cars.  One such car is from Freenove.  I've added an issue in their github, hoping they would implement this; https://github.com/Freenove/Freenove_4WD_Smart_Car_Kit_for_Raspberry_Pi/issues/41  We do have a user [@jjakubassa](https://github.com/jjakubassa) with this car; they have opened a PR for this issue; [add support for four motors differential drive](https://github.com/autorope/donkeycar/pull/1082)

Ezward, 2023-01-09:
@jjakubassa have you been able to collect data and train an autopilot?

jjakubassa, 2023-02-20:
No, I didn't get this far. The car drives left and right as expected, but the turn radius isn't great (at least for the small place I drove in). I think, the radius could be reduced if one side would rotate backwards, as in the driving test in the freenove code. But even with the wide turn radius the vibrations from the friction were too much for the car (wheels or camera falling off). With a different a car design it might work better ([see](https://www.rakeshmondal.info/4-Wheel-Drive-Robot-Design)). Finally, I decided to go with a more standard donkey car build (which works great).

Ezward, 2023-02-27:
Thanks for the response @jjakubassa .  Yes our differential drive code just stops one wheel in the turn, it cannot drive it in reverse.  It is good info to know that these kind of chassis do not work well with donkeycar.  I'm closing the issue.  Thanks.

closed 2023-02-27 by Ezward