autorope/donkeycar #1090: link is not valid encoder.py

Ahrovan, 2023-02-04:
https://github.com/autorope/donkeycar/blob/main/donkeycar/parts/encoder.py

# The microcontroller should be flashed with this sketch (use the Arduino IDE to do that): **https://github.com/zlite/donkeycar/tree/master/donkeycar/parts/encoder/encoder**

This link not valid. please update link to Arduino sketch (ino file)

TCIII, 2023-02-04:
@Ahrovan,
The Arduino sketches for both the mono and quadrature encoders can be found [here](https://github.com/autorope/donkeycar/tree/921-kinematics-pose-estimation/arduino) now.

You can copy and paste the sketch of your choice into a new Arduino IDE sketch and flash into an Arduino of your choice.
I use the Adafruit Metro Mini with USB connector, but an Arduino Mini will work too as long as it has a USB connector.

TCIII

Ezward, 2023-02-11:
@Ahrovan did you give the encoder sketches in the PR at try?

Ahrovan, 2023-02-11:
> 

Next week I will do and reply here

Ezward, 2023-02-21:
@Ahrovan have you been able to try this?

Ezward, 2023-02-27:
Closing as unresponsive.

closed 2023-02-27 by Ezward