autorope/donkeycar #1093: Failed to allocate memory

Ahrovan, 2023-02-11:
`donkey train --tub ./data --model ./models/myModel.h5`
### 2 root error(s)

(0) Resource exhausted:  Failed to allocate memory for the batch of component 0
         [[node IteratorGetNext (defined at /projects/donkeycar/donkeycar/parts/keras.py:183) ]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

         [[IteratorGetNext/_6]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

  (1) Resource exhausted:  Failed to allocate memory for the batch of component 0
         [[node IteratorGetNext (defined at /projects/donkeycar/donkeycar/parts/keras.py:183) ]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

Hardware/Software Details:
- Jetson Nano B01
- donkeycar 4.4.0
- Used #190 for solve libgomp.so.1 Error
- /data folder sizze 98MB, About 3k Images
- Tensorflow 2.3.1
- Used [jetson-nano-jp451-sd-card-image.zip](https://developer.nvidia.com/embedded/l4t/r32_release_v5.1/r32_release_v5.1/jeston_nano/jetson-nano-jp451-sd-card-image.zip) as Image
-

Ezward, 2023-02-11:
That branch is 6 years old.  You should try the main branch.  We are also close to merging a new branch that upgrades to jp4.6 and Tensorflow 2.9.

Ahrovan, 2023-02-11:
Thank you, @Ezward how to try new? 
must to be install jp4.6 ?

Ahrovan, 2023-02-12:
@Ezward Failed to allocate memory

using donkey v4.4.4-main ...
INFO:donkeycar.config:loading config file: ./config.py
INFO:donkeycar.config:loading personal config over-rides from ./myconfig.py
2023-02-11 19:44:42.958070: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.10.2
WARNING:donkeycar.pipeline.database:No model database found at /home/ahrovan/Projects/myCar/models/database.json
INFO:donkeycar.utils:get_model_by_type: model type is: linear
2023-02-11 19:44:52.458462: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcuda.so.1
2023-02-11 19:44:52.499566: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1046] ARM64 does not support NUMA - returning NUMA node zero
2023-02-11 19:44:52.499729: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1742] Found device 0 with properties:
pciBusID: 0000:00:00.0 name: NVIDIA Tegra X1 computeCapability: 5.3
coreClock: 0.9216GHz coreCount: 1 deviceMemorySize: 3.87GiB deviceMemoryBandwidth: 194.55MiB/s
2023-02-11 19:44:52.499810: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.10.2
2023-02-11 19:44:52.663623: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcublas.so.10
2023-02-11 19:44:52.736950: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcufft.so.10
2023-02-11 19:44:52.843827: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcurand.so.10
2023-02-11 19:44:52.975454: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcusolver.so.10
2023-02-11 19:44:53.048683: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcusparse.so.10
2023-02-11 19:44:53.052396: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudnn.so.8
2023-02-11 19:44:53.052787: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1046] ARM64 does not support NUMA - returning NUMA node zero
2023-02-11 19:44:53.053117: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1046] ARM64 does not support NUMA - returning NUMA node zero
2023-02-11 19:44:53.053198: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1884] Adding visible gpu devices: 0
2023-02-11 19:44:53.084356: W tensorflow/core/platform/profile_utils/cpu_utils.cc:108] Failed to find bogomips or clock in /proc/cpuinfo; cannot determine CPU frequency
2023-02-11 19:44:53.085005: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x3eb00150 initialized for platform Host (this does not guarantee that XLA will be used). Devices:
2023-02-11 19:44:53.085078: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): Host, Default Version
2023-02-11 19:44:53.222570: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1046] ARM64 does not support NUMA - returning NUMA node zero
2023-02-11 19:44:53.223062: I tensorflow/compiler/xla/service/service.cc:168] XLA service 0x3eb7a360 initialized for platform CUDA (this does not guarantee that XLA will be used). Devices:
2023-02-11 19:44:53.223133: I tensorflow/compiler/xla/service/service.cc:176]   StreamExecutor device (0): NVIDIA Tegra X1, Compute Capability 5.3
2023-02-11 19:44:53.248679: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1046] ARM64 does not support NUMA - returning NUMA node zero
2023-02-11 19:44:53.248849: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1742] Found device 0 with properties:
pciBusID: 0000:00:00.0 name: NVIDIA Tegra X1 computeCapability: 5.3
coreClock: 0.9216GHz coreCount: 1 deviceMemorySize: 3.87GiB deviceMemoryBandwidth: 194.55MiB/s
2023-02-11 19:44:53.248996: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.10.2
2023-02-11 19:44:53.249223: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcublas.so.10
2023-02-11 19:44:53.249356: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcufft.so.10
2023-02-11 19:44:53.249468: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcurand.so.10
2023-02-11 19:44:53.249573: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcusolver.so.10
2023-02-11 19:44:53.249676: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcusparse.so.10
2023-02-11 19:44:53.249780: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudnn.so.8
2023-02-11 19:44:53.250062: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1046] ARM64 does not support NUMA - returning NUMA node zero
2023-02-11 19:44:53.250353: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:1046] ARM64 does not support NUMA - returning NUMA node zero
2023-02-11 19:44:53.250446: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1884] Adding visible gpu devices: 0
2023-02-11 19:44:53.250572: I tensorflow/stream_executor/platform/default/dso_loader.cc:49] Successfully opened dynamic library libcudart.so.10.2
[... cut ...]
         [[node IteratorGetNext (defined at /Projects/donkeycar/donkeycar/parts/keras.py:183) ]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

         [[IteratorGetNext/_6]]
Hint: If you want to see a list of allocated tensors when OOM happens, add report_tensor_allocations_upon_oom to RunOptions for current allocation info.

0 successful operations.
0 derived errors ignored. [Op:__inference_train_function_1794]

Function call stack:
train_function -> train_function

Ezward, 2023-02-21:
What machine are you running training on?  Maybe your are trying to train on a Jetson Nano?  I have heard of others that did this, but it is not a supported training machine.  I guess I would try to reduce the batch size by changing `BATCH_SIZE` configuration; try 16.

Ahrovan, 2023-02-22:
@Ezward yes try to train with Jetson Nano B01, I changed BATCH_SIZE to 1, but failed

DocGarbanzo, 2023-02-22:
@Ahrovan - you seem to be using an image size of 600x800. The linear model that you are trying to train is geared towards an image size of 120x160, and has about 500k parameters for that size. For a 600x800 you would need a model with higher compression, i.e. more layers or larger strides. You can see that your model now has 40,000,000 parameters (look at the Flattened layer from your model: you are coming out with a 400k dimensional vector and then going into a 100d dense layer, this basically gives you the 40m parameters). This model is far to big to fit into the ram of the nano. So either, set the image size to the standard of 120x160 or modify the model architecture.

Ahrovan, 2023-02-23:
@DocGarbanzo Thank you

Ezward, 2023-02-27:
Note that image should be 120 high, 160 wide.  @Ahrovan have you retried using the new image size?

Ezward, 2023-03-04:
@Ahrovan I am going to close this; if you have more info please do add a comment to the closed issue.

closed 2023-03-04 by Ezward