autorope/donkeycar #1097: Map controller button/axis changes to behavior

Ezward, 2023-02-18:
## Problem Statement

Our current system hard-codes behavior in the joystick parts.  This has unwanted side-effects;
- changes in linux drivers break our button/axis -> behavior mappings.
- users are stuck with the way we have mapped buttons/axis to behaviors; they cannot change which buttons do what (see Note 1 at bottom) and they cannot assign new behaviors.
- The current mappings assume the deep learning template.  However we also have the path follow and computer vision templates and they have very different needs for the controller buttons/axes.  In the near future we will make it easy for users to construct their own 'template'/vehicle and so they will need to decide how buttons should be mapped for their purposes.
- The code is complex, redundant and brittle.  Game controllers are difficult to add or modify because the code does too much.

We know that XBox joystick has different mapping under different SBCs and even within different releases of the OS on a given SBC.  We have split the joystick parts into two pieces; 
- **Joystick**: Each kind of controller has a part subclassed from the Joystick() abstract class.  This maps axis (analog) values and button presses (boolean 0 or 1 values) to names.
- **JoystickController**: Each kind of controller also has a part subclassed from the JoystickController() class.  This calls functions and/or sets output values based on the changes in the axes and button states.

Together these two parts effectively map a button code to a function.  The problem is that different drivers for the same controller (different across drive versions or OS versions of SBC versions) output different codes.  So the 'X' button on an xbox controller may be 0x05 on one driver and 0x23 on another.  The breaks everything because the mapping of code to name in the Joystick() may be completely missing or may map to the wrong actually axis or button.  The canonical example is the difference between the x-box controller on RaspberryPi and Jetson Nano.

The other issue is that every JoystickController subclass must implement the behaviors itself.  (1) this is super redundant (2) this makes it very hard to modify what the axes and buttons do (3) this breaks if the driver changes the output codes and see (2). (4) We now have not just the deep learning template; we have the path follow and computer vision templates and they have different needs for the controller should behave.

## Goals:

- Make it easy to handle the fact that the same game controller may publish buttons and axes using different codes on different OS or version of an OS or SBC.    
- Make it very easy to remap the behaviors the run when a given button is pressed or joystick is moved.  So make it easy using configuration to change which button toggles recording for instance.
- In the process greatly simplify how we support game controllers.

## Proposed Direction:
Replace hard-coded behavior we have in our controllers with an event system where the controller emits a event with a name and value when an axis/button changes, then an event handler (a part) can handle the event to implement the behavior.  
    - We could implement this as a single-stage mapping where low-level codes are mapped directly to high level event names.  So we may map a button with code 0x05 to the 'toggle-recording' event.
    - OR we could implement a two-level mapping where we map low-level controller codes to low-level names, then map low-level names to high level events.  So we have a table that maps low-level button code 0x05 to 'button-1' and then we have a separate part to uses a lookup table to map 'button-1' to the 'toggle-recording'.  

Using more generic low-level button names, like 'button-1' means we can assign these simply by asking the `/input/js0` device for the list of axes and buttons, then we just assign names base on order.  This is exactly what `pygame` does.  So then we would map these low level names to high-level events.  That second mapping is the one that we want to design so it is easy to change the mapping in configuration.  We will want a set of lookup tables that handle all of our current cases.  

We would still be left with the issue that the button/axis names might differ between drivers.  We can handle differences between drivers by including extra tables.  So we might have 'xbox' lookup table to handle the xbox controller on Raspberry Pi and an 'xbox-jetson' lookup table to handle the same controller on the Jetson because the xbox driver on Jetson outputs a different set of button-#/axis-# names than it does on Raspberry Pi.

Beyond the mapping of low-level codes to high-level events (be that in a single-stage to two-stage system) we need a system that delivers the event to a part that implements the necessary behavior.  So we would have a part that handles toggling the recording state and it would run when we get a 'toggle-recording' event. 
- The way I do this in path_follow is that I treat the high-level event name as a latched run_condition; when the event is emitted it is simply added to the vehicle memory using it's name.  Then there is a part that uses that name as it's run_condition; so it will run when the event name is added to the vehicle state.  At the end of the vehicle loop the event will be removed from the vehicle memory; this makes it so it only fires for one iteration the vehicle loop.  This system makes it easy to fix the ordering of event handling relative to other parts; it's just a normal vehicle part with a run_condition; it will run in the order it is added to the vehicle and only when the run condition exists.
- Alternatively we might add a true event pub/sub system. We have to think about the order that things run; a true pub-sub system might run thing in different order at different times. 
- Perhaps we compromise and rather than create a full pub-sub system we enhance what we already have to use a 'subscribe=<event-name>' argument when adding a part; the part will run if an event is present with the name; all events are automatically cleared at the end of a vehicle loop.  This is basically like what I did using the run_condition, but it would not overload that system and it might have clearer semantics.  It would have the benefit of working like parts work now, including ordering guarantees.

Note 1: We have the JoystickWizard that creates a custom joystick part that maps low-level code to a low level name. So technically users can remap buttons/axes to existing functions; but they cannot assign new functions.  This is also complex; the Joystick Wizard ends up generating a Python class that it writes the the root of the my_car folder.  If the user change their joystick preference to "custom" then add_user_controller() looks for my_joystick.py in the my_car folder and imports it dynamically.  It would be much better if the wizard just wrote out a lookup table (a standard csv file) and the custom joystick always used that as it's lookup table.  Then we would not have to had the weird code that dynamically load the 'my_joystick.py' part.  The user could easily remap things by editing the file.

DocGarbanzo, 2023-02-18:
@Ezward - this is a very thorough and comprehensive description of the problem and the possible solutions. I had to  prototype a solution already in order to get the generic vehicle template approach to work, where we need to rely on all parts to communicate through the `run()` or `run_threaded()`.

Without spending too much time on a strategic solution, I directly implemented a solution where the controller would simply spit out a map of the buttons that are pressed or released. This map is returned from the `run()` or `run_threaded` method and then one only needs a new generic part, say `ButtonInterpreter` which translates the button map into boolean variables. These variables simply get added to the vehicle memory and can then be used as `inputs` or `run_condition` in other parts. Obviously this requires the updating of some Part's `run()` methods. For example, the `TubWriter` would need an additional `bool` argument to check if it should remove the last `n` records and similar. If we think we might create too many `bool` arguments, then we would need to pass all of them in another dictionary and share that dictionary as input parameter for all parts that allow for some 'special' controller interaction.  

Note, I don't think these variables should get cleared at the end of the vehicle loop. The reason is that sometimes the parts topology in the car can become complicated, e.g. for some reason you would want the AI pilot to run before your controller and for some other reason you want to run it afterwards, usually because you have some cyclic dependencies, so part one's (`p1`) input depends on part two's (`p2`) output and `p2`'s input depends on `p1`s output. You simply put these part in the order where in which you want to reduce latency, say `p1` before `p2` and then `p1` will pick up `p2`'s output in the next vehicle loop. For buttons to delete records or switch some behaviour on-the-fly, a one-loop latency is absolutely no problem, but if we were to clear them at the end of the loop we would miss out on using the output like mentioned above. 

As you say, the mapping of physical buttons to button names might be platform dependent, so it should be taken care of in the controller itself, I would think.  The mapping of buttons, say `X` to toggle recording should be happening in the vehicle design, because it depends on which vehicle or template you run and which controller you own and what your preferences are. I can also imagine to build an enhanced `ButtonInterpreter` which would allow some debouncing, or latching or distinguish between short and long presses. Something I have started using on my RC control, because it has only a single button (it's 3 channel).

Ezward, 2023-02-19:
@DocGarbanzo can you give me a link to the branch where you are developing generic vehicle template approach?

Ezward, 2023-02-19:
Looking at current Joystick and desired kinds of events, I think the layers wild be

- Joystick tracks low level device events.  This are retrieved, one at a time, by calling the poll() method.  Poll() blocks until an event is available, so it should be called in a thread.  Note that this is NOT a part; it is more like a device driver
  - button_change: (timestamp, button_id, button_value)
      - timestamp is milliseconds
      - button_id is string like, "button_1" 
      - button_value is 1 for down and 0 for up
  - axis_change: (timestamp, axis_id, axis_value)
      - timestamp is milliseconds
      - axis_id is string like, "button_1" 
      - axis_value is -32767 to 32767

- ControllerEvents - A threaded part that turns low level button_change events into clicks.  The update() continually calls the Joystick.poll() in a thread.  It maintains a two lists; a list of button events and a list of axis events.  
  - button_change: these values are simply a pass through of that Joystick.poll() returns.
  - button_click: represents a button_down followed by button_up
    - timestamp is milliseconds of the button_up
    - button_id is string, like "button_1"
    - fast_click is a count of sequential fast_clicks, the number of clicks in sequence where the time between clicks is at or below the quick_threshold.  Basically this indicates double-click, triple-click, etc. without having to have explicit separate events.
    - long_click is a count of sequential long clicks; the number of clicks in sequence whose duration is at or above the long_threshold.
  - axis_change: these values are simply a pass through of what Joystick.poll() returns.
  - The run_threaded() method returns the button and axis event lists and then clears the internal queue to start new lists.

- AxisButtonEvents - A standard (non-threaded) part that takes as inputs the button and axis lists emitted by the ControllerEvents part expands the events into individual events for each button/axis.  This simply takes the most recent button click for a button and the most recent axis change for an axis and publishes them using the name of the button/axis.  For instance, "event/click/button_1" the the most recent button_1 click event.  The data include the timestamp, button_id, fast_click count and long_click count.  For an axis it might be "event/axis/axis_3" and the data would be the timestamp, axis_id and axis_value. 

It is possible for parts that implement behavior to use the outputs of the ControlEvents part or the AxisButtonEvents part directly.  However, it may be better to have another layer that turns these device input events into high level events.  For instance, rather than a part that looks in the list of button events for a click event on "button_1" to toggle recording, it would be better to have a part that maps a "event/click/button_1" event to a "toggle-recording" event.  

I think we don't want to have to have one part per translation.  Ideally this would be done in a single part.  We could use the order of the list of inputs and outputs as the association; so the first input would map to the first output, etc.  We could get the list of inputs and outputs from configuration.

We still have to deal with the different controller and os versions.  We might create a function that takes in the OS name and configured controller name and uses that to get the correct lookups for event translation.

DocGarbanzo, 2023-02-20:
> @DocGarbanzo can you give me a link to the branch where you are developing generic vehicle template approach?

@Ezward - this is here: https://github.com/DocGarbanzo/donkeycar/tree/part_factory

DocGarbanzo, 2024-01-14:
I have one more comment to the Event System. I think if we add one-shot events to the vehicle memory, these events must be cleared by the part that generates those. Say a one-click event would be registered, only if the button is pressed through at least 2 vehicle loops and or a double click requires at at least 2 vehicle loops between the single clicks. Such an event would be issued when registered from the controller and would be cleared in the next loop. This will allow the parts that react to that event to be placed either before or after the controller in the vehicle loop. I think it is important to have that flexibility, and we already have parts which are issuing data into the memory that will be picked up by other parts only in the next loop, for example the current tub index is read by the webcontroller only in the next iteration. The causal topology structure in the car is not a DAG. 

For your questions above, how to correctly recognise short or long or double clicks, I am already using some of this for my channel 3 controller, so I have some code for it. The controller part will issue a different message when pressed long (for me this issues a Ctrl-C like event to stop the vehicle) vs pressing short (which will clear the last 1s of data from the tub).

Ezward, 2024-02-04:
@DocGarbanzo can you share your code with me?

Ezward, 2024-05-28:
A proof of concept is available on branch [1097-refactor-game-controller-system](https://github.com/autorope/donkeycar/tree/1097-refactor-game-controller-system).  I've also opened a [draft pull request](https://github.com/autorope/donkeycar/pull/1180) to make it easier to see the changes and to comment on the code; that PR is not intended to ever be merged.  Please address comments regarding the approach to replacing the legacy joystick code in this issue rather than the PR.  Make comment that are specific to the code in the PR.

This lengthy comment is a discussion of the approach taken to refactoring the joystick code as embodied in the branch.

>> running the POC
>> - These instructions assume the developer installation has been done.
>> - Change the working directory to the root of the donkeycar project folder and check out the branch `1097-refactor-game-controller-system`
>> - Activate the donkeycar environment, `source env/bin/activate`
>> - to run the vehicle loop `python3 donkeycar/parts/controller_events.py`
>> - to run the simulated loop with event logging: `python3 donkeycar/parts/controller_events.py --vehicle=false`

The current Donkeycar game controller code has tightly bound many behaviors, like toggling the pilot mode with a button press or setting throttle with an joystick (axis) movement, to specific controls on specific makes of game controllers.  That legacy design is a problem because we can't remap buttons/axes to different behaviors.  That is a drag for two reasons; first it makes supporting new templates or new template behavior more difficult and second it makes supporting Dirk's declarative vehicle assembly basically impossible.  This POC shows how we can eliminate `JoystickController` completely and instead generate input events that can be handled by standard donkeycar parts. Once we can handle button presses and axis movements with standard parts, then we are open to remapping template behaviors to any input control that we want.  Further, because these behaviors are standard donkeycar parts, they can be assembled using Dirk's declarative syntax.

The first thing we need to do is surface input control changes as one-shot events.  The  POC implements a one-shot event as values in memory that exist for a single complete pass through the vehicle loop and are then removed.  Contrast this to 'normal' memory state, which is persistent for the entire life of the vehicle loop.  We want events to be one-shot so they can trigger code once in the loop when the are emitted and not over and over again on subsequent passes through the loop.  

In addition to their one-shot nature, button and axis events must be named in a regular, predictable way so using them as inputs or run_conditions is easy.

We already have a set of classes that can read controller inputs and rename them to something predictable (to the name of the buttons/axes on the game controller that is configured).  That is the set of classes in [controller.py](https://github.com/autorope/donkeycar/blob/main/donkeycar/parts/controller.py) based on the [Joystick](https://github.com/autorope/donkeycar/blob/f81750df7fb626a553e757fd792880721fc7c11e/donkeycar/parts/controller.py#L17) class.  The 'Joystick' class is specifically to read input events from the linux `/dev/input/js0` device.  There are other classes in controller.py that implement other kinds of controller, like [RCReceiver](https://github.com/autorope/donkeycar/blob/f81750df7fb626a553e757fd792880721fc7c11e/donkeycar/parts/controller.py#L246) the class that to read buttons and steering/throttle from an RC controller.  This classes don't all conform to a uniform specification because there was not such specification.  The first part of the POC is to create that specification and refactor the code, where necessary to use it.  This begins by creating a standard input controller api, `AbstractInputController`, that all input controllers that want to participate in the input event system must implement.

**`AbstractInputController`**: an abstract class that all input controllers must be derived from.  The most important method is `poll()`, which returns a tuple which can contain one button state hange and one axis state change.  `poll()` must be called continuously to retrieve input changes otherwise device driver buffering can cause problems. This is the method called by InputControllerEvents() it its threaded `update()` loop so input changes are retrieved in the background.  

**`LinuxGameController`**: an implementation of `AbstractInputController` for the linux `/dev/input/js0` devices.  The constructor can take a dictionary that maps from device-driver button/axis numbers to human readable button and axis names.  This is how we map to native game controller names for each make of game controller, like button 0x133 -> 'X' or 0x13b -> 'start' and axis 0x03 -> 'right_stick_horz' for the Logitech F710.  So the `LinuxGameController.poll()` method will emit state changes with the mapped button and axis names.  Any button or axis that is not mapped will get a default name based on the device driver control number; for instance the prior button/axis examples would be 'button(0x133)', 'button(0x13b)', 'axis(0x03)' if no mapping is provided for them.

**`LogitechJoystick`**: an implementation of `LinuxGameController` that provides a button/axis mapping for the Logitech F710 game controller.  This provides an example of how we would implement that mapping for the set of ga [... cut ...]
Throttle = 6.103701895199438e-05
```

The POC has enough in it to make me feel that this is a good direction, but we should discuss it before I put more work in.  I've put a lot of detail into the explanation of the POC and the the event stream in order to elicit some discussion.  I think what is implemented works well for the use cases of which I am aware.  However there may be better alternatives or there may be use cases I don't know about that don't fit well into this scheme.  So constructive input is desired.  Give the