autorope/donkeycar #1100: Create a cone detector and local planner to navigate around cones

Ezward, 2023-02-22:
1. Create a cone detector.  This can be based on the stop sign detector.  We may need to do some transfer learning to add more cones to the dataset.  We only care about the cone category.
- [Cone Detection Dataset](https://www.fsoco-dataset.com/)

2. In path follow template, create a local planner that will modify the robot's path to avoid a cone detected by the cone detector.

Ahrovan, 2023-03-15:
for 1_I think it's better to use Yolo. I am working on a sample. Please try to send me a set of images so that I can prepare the model.

cfox570, 2023-05-19:
See Pull Request StopSign and Cone Detection #1131
I used TensorFlow Lite Object Detection with a custom cone model to implement traffic cone avoidance.  It was tested with Raspberry Pi 4.  I am also adding a pull request for the docs.