autorope/donkeycar #1101: Nvidia Jetson mismatch

Ahrovan, 2023-02-23:
- According to the installation doc for [Nvidia Jetson](https://docs.donkeycar.com/guide/robot_sbc/setup_jetson_nano/), will be installed tensorflow==2.3.1
- According to the installation doc for [Host PC ubuntu](https://docs.donkeycar.com/guide/host_pc/setup_ubuntu/), will be installed tensorflow==2.2.0
- After creating the model in Host and move it to Jetson, the following error is displayed:
```
  File "/home/ahrovan/env2/lib/python3.6/site-packages/tensorflow/python/framework/ops.py", line 6843, in raise_from_not_ok_status
    six.raise_from(core._status_to_exception(e.code, message), None)
  File "<string>", line 3, in raise_from
tensorflow.python.framework.errors_impl.InvalidArgumentError: Matrix size-incompatible: In[0]: [1,67712], In[1]: [6656,100] [Op:MatMul]
```
- But when the model is trained in Jetson. Runs without problems.

Why are different versions of Tensorflow incompatible? And if they do not match. Why these versions not corrected in the documents?

Ahrovan, 2023-02-23:
@DocGarbanzo Is this problem related to tensorflow version 2.3.1?
[tf_2_9](https://github.com/autorope/donkeycar/tree/tf_2_9) Can be installed and used?

DocGarbanzo, 2023-02-24:
@Ahrovan - can you check that you have the same image setting on your trained model as well as on your robot. I.e. `IMAGE_W, IMAGE_H` in your `myconfig.py`?

Ezward, 2023-02-27:
ubuntu.yml installs tensorflow 2.2.0, see [ubuntu.yml#L39](https://github.com/autorope/donkeycar/blob/de4d97be59e536154a98e260c24142b42aeb05f8/install/envs/ubuntu.yml#L39) but Jetson Nano installs 2.3.1, see docs [setup_jetson_nano.md#L66](https://github.com/autorope/donkeydocs/blob/dde1cf777b914007af025cd45d797e9b33f78079/docs/guide/robot_sbc/setup_jetson_nano.md?plain=1#L66)

In addition mac.yml installs tensorflow 2.2.0, see [mac.yml#L43](https://github.com/autorope/donkeycar/blob/de4d97be59e536154a98e260c24142b42aeb05f8/install/envs/mac.yml#L43)  

Windows docs also shows to install TF 2.2.0, see [setup_windows.md#L59](https://github.com/autorope/donkeydocs/blob/dde1cf777b914007af025cd45d797e9b33f78079/docs/guide/host_pc/setup_windows.md?plain=1#L59) and [setup_windows.md#L69](https://github.com/autorope/donkeydocs/blob/dde1cf777b914007af025cd45d797e9b33f78079/docs/guide/host_pc/setup_windows.md?plain=1#L69)

Ahrovan, 2023-02-27:
pip install tensorflow==2.3.1 in PC Host env/conda solved. just need test. working on it...

Ezward, 2023-03-06:
We have to do a little work here; I think the final fix will be when @DocGarbanzo lands the tensorflow 2.9 branch; then both raspberry pi and jetson nano will use the same version, so we can update the docs to that version.

Ezward, 2023-03-06:
@Ahrovan you have a workaround, is that correct?  I am re-opening; this is a real issue and we appreciate you bringing this up.  I will close when the tf 2.9 branch lands.

Ahrovan, 2023-03-16:
[image]
I have prepared this table. Please note that there may be minor mistakes.