autorope/donkeycar #1103: KeyError: 'user/angle'

Ahrovan, 2023-02-26:
Command : `donkey train --tub ./data --model ./models/mypilot.h5`

ERROR : `2023-02-26 22:35:30.444913: W tensorflow/core/framework/op_kernel.cc:1755] Unknown: KeyError: 'user/angle'`

How to solve:

- change manifest.json and catalog_x.catalog 
 all "user/steering" >> "user/angle"

Why occurred ? donkey car web controller saved all steering values as "user/angle" not "user/steering"
[image]

Ezward, 2023-02-26:
This is my error.  I just merged a change that renamed 'user/angle' to 'user/steering' as part of a larger change, forgetting that this would obsolete old data.  I will work on a fix immediately.

Ezward, 2023-02-26:
@Ahrovan I've made a change in this branch; https://github.com/autorope/donkeycar/tree/1104-fix-user-angle-in-deep-learning  It will now again save steering data with the 'user/angle' label ratehr than 'user/steering'  However this does not fix data that was saved with 'user/steering',but you say you have a work-around for data that was erroneously saved using 'user/steering':
- To fix data saved with 'user/steering', change manifest.json and catalog_x.catalog all "user/steering" >> "user/angle"

Please give the branch a test and let me know here is this fixes the data recording issue.  Thanks and sorry for the trouble.

Ezward, 2023-02-26:
See PR https://github.com/autorope/donkeycar/pull/1105 for a fix

Ezward, 2023-02-27:
That PR merged.  I've tested in by collecting data, training and driving on autopilot.

closed 2023-02-27 by Ezward