autorope/donkeycar #1104: train done -Always no result

Ahrovan, 2023-02-26:
using donkey v4.4.dev6 ...

I have not been able to train and run for several days. I doubted my data. Finally, I repeated the work of train with your data

- [sim_warehouse_manual.tar.gz](https://github.com/autorope/donkey_datasets/blob/master/large_dataset/sim_warehouse_manual.tar.gz)
- myconfig.py not changed just added three lines
```
DONKEY_GYM = True
DONKEY_SIM_PATH = "...../projects/DonkeySimWin/donkey_sim.exe"
DONKEY_GYM_ENV_NAME = "donkey-generated-track-v0"
```
- Pilot throttle and angle always is constant or wrong
- please check and provide a solution.

Ezward, 2023-02-26:
I've made a change to the deep learning template that caused this.  It is saving steering data as 'user/steering' rather than 'user/angle'.  I've made a fix in this branch so that future data will again be recording as 'user/angle' https://github.com/autorope/donkeycar/tree/1104-fix-user-angle-in-deep-learning  Please give that a try and let me know if it is recording data correctly.  I'll look into how to fix the data that has been recorded as 'user/steering'

Ezward, 2023-02-27:
@Ahrovan Given the 'user/steering'->'user/angle' bug that was just fixed, can you checkout the latest main and rewrite your mycar folder (or whatever folder you use for the donkeycar application).  So from the root of your donkeycar repo:
```
git checkout main
donkey createcar --overwrite --path=<path-to-your-mycar>
```
and give it another try.

Ezward, 2023-02-27:
PS: I gathered data, trained and ran autopilot using the new code.  Remember that the steering data is labelled as 'user/angle' so if you changed that to 'user/steering' then you should change it back.

Ahrovan, 2023-02-27:
[sim_warehouse_manual.tar.gz](https://github.com/autorope/donkey_datasets/blob/master/large_dataset/sim_warehouse_manual.tar.gz) extracted again. 

[image]

Ezward, 2023-02-27:
@Ahrovan Can you try training using the command line?  See https://docs.donkeycar.com/guide/deep_learning/train_autopilot/#train-a-model  That way we can see if it is DonkeyUI or the donkeycar train command itself.

PS: I trained a model yesterday from the command line on a Mac and it worked file.

Ezward, 2023-02-27:
@Ahrovan Also are you training a 'linear' model but running it as a 'categorical' model; that is what I see in the screen captures above.

Ahrovan, 2023-02-28:
@Ezward no, but is linear. Problem just related to recording. I know but need to same

Ahrovan, 2023-02-28:
@Ezward @DocGarbanzo 

Python 3.7.16 | donkey v4.4.dev6 | tensorflow '2.2.0' | windows | linear model | Created New Conda Env today

- train always finish without any Error.
`INFO:donkeycar.parts.interpreter:Convert model C:\Users\Hadi\workspace\mycar\models\mypilot.h5 to TFLite C:\Users\Hadi\workspace\mycar\models\mypilot.tfl
`
- tested with donkey ui and command line
- [sim_warehouse_manual.tar.gz](https://github.com/autorope/donkey_datasets/blob/master/large_dataset/sim_warehouse_manual.tar.gz) used
- [h5 file ] 9,661 KB after train completed

[image]

[image]

Ezward, 2023-02-28:
@Ahrovan that does look wrong.  Can you try running tubplot with your model and data to see what it outputs; https://docs.donkeycar.com/utility/donkey/#plot-predictions

Ezward, 2023-02-28:
Please also check your data again; I believe you renamed 'user/angle' to 'user/steering';  if you did you need to restore it to 'user/angle'.

Ahrovan, 2023-02-28:
[sim_warehouse_manual.tar.gz](https://github.com/autorope/donkey_datasets/blob/master/large_dataset/sim_warehouse_manual.tar.gz) extracted without any change.  
[image]
**'user/angle' exist.**

Ahrovan, 2023-03-01:
@Ezward The problem is related to train with gpu. If train operation is done with cpu, the problem will be solved.

- No error is reported during tarin with GPU. But the result has a fixed value.
- cuda is installed correctly.
- cudnn is installed correctly.
- Our GPU is from Nvidia series.
**- Important >> This problem occurred in Jetson Device Like NX, Nano** and PC, Windows and Ubuntu, WSL2

Ezward, 2023-03-01:
@Ahrovan can you try to train with an earlier version of donkeycar so we can see if it is a recently introduced bug?  Maybe checkout tag `4.3.6.2` or `4.4.0`

Ahrovan, 2023-03-02:
Report - donkey v4.4.dev6 worked on Jetson NX - Trained with GPU

Ezward, 2023-03-03:
to summarize
- train on cpu works
- train with v4.4.dev6 on Jetson NX GPU works
- train with TensorFlow version 2.3.1 | donkey v4.4.dev6 | windows | h5 model produces constant throttle/angle.  Assume training using GPU?
- train with Python 3.7.16 | donkey v4.4.dev6 | tensorflow '2.2.0' | windows | h5 model produces constant throttle/angle.  Assume training using GPU?

Is that correct?

On the failing system can you train using the command line and copy the console output here?

Ahrovan, 2023-03-03:
- donkey ui with GPU = wrong pilot play (Video first part)
- donkey ui without GPU = True result (Video second part)

This shows that the model has no problem. Only the ui display has a problem. I have tested this model on the robot. Robot runs without problems

[image]

Ezward, 2023-03-03:
Except that you also did a tubplot and it also showed constant throttle and steering.

Ahrovan, 2023-03-04:
[image]
[log.txt](https://github.com/autorope/donkeycar/files/10887869/log.txt)

Ezward, 2023-03-06:
I don't understand what is happening in the prior comment; can you add an explanation?

Ezward, 2023-03-11:
Maybe you are saying one of the modes is always getting that bad record NaN error?  And a white screen.  Is that the same thing that was going on before?  It looks different.

Ahrovan, 2023-03-12:
yes, white screen is related model trained with GPU always getting bad record NaN error. yes, i don't know

Ezward, 2023-03-13:
@DocGarbanzo have you seen this issue before in the DonkeyUI?  Do you know what this might be?

Ahrovan, 2023-03-16:
Problem solved. 
cudnn 8.1 CUDA 11.2 TensorFlow 2.5 (CUDA >= 11)
now I am working on compatibility issues on Host and Jetson platform.

Ezward, 2023-03-18:
Great. I am glad you have a solution.  Closing this issue.

closed 2023-03-18 by Ezward