autorope/donkeycar #1108: adafruit_python_pca9685 and adafruit_circuitpython_pca9685 conflict

quebeh, 2023-03-02:
Hey guys, i want to use the newer [adafruit_circuitpython_pca9685](https://pypi.org/project/adafruit-circuitpython-pca9685/) library for other utilities. But it seems like donkeycar is using the old, deprecated library [adafruit_python_pca9685](https://pypi.org/project/Adafruit-PCA9685/).

Is there any replacement for donkeycar.parts.actuator.PWMThrottle (ESC module) that uses the new library?
I have tried to download the old one's source code, but it seems like having both the modules having its own instances of PCA9685 objects will reduce the efficiency. 

Any help is appreciated
Thank you

Ezward, 2023-03-02:
We have no plans to change that library, but we are definitely open to accepting a PR.

The old [PCA9685](https://github.com/autorope/donkeycar/blob/08adc76ec184dd6a0dde1c8f53aa82aaa8b5204c/donkeycar/parts/actuator.py#L128) class is being deprecated in favor of [PulseController](https://github.com/autorope/donkeycar/blob/08adc76ec184dd6a0dde1c8f53aa82aaa8b5204c/donkeycar/parts/actuator.py#L85) which takes a [PwmPin](https://github.com/autorope/donkeycar/blob/08adc76ec184dd6a0dde1c8f53aa82aaa8b5204c/donkeycar/parts/pins.py#L156) which can source Pwm from GPIO or PCA9685.  That is what [PWMSteering](https://github.com/autorope/donkeycar/blob/08adc76ec184dd6a0dde1c8f53aa82aaa8b5204c/donkeycar/parts/actuator.py#L282) and [PWMThrottle](https://github.com/autorope/donkeycar/blob/08adc76ec184dd6a0dde1c8f53aa82aaa8b5204c/donkeycar/parts/actuator.py#L331) use to source the PWM signal.  So there is no need to touch those; you want to change the implementation of the PCA9685 pwm pin at [PwmPinPCA9685](https://github.com/autorope/donkeycar/blob/08adc76ec184dd6a0dde1c8f53aa82aaa8b5204c/donkeycar/parts/pins.py#L697). 

"both the modules having its own instances of PCA9685 objects will reduce the efficiency"

You are correct that the [PwmPinPCA9685](https://github.com/autorope/donkeycar/blob/08adc76ec184dd6a0dde1c8f53aa82aaa8b5204c/donkeycar/parts/pins.py#L697) implementation does use one PCA9685 instance per pin via the factory method [pca9685](https://github.com/autorope/donkeycar/blob/08adc76ec184dd6a0dde1c8f53aa82aaa8b5204c/donkeycar/parts/pins.py#L617).  I don't think this is an issue but you are free to change that implementation detail if you decide to open a PR.

quebeh, 2023-03-03:
Alright. Thank you very much. I'll get a better understanding of the pins.py code and possibly make a pull request.

closed 2023-03-03 by quebeh