autorope/donkeycar #1110: Adding Timms Models for transfer learning for FastAI(Pytorch)

adricl, 2023-03-04:
So Timms models is a large collection of pre-trained networks (https://github.com/rwightman/timm#models) that can be used for transfer learning. 

I have setup this up for Fast AI (Pytorch) transfer learning. I was wondering if you were interested in me pushing a PR for it. 

The models include 
EfficientNet (and family models)
MobileNet 

All small fast models that could run on Pi's 

Thanks.

Ezward, 2023-03-06:
I think we would be interested.  Can you add a comment to this issue that includes some general documentation as markdown that we could get into docs.donkeycar.com; or do a parallel PR in https://github.com/autorope/donkeydocs

adricl, 2023-03-06:
I can add some documentation around this as a PR as well as the PR for donkey car. It will just take me a few days.