autorope/donkeycar #1115: Allow a constant throttle in deep learning template

Ezward, 2023-03-23:
Both the path follow and the computer vision templates allow a constant throttle to be used when in autopilot mode.  That would be a useful feature in the deep learning template.  

- `USE_CONSTANT_THROTTLE` is the property used by the other templates to turn this on.  We can re-use that in the deep learning template

- The computer vision template keeps a series of values whereby the throttle can be varied based on the steering value; slow down when turning; speed up when going straight.  There are initial, minimum and maximum values that can be specified; so making min and max the same creates a constant throttle.  But it would also be interesting to keep the step control as in the computer vision template.  
```
THROTTLE_MAX = 0.3    # maximum throttle value the controller will produce
THROTTLE_MIN = 0.15   # minimum throttle value the controller will produce
THROTTLE_INITIAL = THROTTLE_MIN  # initial throttle value
THROTTLE_STEP = 0.05  # how much to change throttle when on/off line (adding/subtracting respectively)
```

Perhaps this is a capability we add to all templates; we can port this from the [LineFollower](https://github.com/autorope/donkeycar/blob/5e234c3101cc5f54935c240819e3840596c753a3/donkeycar/parts/line_follower.py#L90) part to a separate part (and so separate throttle control from steering control in the computer vision template.

Yann377, 2023-10-23:
Hi, I'm an IT student, I'm going to work on it as part of my computer science studies :-)

Ezward, 2024-01-13:
@Yann377 Great, this is a good project.  So how can I help?