autorope/donkeycar #1117: Add feature - connect to a new simulator

Ahrovan, 2023-03-25:
We want to connect to another simulator. This simulator has several simple methods: connect, setSteering, setSpeed, etc. Just please guide to add this feature in donkeycar code.

Maximellerbach, 2023-03-25:
Could you be more specific about which simulator?

Ahrovan, 2023-03-25:
https://github.com/AvisEngine/AVIS-Engine-Python-API

AVIS Engine Python API

Ezward, 2023-03-25:
@Maximellerbach I don't think we should do this.  I don't think we want to support a 3rd party api that will change over time as AWS changes their simulator.  We are open source, so @Ahrovan can write the necessary Donkey part it an keep it in a personal repo.

Maximellerbach, 2023-03-25:
@Ezward yep, that's my point of view too

Ezward, 2023-03-25:
Ok, I'm closing this as won't do.  @Ahrovan we can provide support for you in the donkey discord if you choose to do this work.

closed 2023-03-25 by Ezward