autorope/donkeycar #1122: Bad Jetson.GPIO version in TF29 branch

BrianHenryIE, 2023-04-17:
When I tried to calibrate my car, I got the error `The current user does not have permissions...`

`$ donkey calibrate --pwm-pin=PCA9685.1:40.1`

```
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
Traceback (most recent call last):
  File "/home/brian/mambaforge/envs/donkey/bin/donkey", line 33, in <module>
    sys.exit(load_entry_point('donkeycar', 'console_scripts', 'donkey')())
  File "/home/brian/projects/donkeycar/donkeycar/management/base.py", line 626, in execute_from_command_line
    c.run(args[2:])
  File "/home/brian/projects/donkeycar/donkeycar/management/base.py", line 219, in run
    from donkeycar.parts.actuator import PulseController
  File "/home/brian/projects/donkeycar/donkeycar/parts/actuator.py", line 19, in <module>
    import RPi.GPIO as GPIO
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that {} does not exist. Please check if that file is present.".format(_GPIOCHIP_ROOT))
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that /dev/gpiochip0 does not exist. Please check if that file is present.
```

I tried a few fixes, the [Jetson GPIO docs](https://github.com/NVIDIA/jetson-gpio/blob/master/README.md#setting-user-permissions) for setting user permissions:

```
sudo groupadd -f -r gpio
sudo usermod -a -G gpio $USER
sudo cp ~/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/99-gpio.rules /etc/udev/rules.d/
```

Which did not work.

Then tried `sudo chmod a+rw /dev/gpiochip0` ([as suggested](https://github.com/warthog618/gpiod/issues/10)) which changed the error to `TypeError: can only concatenate str (not "int") to str`

`$ donkey calibrate --pwm-pin=PCA9685.1:40.1`
```
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
Traceback (most recent call last):
  File "/home/brian/mambaforge/envs/donkey/bin/donkey", line 33, in <module>
    sys.exit(load_entry_point('donkeycar', 'console_scripts', 'donkey')())
  File "/home/brian/projects/donkeycar/donkeycar/management/base.py", line 626, in execute_from_command_line
    c.run(args[2:])
  File "/home/brian/projects/donkeycar/donkeycar/management/base.py", line 219, in run
    from donkeycar.parts.actuator import PulseController
  File "/home/brian/projects/donkeycar/donkeycar/parts/actuator.py", line 19, in <module>
    import RPi.GPIO as GPIO
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/gpio.py", line 69, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/gpio_pin_data.py", line 581, in get_data
    d = prefix + pwm_chip_name
TypeError: can only concatenate str (not "int") to str
```

I found the issue https://github.com/NVIDIA/jetson-gpio/issues/79 where the official response was:

> For Jetson Xavier and before you should be fine using version 2.0.21 for now.

The TF29 branch installs v2.1:

`$ conda list`
```
jetson-gpio               2.1.0                    pypi_0    pypi
```

I installed v2.0.21 using:

`pip install --force-reinstall -v "Jetson.GPIO==2.0.21"`

And the error went away.

TCIII, 2023-04-17:
@BrianHenryIE,

`Jetson.gpio` normally comes already installed on JP so this error that you experienced is very interesting.

TCIII

BrianHenryIE, 2023-04-17:
I see [v2.1.1 was released March 8](https://github.com/NVIDIA/jetson-gpio/releases/tag/v2.1.1) with the comment:

> Fix Jetson Nano ... implementation

I'll try install that when I get my car driving.

DocGarbanzo, 2023-04-21:
@BrianHenryIE - here is a solution from @Heavy02011: `sudo chmod 666 /dev/gpiochip*` who encountered the same issue - I'll add this to the docs after I figured out when exactly we have to run this.

BrianHenryIE, 2023-04-21:
I did a fresh install of the tf29 branch yesterday and just calibrated now. I can confirm `sudo chmod 666 /dev/gpiochip*` worked to fix the error from this thread.

This time `conda list` had `jetson-gpio               2.1.1                    pypi_0    pypi` (i.e. using 2.1.1 wasn't the solution as hoped).

TCIII, 2023-04-22:
@BrianHenryIE,

I don't encounter the permissions error when running `jetson-gpio` version 2.0.20, therefore getson-gpio version 2.1.1 is causing the error.

TCIII

TCIII, 2023-04-22:
@DocGarbanzo,

NVIDIA solution for jetson-gpio permissions issue:

```
If you installed Jetson.GPIO from a package, e.g. using pip into a virtual environment:

sudo cp venv/lib/pythonNN/site-packages/Jetson/GPIO/99-gpio.rules /etc/udev/rules.d/
For the new rule to take place, you either need to reboot or reload the udev rules by running:

sudo udevadm control --reload-rules && sudo udevadm trigger
```
TCIII

BrianHenryIE, 2023-04-22:
Our data from this thread is:

* 2.1.1 – error for me
* 2.1.0 – error for me
* 2.0.21 – no error for me 
* 2.0.20 – no error for you

After `sudo chmod 666 /dev/gpiochip* `:

* 2.1.1 – no error for me

I'm under the impression, from https://github.com/NVIDIA/jetson-gpio/issues/79 and https://github.com/NVIDIA/jetson-gpio/releases/tag/v2.1.1, that this will be fixed by Nvidia in a later update.

I'm pretty sure I tried `sudo udevadm control --reload-rules && sudo udevadm trigger` earlier without effect but forgot to copy that into my original post. As I said, I did run the `cp ... 99-gpio.rules` line.

TCIII, 2023-04-22:
@BrianHenryIE,

Did you run `sudo cp venv/lib/pythonNN/site-packages/Jetson/GPIO/99-gpio.rules /etc/udev/rules.d/` before running `sudo udevadm control --reload-rules && sudo udevadm trigger` per the NVIDIA solution above?

TCIII

BrianHenryIE, 2023-04-22:
Yes. And rebooted. I tested that just now on a fresh install to be sure:

```
(donkey) brian@brian-desktop:~$ donkey calibrate --pwm-pin=PCA9685.1:40.1
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
Traceback (most recent call last):
  File "/home/brian/mambaforge/envs/donkey/bin/donkey", line 33, in <module>
    sys.exit(load_entry_point('donkeycar', 'console_scripts', 'donkey')())
  File "/home/brian/projects/donkeycar/donkeycar/management/base.py", line 626, in execute_from_command_line
    c.run(args[2:])
  File "/home/brian/projects/donkeycar/donkeycar/management/base.py", line 219, in run
    from donkeycar.parts.actuator import PulseController
  File "/home/brian/projects/donkeycar/donkeycar/parts/actuator.py", line 19, in <module>
    import RPi.GPIO as GPIO
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that {} does not exist. Please check if that file is present.".format(_GPIOCHIP_ROOT))
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that /dev/gpiochip0 does not exist. Please check if that file is present.

(donkey) brian@brian-desktop:~$ sudo groupadd -f -r gpio
[sudo] password for brian: 
(donkey) brian@brian-desktop:~$ sudo usermod -a -G gpio brian
(donkey) brian@brian-desktop:~$ sudo cp ~/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/99-gpio.rules /etc/udev/rules.d/
(donkey) brian@brian-desktop:~$ sudo udevadm control --reload-rules && sudo udevadm trigger

(donkey) brian@brian-desktop:~$ donkey calibrate --pwm-pin=PCA9685.1:40.1
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
Traceback (most recent call last):
  File "/home/brian/mambaforge/envs/donkey/bin/donkey", line 33, in <module>
    sys.exit(load_entry_point('donkeycar', 'console_scripts', 'donkey')())
  File "/home/brian/projects/donkeycar/donkeycar/management/base.py", line 626, in execute_from_command_line
    c.run(args[2:])
  File "/home/brian/projects/donkeycar/donkeycar/management/base.py", line 219, in run
    from donkeycar.parts.actuator import PulseController
  File "/home/brian/projects/donkeycar/donkeycar/parts/actuator.py", line 19, in <module>
    import RPi.GPIO as GPIO
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/RPi/GPIO/__init__.py", line 1, in <module>
    from Jetson.GPIO import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that {} does not exist. Please check if that file is present.".format(_GPIOCHIP_ROOT))
RuntimeError: The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that /dev/gpiochip0 does not exist. Please check if that file is present.

(donkey) brian@brian-desktop:~$ sudo reboot now
Connection to 192.168.0.139 closed by remote host.
Connection to 192.168.0.139 closed.
(base) brianhenry@Brians-MacBook-Air-2 donkeycar % ssh brian@192.168.0.139
brian@192.168.0.139's password: 
Welcome to Ubuntu 18.04.6 LTS (GNU/Linux 4.9.299-tegra aarch64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
[... 61 lines cut ...]
    from Jetson.GPIO import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/Jetson/GPIO/gpio.py", line 33, in <module>
    raise RuntimeError("The current user does not have permissions set to access the library functionalites. Please configure permissions or use the root user to run this. It is also possible that {} does not exist. Please check if that file is present.".format(_G [... cut ...]
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
WARNING:donkeycar.parts.pins:pigpio was not imported.
init pin PCA9685.1:40.1
Using PWM freq: 60

Enter a PWM setting to test ('q' for quit) (0-1500): 

```

BrianHenryIE, 2023-04-23:
`sudo chmod 666 /dev/gpiochip*` does not persist across all reboots.

Ezward, 2023-05-20:
Marked as solved.  Discussion can continue if needed.

closed 2023-05-20 by Ezward