autorope/donkeycar #1123: TF29 branch, Unable to open CSICamera.

BrianHenryIE, 2023-04-17:
`$ python ~/mycar/manage.py drive`
```
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
INFO:donkeycar.config:loading config file: /home/brian/mycar/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
INFO:__main__:PID: 8161
WARNING:donkeycar.parts.pins:pigpio was not imported.
cfg.CAMERA_TYPE CSIC
INFO:__main__:cfg.CAMERA_TYPE CSIC
Traceback (most recent call last):
  File "/home/brian/mycar/manage.py", line 1142, in <module>
    drive(cfg, model_path=args['--model'], use_joystick=args['--js'],
  File "/home/brian/mycar/manage.py", line 104, in drive
    add_camera(V, cfg, camera_type)
  File "/home/brian/mycar/manage.py", line 880, in add_camera
    cam = get_camera(cfg)
  File "/home/brian/mycar/manage.py", line 808, in get_camera
    cam = CSICamera(image_w=cfg.IMAGE_W, image_h=cfg.IMAGE_H, image_d=cfg.IMAGE_DEPTH,
  File "/home/brian/projects/donkeycar/donkeycar/parts/camera.py", line 218, in __init__
    self.init_camera()
  File "/home/brian/projects/donkeycar/donkeycar/parts/camera.py", line 246, in init_camera
    raise RuntimeError("Unable to open CSICamera.")
RuntimeError: Unable to open CSICamera.
```

Jetson Nano on TF2.9 branch.

I was able to get past the error by adding `self.camera.open(0)` to `donkeycar/parts/camera.py:237`. I'm having another problem now so I'm not 100% sure this is fixed.

TCIII, 2023-04-17:
@BrianHenryIE,

Try running the following commands to ensure that GStreamer is installed on your Nano:
```
python
>>>import cv2
>>>print(cv2.getBuildInformation())   // Check for GStreamer installed version
```

TCIII

TCIII, 2023-04-17:
@BrianHenryIE,
Here is the CSCI camera output from running `manage.py drive`:

```
cfg.CAMERA_TYPE CSIC
INFO:__main__:cfg.CAMERA_TYPE CSIC
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0
   Camera mode  = 5
   Output Stream W = 1280 H = 720
   seconds to Run    = 0
   Frame Rate = 120.000005
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0@3.619] global /home/thomas/opencv/modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
INFO:donkeycar.parts.camera:CSICamera opened...
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:CSICamera ready.
INFO:donkeycar.vehicle:Adding part CSICamera.
```
I would hazard a guess that either the CSIC camera ribbon cable is not connected to your Nano correctly or that GStreamer is not installed.
TCIII

BrianHenryIE, 2023-04-18:
```
python
>>>import cv2
>>>print(cv2.getBuildInformation())
```

```
GStreamer:                   NO
```

Full output:

```
General configuration for OpenCV 4.7.0 =====================================
  Version control:               4.7.0

  Platform:
    Timestamp:                   2023-02-22T13:23:24Z
    Host:                        Linux 5.3.0-28-generic aarch64
    CMake:                       3.25.2
    CMake generator:             Unix Makefiles
    CMake build tool:            /bin/gmake
    Configuration:               Release

  CPU/HW features:
    Baseline:                    NEON FP16

  C/C++:
    Built as dynamic libs?:      NO
    C++ standard:                11
    C++ Compiler:                /opt/rh/devtoolset-10/root/usr/bin/c++  (ver 10.2.1)
    C++ flags (Release):         -Wl,-strip-all   -fsigned-char -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -Wl,-strip-all   -fsigned-char -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /opt/rh/devtoolset-10/root/usr/bin/cc
    C flags (Release):           -Wl,-strip-all   -fsigned-char -W -Wall -Wreturn-type -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -Wl,-strip-all   -fsigned-char -W -Wall -Wreturn-type -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -L/ffmpeg_build/lib  -Wl,--gc-sections -Wl,--as-needed -Wl,--no-undefined  
    Linker flags (Debug):        -L/ffmpeg_build/lib  -Wl,--gc-sections -Wl,--as-needed -Wl,--no-undefined  
    ccache:                      YES
    Precompiled headers:         NO
    Extra dependencies:          /lib64/libopenblas.so /usr/local/lib/libpng.so /lib64/libz.so dl m pthread rt
    3rdparty dependencies:       libprotobuf ade ittnotify libjpeg-turbo libwebp libtiff libopenjp2 IlmImf quirc tegra_hal

  OpenCV modules:
    To be built:                 calib3d core dnn features2d flann gapi highgui imgcodecs imgproc ml objdetect photo python3 stitching video videoio
    Disabled:                    world
    Disabled by dependency:      -
    Unavailable:                 java python2 ts
    Applications:                -
    Documentation:               NO
    Non-free algorithms:         NO

  GUI:                           NONE
    VTK support:                 NO

  Media I/O: 
    ZLib:                        /lib64/libz.so (ver 1.2.7)
    JPEG:                        libjpeg-turbo (ver 2.1.3-62)
    WEBP:                        build (ver encoder: 0x020f)
    PNG:                         /usr/local/lib/libpng.so (ver 1.6.37)
    TIFF:                        build (ver 42 - 4.2.0)
    JPEG 2000:                   build (ver 2.4.0)
    OpenEXR:                     build (ver 2.3.0)
    HDR:                         YES
    SUNRASTER:                   YES
    PXM:                         YES
    PFM:                         YES

  Video I/O:
    DC1394:                      NO
[... 42 lines cut ...]

I don't get the sensor modes as I start to drive the car: 

```
using donkey v5.0.dev1 ...
INFO:donkeycar.config:loading config file: /home/brian/mycar/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
INFO:__main__:PID: 8135
WARNING:donkeycar.parts.pins:pigpio was not imported.
cfg.CAMERA_TYPE CSIC
INFO:__main__:cfg.CAMERA_TYPE CSIC
INFO:donkeycar.parts.camera:CSICamera opened...
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:CSICamera ready.
INFO:donkeycar.vehicle:Adding part CSICamera.
INFO:donkeycar.parts.web_controller.web:Starting Donkey Server...
```

And the images that are recorded are just solid green.

[image]

And I tried the ribbon cable in both Jetson Nano connections.

I guess the best thing is to reinstall from scratch and ensure the CMake command is correct.

TCIII, 2023-04-18:
@BrianHenryIE,

From your Full Output:
You built opencv4.7 instead of opencv4.6, per the install instructions, and you are using Python 3.7 instead of Python 3.9: [tf_2_9 jetson46.yml](https://github.com/autorope/donkeycar/blob/tf_2_9/install/envs/jetson46.yml).

I have built opencv4.6 + GStreamer using the Q-Engineering manual install instructions and the accompanying modified CMake arguments four times for two Nano 4GBs and two Xavier NX 8GBs without issue once the CMake arguments were correctly determined.

Regards,
TCIII

BrianHenryIE, 2023-04-18:
Thanks. I think I know what happened. The traditional Donkey setup docs add a line to `~/.bashrc` to activate the virtual environment. The newer TF29 branch ones don't. I [already opened a PR](https://github.com/autorope/donkeydocs/pull/49/files) to address this:

TCIII, 2023-04-18:
@BrianHenryIE,

You must have missed this line, `source ~/.bashrc`, in the Nano JP4.6.1/2 [instructions](https://github.com/autorope/donkeydocs/blob/Update_to_tf_29/docs/guide/robot_sbc/setup_jetson_nano.md):
```
Step 3b: Setup Python Environment
The installation on JP 4.6.X is more tricky, because we need to use a newer version of tensorflow than what is packaged by NVidia for this OS version. Also, we need a compatible OpenCV version which needs to be built on the nano.

Step 1: Install mamba-forge
Download and install mambaforge. When you are asked to run conda init in the script, type yes.

wget https://github.com/conda-forge/miniforge/releases/latest/download/Mambaforge-Linux-aarch64.sh
chmod u+x ./Mambaforge-Linux-aarch64.sh
bash ./Mambaforge-Linux-aarch64.sh
source ~/.bashrc            // I believe that this is the line you are referring to?

TCIII

```

BrianHenryIE, 2023-04-22:
I reinstalled everything, confirmed GStreamer was present, then ran
`pip install -e .[nano] --user pip` because `donkey` wasn't working, and it seems to have disable GStreamer.
```
Python 3.9.16 | packaged by conda-forge | (main, Feb  1 2023, 22:05:40) 
[GCC 11.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cv2
>>> print(cv2.getBuildInformation()) 

General configuration for OpenCV 4.6.0 =====================================
  Version control:               unknown

  Extra modules:
    Location (extra):            /home/brian/opencv_contrib/modules
    Version control (extra):     unknown

  Platform:
    Timestamp:                   2023-04-20T04:50:47Z
    Host:                        Linux 4.9.299-tegra aarch64
    CMake:                       3.10.2
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               RELEASE

  CPU/HW features:
    Baseline:                    NEON FP16
      required:                  NEON

  C/C++:
    Built as dynamic libs?:      YES
    C++ standard:                11
    C++ Compiler:                /usr/bin/c++  (ver 7.5.0)
    C++ flags (Release):         -fsigned-char -ffast-math -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -ffast-math -W -Wall -Wreturn-type -Wnon-virtual-dtor -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -fopenmp -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -fsigned-char -ffast-math -W -Wall -Wreturn-type -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fopenmp -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -ffast-math -W -Wall -Wreturn-type -Waddress -Wsequence-point -Wformat -Wformat-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fopenmp -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,--gc-sections -Wl,--as-needed -Wl,--no-undefined  
    Linker flags (Debug):        -Wl,--gc-sections -Wl,--as-needed -Wl,--no-undefined  
    ccache:                      NO
    Precompiled headers:         NO
    Extra dependencies:          m pthread cudart_static dl rt nppc nppial nppicc nppicom nppidei nppif nppig nppim nppist nppisu nppitc npps cublas cudnn cufft -L/usr/local/cuda/lib64 -L/usr/lib/aarch64-linux-gnu
    3rdparty dependencies:

  OpenCV modules:
    To be built:                 alphamat aruco barcode bgsegm bioinspired calib3d ccalib core cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev datasets dnn dnn_objdetect dnn_superres dpm face features2d flann freetype fuzzy gapi hdf hfs highgui img_hash imgcodecs imgproc intensity_transform line_descriptor mcc ml objdetect optflow phase_unwrapping photo plot python3 quality rapid reg rgbd saliency sfm shape stereo stitching structured_light superres surface_matching text tracking ts video videoio videostab wechat_qrcode xfeatures2d ximgproc xobjdetect xphoto
    Disabled:                    python2 world
    Disabled by dependency:      -
    Unavailable:                 cvv java julia matlab ovis viz
    Applications:                perf_tests apps
    Documentation:               NO
    Non-free algorithms:         YES

  GUI:                           GTK3
    GTK+:                        YES (ver 3.22.30)
      GThread :                  YES (ver 2.56.4)
      GtkGlExt:                  NO
    VTK support:                 NO

  Media I/O: 
    ZLib:                        /usr/lib/aarch64-linux-gnu/libz.so (ver 1.2.11)
    JPEG:                        /usr/lib/aarch64-linux-gnu/libjpeg.so (ver 80)
    WEBP:                        /home/brian/mambaforge/envs/donkey/lib/libwebp.so (ver encoder: 0x020f)
    PNG:                         /usr/lib/aarch64-linux-gnu/libpng.so (ver 1.6.34)
    TIFF:                        build (ver 42 - 4.2.0)
    JPEG 2000:                   build (ver 2.4.0)
    OpenEXR:                     build (ver 2.3.0)
    HDR:                         YES
    SUNRASTER:                   YES
[... cut ...]
    numpy:                       /home/ci/.local/lib/python3.7/site-packages/numpy/core/include (ver 1.19.3)
    install path:                python/cv2/python-3

  Python (for build):            /bin/python2.7

  Java:                          
    ant:                         NO
    JNI:                         NO
    Java wrappers:               NO
    Java tests:                  NO

  Install to:                    /io/_skbuild/linux-aarch64-3.7/cmake-install
-----------------------------------------------------------------
```

TCIII, 2023-04-22:
@BrianHenryIE,

For `donkey calibrate --pwm-pin=PCA9685.1:40.1` to run, you need to be in the `mycar` folder.

TCIII

BrianHenryIE, 2023-04-22:
After a fresh install, the camera is working and recording on throttle! Thanks for the help.

The install order is important. If `pip install -e .[nano]` is run after opencv 4.6 is built, it will remove it as it installs opencv-python-headless-4.7.0.72 as a dependency of albumentations.

Somehow I managed to mess up the install order again yesterday! So I commented out albumentations in setup.py, then ran `pip install albumentations --dry-run` to learn what its dependencies were: 
```
Would install albumentations-1.3.0 opencv-python-headless-4.7.0.72 qudida-0.0.4
```
and ran `pip install qudida==0.0.4 --no-deps` and `pip install albumentations --no-deps`. I'm not sure what, if any, problems this might cause. I'm curious if the install steps are followed correctly, is `opencv-python-headless-4.7.0.72` installed, and opencv 4.6? I notice opencv 4.6 is NOT in my `conda list` but shows correctly in `cv2.getBuildInformation()`.

I also commented out `make clean` from the CMake build script. But I don't know how to reinstall opencv from that build if I did need to.

BrianHenryIE, 2023-04-22:
I was able to drive but got an error when trying to run autopilot:

```
TensorFlow was not built with CUDA kernel binaries compatible with compute capability 5.3. CUDA kernels will be jit-compiled from PTX, which could take 30 minutes or longer.
```

With both a .tflite and a .h5 model.

I ran `pip install ../jetson/tensorflow-2.9.3-cp39-cp39-linux_aarch64.whl --force-reinstall` and in the output was:

```
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
albumentations 1.3.0 requires opencv-python-headless>=4.1.1, which is not installed.
qudida 0.0.4 requires opencv-python-headless>=4.0.1, which is not installed.
spacy 3.3.1 requires pydantic!=1.8,!=1.8.1,<1.9.0,>=1.7.4, but you have pydantic 1.10.7 which is incompatible.
thinc 8.0.15 requires pydantic!=1.8,!=1.8.1,<1.9.0,>=1.7.4, but you have pydantic 1.10.7 which is incompatible.
```

I'm reinstalling from scratch again now.

TCIII, 2023-04-22:
@BrianHenryIE,

Possible solution:

From the Albumentations website:

Note on OpenCV dependencies¶
By default, pip downloads a wheel distribution of Albumentations. This distribution has opencv-python-headless as its dependency.

If you already have some OpenCV distribution (such as opencv-python-headless, opencv-python, opencv-contrib-python or opencv-contrib-python-headless) installed in your Python environment, you can force Albumentations to use it by providing the --no-binary qudida,albumentations argument to pip, e.g.

`pip install -U albumentations --no-binary qudida,albumentations`

TCIII

Ezward, 2023-05-13:
@DocGarbanzo When you update the nano install to address the albumentations installation issue can you close this issue?

Ezward, 2023-05-20:
Marked as solved.

closed 2023-05-20 by Ezward