autorope/donkeycar #1124: [tf_2_9] TensorFlow was not built with CUDA kernel binaries compatible with compute capability 5.3.

BrianHenryIE, 2023-04-22:
When I install the tf_2_9 branch on Jetson Nano 4GB B01, capture some training data, and train it on MacOS M1, when I try to run the model on the Nano, I get the error:

> TensorFlow was not built with CUDA kernel binaries compatible with compute capability 5.3. CUDA kernels will be jit-compiled from PTX, which could take 30 minutes or longer.

```
 python manage.py drive --model=models/mypiolt.h5 --type=linear
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
INFO:donkeycar.config:loading config file: /home/brian/mycar/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
INFO:__main__:PID: 7669
WARNING:donkeycar.parts.pins:pigpio was not imported.
cfg.CAMERA_TYPE CSIC
INFO:__main__:cfg.CAMERA_TYPE CSIC
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0@2.141] global /home/brian/opencv/modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
INFO:donkeycar.parts.camera:CSICamera opened...
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:CSICamera ready.
INFO:donkeycar.vehicle:Adding part CSICamera.
INFO:donkeycar.parts.web_controller.web:Starting Donkey Server...
INFO:donkeycar.parts.web_controller.web:You can now go to brian-desktop.local:8887 to drive your car.
INFO:donkeycar.vehicle:Adding part LocalWebController.
INFO:donkeycar.vehicle:Adding part PS4JoystickController.
INFO:donkeycar.vehicle:Adding part Pipe.
INFO:donkeycar.vehicle:Adding part ExplodeDict.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part ThrottleFilter.
INFO:donkeycar.vehicle:Adding part UserPilotCondition.
INFO:donkeycar.vehicle:Adding part RecordTracker.
/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/numpy/core/getlimits.py:500: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
  setattr(self, word, getattr(machar, word).flat[0])
/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
  return self._float_to_str(self.smallest_subnormal)
/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/numpy/core/getlimits.py:500: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
  setattr(self, word, getattr(machar, word).flat[0])
/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
  return self._float_to_str(self.smallest_subnormal)
INFO:numexpr.utils:NumExpr defaulting to 4 threads.
INFO:donkeycar.utils:get_model_by_type: model type is: linear
2023-04-22 16:09:30.601102: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-04-22 16:09:30.936621: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-04-22 16:09:30.937182: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-04-22 16:09:30.937303: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1943] TensorFlow was not built with CUDA kernel binaries compatible with compute capability 5.3. CUDA kernels will be jit-compiled from PTX, which could take 30 minutes or longer.
2023-04-22 16:09:30.939402: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-04-22 16:09:30.939821: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-04-22 16:09:30.940259: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-04-22 16:09:30.940427: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1943] TensorFlow was not built with CUDA kernel binaries compatible with compute capability 5.3. CUDA kernels will be jit-compiled from PTX, which could take 30 minutes or longer.
```

Training was done with:
```
donkey train --model=/Users/brianhenry/Sites/donkeycar-2023-04/mycar/models/mypilot.h5 --type=linear --tub=/Users/brianhenry/Sites/donkeycar-2023-04/mycar/data
```

I'm going to try train on device now to see is it the fault of MacOS M1.

BrianHenryIE, 2023-04-23:
Letting the JIT compile run does work, but takes 20 minutes and runs every time.

I tried explicitly _not_ disabling the cache, but it didn't work to avoid the recompile `CUDA_CACHE_DISABLE=0 CUDA_DISABLE_PTX_JIT=0 `

`$ python manage.py drive --model=models/mypilot.h5 --type=linear`
```
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
INFO:donkeycar.config:loading config file: /home/brian/mycar/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
INFO:__main__:PID: 8467
WARNING:donkeycar.parts.pins:pigpio was not imported.
cfg.CAMERA_TYPE CSIC
INFO:__main__:cfg.CAMERA_TYPE CSIC
GST_ARGUS: Creating output stream
CONSUMER: Waiting until producer is connected...
GST_ARGUS: Available Sensor modes :
GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

GST_ARGUS: Running with following settings:
   Camera index = 0 
   Camera mode  = 5 
   Output Stream W = 1280 H = 720 
   seconds to Run    = 0 
   Frame Rate = 120.000005 
GST_ARGUS: Setup Complete, Starting captures for 0 seconds
GST_ARGUS: Starting repeat capture requests.
CONSUMER: Producer has connected; continuing.
[ WARN:0@3.406] global /home/brian/opencv/modules/videoio/src/cap_gstreamer.cpp (1405) open OpenCV | GStreamer warning: Cannot query video position: status=0, value=-1, duration=-1
INFO:donkeycar.parts.camera:CSICamera opened...
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:CSICamera ready.
INFO:donkeycar.vehicle:Adding part CSICamera.
INFO:donkeycar.parts.web_controller.web:Starting Donkey Server...
INFO:donkeycar.parts.web_controller.web:You can now go to brian-desktop.local:8887 to drive your car.
INFO:donkeycar.vehicle:Adding part LocalWebController.
INFO:donkeycar.vehicle:Adding part PS4JoystickController.
INFO:donkeycar.vehicle:Adding part Pipe.
INFO:donkeycar.vehicle:Adding part ExplodeDict.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part ThrottleFilter.
INFO:donkeycar.vehicle:Adding part UserPilotCondition.
INFO:donkeycar.vehicle:Adding part RecordTracker.
/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/numpy/core/getlimits.py:500: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
  setattr(self, word, getattr(machar, word).flat[0])
/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
  return self._float_to_str(self.smallest_subnormal)
/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/numpy/core/getlimits.py:500: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
  setattr(self, word, getattr(machar, word).flat[0])
/home/brian/mambaforge/envs/donkey/lib/python3.9/site-packages/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
[... 98 lines cut ...]
device_identifier: "NVIDIA Tegra X1 sm_5.3 with 4156514304B RAM and 1 cores"
version: 1

2023-04-23 15:01:20.784663: W tensorflow/core/kernels/conv_ops_gpu.cc:336] None of the algorithms provided by cuDNN frontend heuristics worked; trying fallback algorithms.  Conv: batch: 1
in_depths: 64
out_depths: 64
in: 23
in: 23
data_format: 1
filter: 3
filter: 3
filter: 64
dilation: 1
dilation: 1
stride: 1
stride: 1
padding: 0
padding: 0
dtype: DT_FLOAT
group_count: 1
device_identifier: "NVIDIA Tegra X1 sm_5.3 with 4156514304B RAM and 1 cores"
version: 1
```

But the model did work to drive the car!

DocGarbanzo, 2023-06-05:
@BrianHenryIE - Newer versions of tensorflow are not build with TensorRT support any longer. The idea for using TRT in the tf 2.9 branch is to train the model in standard format (which is `.savedmodel` btw and not `.h5`) and then load the model on the jetson with `... --model models/mymodel.savedmodel --type tensorrt_linear` for example. It should then create the tensorrt engines at runtime and give you the fasted version of tensorrt on the target architecture. I believe that only this procedure will give you all of the performance of tensorrt on the jetson platform. If the jit compilation is too slow, I suggest you file an issue w/ NVidia support. I agree that a 20min waiting time is unacceptable, given this is a very small model in the end. 

Please note, in training you can omit the `--model ...` as a name will be auto-generated for you (and it will also create a `.savedmodel` format by default).

Heavy02011, 2023-06-06:
trying to reproduce with the following set of parameters:
```
DEFAULT_MODEL_TYPE = 'linear'
#DEFAULT_MODEL_TYPE = 'tensorrt_linear'
BATCH_SIZE = 8#16 # 8 #128                #how many records to use when doing one pass of gradient decent. Use a smaller number if your gpu is running out of memory.
# TRAIN_TEST_SPLIT = 0.8          #what percent of records to use for training. the remaining used for validation.
# MAX_EPOCHS = 100                #how many times to visit all records of your data
# SHOW_PLOT = True                #would you like to see a pop up display of final loss?
# VERBOSE_TRAIN = True            #would you like to see a progress bar with text during training?
# USE_EARLY_STOP = True           #would you like to stop the training if we see it's not improving fit?
# EARLY_STOP_PATIENCE = 5         #how many epochs to wait before no improvement
# MIN_DELTA = .0005               #early stop will want this much loss change before calling it improved.
# PRINT_MODEL_SUMMARY = True      #print layers and weights to stdout
# OPTIMIZER = None                #adam, sgd, rmsprop, etc.. None accepts default
# LEARNING_RATE = 0.001           #only used when OPTIMIZER specified
# LEARNING_RATE_DECAY = 0.0       #only used when OPTIMIZER specified
# SEND_BEST_MODEL_TO_PI = False   #change to true to automatically send best model during training
# CREATE_TF_LITE = True           # automatically create tflite model in training
CREATE_TENSOR_RT = True #False        # automatically create tensorrt model in training
# SAVE_MODEL_AS_H5 = False        # if old keras format should be used instead of savedmodel
CACHE_IMAGES = False             # if images are cached in training for speed up
```
getting output/error:
```
(donkey) rainer@donkeynano11:~/data/d2$ python train.py --tub=data/tub_7_23-06-05/ --model=models/trt1
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
INFO:numexpr.utils:NumExpr defaulting to 4 threads.
INFO:donkeycar.config:loading config file: /home/rainer/data/d2/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
WARNING:donkeycar.pipeline.database:No model database found at /home/rainer/data/d2/models/database.json
INFO:donkeycar.utils:get_model_by_type: model type is: linear
2023-06-06 20:31:02.974359: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:31:03.292123: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:31:03.292877: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:31:03.292962: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1943] TensorFlow was not built with CUDA kernel binaries compatible with compute capability 5.3. CUDA kernels will be jit-compiled from PTX, which could take 30 minutes or longer.
2023-06-06 20:31:03.294917: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:31:03.295515: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:31:03.295998: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:31:03.296096: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1943] TensorFlow was not built with CUDA kernel binaries compatible with compute capability 5.3. CUDA kernels will be jit-compiled from PTX, which could take 30 minutes or longer.
2023-06-06 20:48:53.593966: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:48:53.595140: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:48:53.595306: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1616] Could not identify NUMA node of platform GPU id 0, defaulting to 0.  Your kernel may not have been built with NUMA support.
2023-06-06 20:48:53.610903: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
[... cut ...]
    File "/home/rainer/mambaforge/envs/donkey/lib/python3.9/site-packages/keras/optimizers/optimizer_v2/optimizer_v2.py", line 946, in add_slot
        weight = tf.Variable(
    File "/home/rainer/mambaforge/envs/donkey/lib/python3.9/site-packages/keras/initializers/initializers_v2.py", line 152, in __call__
        return tf.zeros(shape, dtype)

    ResourceExhaustedError: OOM when allocating tensor with shape[5,5,3,24] and type float on /job:localhost/replica:0/task:0/device:GPU:0 by allocator GPU_0_bfc [Op:Fill]
```

Heavy02011, 2023-06-06:
After setting BATCH_SIZE = 1

```
(donkey) rainer@donkeynano11:~/data/d2$ python train.py --tub=data/tub_7_23-06-05/ --model=models/trt1
________             ______                   _________              
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /    
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/     
                                 /____/                              

using donkey v5.0.dev1 ...
INFO:numexpr.utils:NumExpr defaulting to 4 threads.
INFO:donkeycar.config:loading config file: /home/rainer/data/d2/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
WARNING:donkeycar.pipeline.database:No model database found at /home/rainer/data/d2/models/database.json
INFO:donkeycar.utils:get_model_by_type: model type is: linear
2023-06-06 20:54:16.979935: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:54:17.307749: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:54:17.308414: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:54:17.308521: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1943] TensorFlow was not built with CUDA kernel binaries compatible with compute capability 5.3. CUDA kernels will be jit-compiled from PTX, which could take 30 minutes or longer.
2023-06-06 20:54:17.310622: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:54:17.311170: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:54:17.311605: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 20:54:17.311684: W tensorflow/core/common_runtime/gpu/gpu_device.cc:1943] TensorFlow was not built with CUDA kernel binaries compatible with compute capability 5.3. CUDA kernels will be jit-compiled from PTX, which could take 30 minutes or longer.
2023-06-06 21:11:25.993902: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 21:11:25.994608: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 21:11:25.994845: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1616] Could not identify NUMA node of platform GPU id 0, defaulting to 0.  Your kernel may not have been built with NUMA support.
2023-06-06 21:11:25.995356: I tensorflow/stream_executor/cuda/cuda_gpu_executor.cc:961] could not open file to read NUMA node: /sys/bus/pci/devices/0000:00:00.0/numa_node
Your kernel may have been built without NUMA support.
2023-06-06 21:11:25.995610: I tensorflow/core/common_runtime/gpu/gpu_device.cc:1532] Created device /job:localhost/replica:0/task:0/device:GPU:0 with 177 MB memory:  -> device: 0, name: NVIDIA Tegra X1, pci bus id: 0000:00:00.0, compute capability: 5.3
INFO:donkeycar.parts.keras:Created KerasLinear with interpreter: KerasInterpreter
Model: "linear"
__________________________________________________________________________________________________
 Layer (type)                   Output Shape         Param #     Connected to                     
==================================================================================================
 img_in (InputLayer)            [(None, 120, 160, 3  0           []                               
                                )]                                                                
                                                                                                  
 conv2d_1 (Conv2D)              (None, 58, 78, 24)   1824        ['img_in[0][0]']                 
                                                                                                  
 dropout (Dropout)              (None, 58, 78, 24)   0           ['conv2d_1[0][0]']               
                                                                                                  
 conv2d_2 (Conv2D)              (None, 27, 37, 32)   19232       ['dropout[0][0]']                
                                                                                                  
 dropout_1 (Dropout)            (None, 27, 37, 32)   0           ['conv2d_2[0][0]']               
                                                                                                  
[... cut ...]
INFO:donkeycar.parts.interpreter:Converting SavedModel /home/rainer/data/d2/models/trt1.savedmodel to TensorRT /home/rainer/data/d2/models/trt1.trt
INFO:tensorflow:Linked TensorRT version: (8, 2, 1)
INFO:tensorflow:Loaded TensorRT version: (8, 2, 1)
ERROR:donkeycar.parts.interpreter:TensorRT conversion failed because: SavedModel file does not exist at: /home/rainer/data/d2/models/trt1.savedmodel/{saved_model.pbtxt|saved_model.pb}
INFO:donkeycar.pipeline.database:Writing database file: /home/rainer/data/d2/models/database.json
```

Heavy02011, 2023-06-08:
my conclusion so far: we need to rebuild tensorflow with „compute_53“ capability (should be asked during building process) as I failed using caching as describes by [Brian](https://github.com/BrianHenryIE).