autorope/donkeycar #1136: Remove albumentations

Ezward, 2023-07-02:
Albumentations still has the problem on jetson nano that it causes a crash because underneath it still uses a dependency that installs opencv-headless.  

The reality is that we only use two augmentations from it; a brightness augmentation and a guassian blur.  These are both easily implemented with opencv.

Replace the brightness augmentation and the guassian blur augmentation in https://github.com/autorope/donkeycar/blob/1398eec8309499c4d54fc45d510348bcb819daff/donkeycar/pipeline/augmentations.py#L13. with ones based on opencv so we can remove albumentations.

DocGarbanzo, 2023-07-05:
Yes, in the current setup that would work. But there are other augmentations, like CoarseDropout for example, which might be quite useful (one can play with them here: https://demo.albumentations.ai/). However, they might be easy to generate with `cv` too.

mdlopezme, 2024-02-23:
Hello I am trying to train in a Nvidia Jetson AGX Xavier, 
I am having problems with albumentations.
When I pip install albumentations, it seems to uninstall my current version of opencv-python. ( The problem is that I have to use a custom version of this library to be able to do gpu accelation in my platform).
I get this error after installing albumentations: pip install albumentations
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/usr/local/lib/python3.8/dist-packages/cv2/__init__.py", line 175, in bootstrap
    if __load_extra_py_code_for_module("cv2", submodule, DEBUG):
  File "/usr/local/lib/python3.8/dist-packages/cv2/__init__.py", line 28, in __load_extra_py_code_for_module
    py_module = importlib.import_module(module_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/local/lib/python3.8/dist-packages/cv2/mat_wrapper/__init__.py", line 39, in <module>
    cv._registerMatType(Mat)
AttributeError: partially initialized module 'cv2' has no attribute '_registerMatType' (most likely due to a circular import)
```

Is there a reason why albumentations is needed? I need guidance to know I can to try to use opencv version.
Would you like me to send a pull request if does work with opencv implementation?

mdlopezme, 2024-02-23:
Little update, I was able to train. After committing the above update on my Jetson. 
Will test on real data soon and send a little clip within the next few days.

mdlopezme, 2024-02-24:
https://youtu.be/rgLV-9JGVLM?si=X8n_8SM7aP7GspVT

@Ezward 
@DocGarbanzo