autorope/donkeycar #1141: Car not turning on when running python manage.py drive

sherifayantayo, 2023-08-11:
[image]

(myenv) sherif@raspberrypi:~/mycar $ python manage.py drive
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v5.0.dev3 ...
INFO:donkeycar.config:loading config file: /home/sherif/mycar/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
INFO:__main__:PID: 3734
WARNING:donkeycar.parts.pins:pigpio was not imported.
cfg.CAMERA_TYPE PICAM
INFO:__main__:cfg.CAMERA_TYPE PICAM
Traceback (most recent call last):
  File "/home/sherif/mycar/manage.py", line 1142, in <module>
    drive(cfg, model_path=args['--model'], use_joystick=args['--js'],
  File "/home/sherif/mycar/manage.py", line 104, in drive
    add_camera(V, cfg, camera_type)
  File "/home/sherif/mycar/manage.py", line 880, in add_camera
    cam = get_camera(cfg)
  File "/home/sherif/mycar/manage.py", line 798, in get_camera
    cam = PiCamera(image_w=cfg.IMAGE_W, image_h=cfg.IMAGE_H, image_d=cfg.IMAGE_DEPTH,
  File "/home/sherif/donkeycar/projects/donkeycar/donkeycar/parts/camera.py", line 22, in __init__
    from picamera.array import PiRGBArray
  File "/home/sherif/donkeycar/projects/donkeycar/myenv/lib/python3.9/site-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/home/sherif/donkeycar/projects/donkeycar/myenv/lib/python3.9/site-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/home/sherif/donkeycar/projects/donkeycar/myenv/lib/python3.9/site-packages/picamera/mmal.py", line 49, in <module>
    _lib = ct.CDLL('libmmal.so')
  File "/usr/lib/python3.9/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmmal.so: cannot open shared object file: No such file or directory

zlite, 2023-08-11:
Did you do a "pip install pigpio" first, as instructed?

sherifayantayo, 2023-08-11:
> pip install pigpio

I didnt see this in the instructions, I did do it now and ran it but i got the same error

zlite, 2023-08-11:
That missing file is part of the standard RPi camera library. Are you using
the latest RaspberryPi OS (Bullseye 64 bit)?

This ChatGPT answer can help:
https://chat.openai.com/share/09971d0d-fb1d-4d28-befe-fcdd760dc622

sherifayantayo, 2023-08-11:
Yes , it is the latest RaspberryPi OS (Bullseye 64 bit, i followed the instructions
[image]

sherifayantayo, 2023-08-11:
But i got the same error

zlite, 2023-08-11:
Did you enable the camera (don't select legacy camera) in raspi-config?

sherifayantayo, 2023-08-11:
[image]
I didnt find any other camera apart from legacy camera

zlite, 2023-08-11:
Try disabling legacy camera

zlite, 2023-08-11:
Failing that, do sudo apt-get install libraspberrypi-dev

sherifayantayo, 2023-08-11:
[image]
It didnt work

zlite, 2023-08-11:
And do you have an actual Pi camera attached?

sherifayantayo, 2023-08-11:
Yes i have one installed. The car wheels work with the calibration steering left to right and back to forth but for whatever reason when I run the python manage.py command as you saw I got these issues

zlite, 2023-08-11:
Might be something wrong with your camera. Try the usual RPi camera tests

TCIII, 2023-08-11:
@zlite 
I think that the problem is that `picamera` does not work on the Bullseye 64-bit OS and `manage.py` is trying to use `picamera` instead of `picamera2` for some reason.
There is no need to activate a camera on the BullsEye OS as it is done automatically (picamera2) during the Bullseye OS install unless a legacy camera is in use and is selected in the raspi-config app.

TCIII

sherifayantayo, 2023-08-11:
I ran this . it seems the camera is working. 
(myenv) sherif@raspberrypi:~/mycar $ vcgencmd get_camera
supported=1 detected=1, libcamera interfaces=0

TCIII, 2023-08-11:
@sherifayantayo,

As I posted above, `manage.py` is trying to use `picamera` and not `picamera2` for some reason.

You might have to uninstall DC, disable the legacy camera, reboot and reinstall DC so that python uses `picamera2` and not `picamera`.

TCIII

sherifayantayo, 2023-08-11:
Yes i installed with these instructions

> Did you install DC using these instructions: Installation for Donkey Car main (>= 5.0) using Raspberry Pi OS Bullseye
> 
> TCIII

TCIII, 2023-08-11:
@sherifayantayo,

Did you select enable the legacy camera in the raspi-config app?

sherifayantayo, 2023-08-11:
I certainly did after , im not sure if it was before i installed DC

sherifayantayo, 2023-08-11:
> @sherifayantayo,
> 
> Did you select `enable the legacy camera` in the raspi-config app before installing DC?

These werent in the instructions for Installation for Donkey Car main (>= 5.0) using Raspberry Pi OS Bullseye

sherifayantayo, 2023-08-11:
> > @sherifayantayo,
> > Did you select `enable the legacy camera` in the raspi-config app before installing DC?
> 
> These werent in the instructions for Installation for Donkey Car main (>= 5.0) using Raspberry Pi OS Bullseye

So I didn’t enable legacy camera before i installed donkey car . 
since you just recently installed donkey car , did you have issues with the adafruit pca9685 module when you ran donkey calibrate ?

TCIII, 2023-08-11:
@TCIII,

No, as I am using an RC HAT.

However, I have not had any issues using the RPi with the PCA9685 module.

Suggestion: I think that you need to take this issue to the [DC software forum](https://discord.com/channels/662098530411741184/671604367706554388) which will give you much more exposure and a faster solution for your problem.

TCIII

DocGarbanzo, 2023-08-11:
@sherifayantayo - there should be no `picamera` in your venv's `site-packages` path, as @TCIII and @zlite already mentioned. We are now using `picamera2`. For me, `picamera2` is already part of the system python install (living in `/usr/lib/python3/site-packages`).  So even installing it with `pip` in our virtual env, doesn't do anything for me. What does `ls /usr/lib/python3/dist-packages/picamera2` return for you? If it doesn't return anything then I recommend you do:
```bash
sudo apt-get update
sudo apt-get upgrade
```
See if this does the `picamera2` install in the system python. If not, you can install it manually by
```bash
sudo apt install -y python3-picamera2
```
Remove your current virtual env (just delete it) and re-create it, following the donkey docs. Even if you don't have `pycamera2` in your system python, the donkey install would install it there as it is part of the dependencies. Make sure you don't enable the legacy cam in `raspi-config`.  So long story short, `picamera2` should be either in your system python or your venv but `picamera` should not be in any of those. This will likely get the camera to work.

Ezward, 2023-08-12:
@sherifayantayo I have a hunch that legacy camera was chosen at some point and now you have picamera instead of pycamera2.  I suggest you reinstall from scratch, again using the 64bit OS, but make sure NOT to enable legacy camera at any point.  After installing the OS just go on with the install as documented and everything should work.

zlite, 2023-08-12:
Also make double sure you're using the newer 64-bit Bullseye OS and not the
old Buster