autorope/donkeycar #1142: Release testing for Donkeycar 5.0.dev3

Ezward, 2023-08-12:
Below is a test matrix 
>> When @DocGarbanzo is ready then we can fill in the links to the proposed documentation changes in the sections below.  We will do the install based on those instructions.

## [Install Software on Host PC]

On each of the platforms listed below we want to
- Install Donkeycar
- Train a deep learning autopilot.  This can be done by using your own data or downloading Circuit Launch data from [donkey_datasets](https://github.com/autorope/donkey_datasets/tree/master/circuit_launch_20210716).  We want to train using the command line tool [donkey train](http://docs.donkeycar.com/utility/donkey/#train-the-model) and DonkeyUI [trainer](http://docs.donkeycar.com/utility/ui/#the-trainer)
- Test inference by running [tubplot](http://docs.donkeycar.com/utility/donkey/#plot-predictions) using the data and the model. 

| OS           | Install | donkey train | DonkeyUI train | tubplot |
| :----------- | :-----: | :----------: | :------------: | :-----: |
| Intel Linux  | todo    | todo         | todo           | todo    |
| Nvidia Linux | todo    | todo         | todo           | todo    |
| Windows WSL  | todo    | todo         | todo           | todo    |
| Intel Mac    | todo    | todo         | todo           | todo    |
| Arm Mac      | todo    | todo         | todo           | todo    |

>> ?Are we officially supporting Nvidia as a host PC?)

>> NOTE: we are not supporting Windows native installation, just WSL.  Google has dropped support for Tensorflow on native Windows.

## [Install Software on Donkeycar](http://docs.donkeycar.com/guide/install_software/#step-2-install-software-on-donkeycar)

- [ ] [Install Donkeycar on Raspberry Pi]
- [ ] [Install Donkeycar on Nvidia SBC (> Nano)]
>> NOTE: we are not supporting the Jetson Nano for versions 5.x.  The Nano is end-of-life and Nvidia will not be updating the Jetpack OS for it.

Heavy02011, 2023-08-28:
I can confirm all tests passed for the *simulator* using DC 5.0dev3 running on Intel Linux (Ubuntu 22.04) using the latest gym-donkeycar & simulator release (22.11.06 from Nov. 6th, 2022)

necessary adjustments in gym-donkeycar: use gym==0.22 instead of version 0.21 in setup.py

DocGarbanzo, 2023-08-29:
> I can confirm all tests passed for the _simulator_ using DC 5.0dev3 running on Intel Linux (Ubuntu 22.04) using the latest gym-donkeycar & simulator release (22.11.06 from Nov. 6th, 2022)
> 
> necessary adjustments in gym-donkeycar: use gym==0.22 instead of version 0.21 in setup.py

@Heavy02011 - can you submit this as a PR in gym-donkeycar? I'm sure @Maximellerbach can test and merge this.

Heavy02011, 2023-09-02:
done

Heavy02011, 2023-09-04:
20230903: setup Jetson Nano Orin 8GB successfully with JetPack 5.1.2: BUT DC installation fails with CV2. further testing…

Heavy02011, 2023-09-09:
setup Jetson Orin 8GB successfully with JetPack 5.1.2 using *VENV*:
- working: install, deep learning
- not tested: calibrate, path follow, computer vision
- problems detected: 
   * installing qudida installs opencv-python-headless>=4.0.1 which breaks GSTREAMER support in OpenCV. fix by ```pip uninstall opencv-python-headless```
   * donkey ui crashes
logs under https://github.com/Heavy02011/donkeycar-test-DC50dev3/blob/main/logs/Try03_test-result.md

Ezward, 2023-09-09:
Failed to run donkey ui on Windows WSL Ubuntu 22.04.  

```
(donkey5) ed@MSI:~$ donkey ui
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v5.1.dev13 ...
[WARNING] [Config      ] Older configuration version detected (0 instead of 24)
WARNING:kivy:Config: Older configuration version detected (0 instead of 24)
[WARNING] [Config      ] Upgrading configuration in progress.
WARNING:kivy:Config: Upgrading configuration in progress.
[INFO   ] [Logger      ] Record log in /home/ed/.kivy/logs/kivy_23-09-09_0.txt
INFO:kivy:Logger: Record log in /home/ed/.kivy/logs/kivy_23-09-09_0.txt
[INFO   ] [Kivy        ] v2.1.0
INFO:kivy:Kivy: v2.1.0
[INFO   ] [Kivy        ] Installed at "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/kivy/__init__.py"
INFO:kivy:Kivy: Installed at "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/kivy/__init__.py"
[INFO   ] [Python      ] v3.9.17 (main, Jul  5 2023, 20:41:20)
[GCC 11.2.0]
INFO:kivy:Python: v3.9.17 (main, Jul  5 2023, 20:41:20)
[GCC 11.2.0]
[INFO   ] [Python      ] Interpreter at "/home/ed/miniconda3/envs/donkey5/bin/python"
INFO:kivy:Python: Interpreter at "/home/ed/miniconda3/envs/donkey5/bin/python"
[INFO   ] [Logger      ] Purge log fired. Processing...
INFO:kivy:Logger: Purge log fired. Processing...
[INFO   ] [Logger      ] Purge finished!
INFO:kivy:Logger: Purge finished!
[INFO   ] [Factory     ] 189 symbols loaded
INFO:kivy:Factory: 189 symbols loaded
[INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
INFO:kivy:Image: Providers: img_tex, img_dds, img_sdl2, img_pil (img_ffpyplayer ignored)
[INFO   ] [Text        ] Provider: sdl2
INFO:kivy:Text: Provider: sdl2
 --- Logging error ---
WARNING:kivy:stderr: --- Logging error ---
 Traceback (most recent call last):
WARNING:kivy:stderr: Traceback (most recent call last):
   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/kivy/input/providers/__init__.py", line 41, in <module>
WARNING:kivy:stderr:   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/kivy/input/providers/__init__.py", line 41, in <module>
     import kivy.input.providers.mtdev
WARNING:kivy:stderr:     import kivy.input.providers.mtdev
   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/kivy/input/providers/mtdev.py", line 94, in <module>
WARNING:kivy:stderr:   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/kivy/input/providers/mtdev.py", line 94, in <module>
     from kivy.lib.mtdev import Device, \
WARNING:kivy:stderr:     from kivy.lib.mtdev import Device, \
   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/kivy/lib/mtdev.py", line 30, in <module>
WARNING:kivy:stderr:   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/kivy/lib/mtdev.py", line 30, in <module>
     libmtdev = cdll.LoadLibrary('libmtdev.so.1')
WARNING:kivy:stderr:     libmtdev = cdll.LoadLibrary('libmtdev.so.1')
   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/ctypes/__init__.py", line 460, in LoadLibrary
WARNING:kivy:stderr:   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/ctypes/__init__.py", line 460, in LoadLibrary
     return self._dlltype(name)
WARNING:kivy:stderr:     return self._dlltype(name)
   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/ctypes/__init__.py", line 382, in __init__
WARNING:kivy:stderr:   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/ctypes/__init__.py", line 382, in __init__
     self._handle = _dlopen(self._name, mode)
WARNING:kivy:stderr:     self._handle = _dlopen(self._name, mode)
 OSError: libmtdev.so.1: cannot open shared object file: No such file or directory
WARNING:kivy:stderr: OSError: libmtdev.so.1: cannot open shared object file: No such file or directory

WARNING:kivy:stderr:
 During handling of the above exception, another exception occurred:
WARNING:kivy:stderr: During handling of the above exception, another exception occurred:

WARNING:kivy:stderr:
[... 257 lines cut ...]
x11 - ImportError: libGL.so.1: cannot open shared object file: No such file or directory
  File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/kivy/core/__init__.py", line 59, in core_select_lib
    mod = importlib.__import__(name='{2}.{0}.{1}'.format(
  File "<frozen importlib._bootstrap>", line 1109, in __import__
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 666, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 565, in module_from_spec
  File "<frozen importlib._bootstrap_external>", line 1173, in create_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed

2023-09-09 16:07:49.179983: W tensorflow/stream_executor/platform/default/dso_loader.cc:64] Could not load dynamic library 'libcudart.so.11.0'; dlerror: libcudart.so.11.0: cannot open shared object file: No such file or directory; LD_LIBRARY_PATH: /home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/cv2/../../lib64:
[... cut ...]
   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/donkeycar/management/base.py", line 626, in execute_from_command_line
     c.run(args[2:])
   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/donkeycar/management/base.py", line 597, in run
     from donkeycar.management.kivy_ui import main
   File "/home/ed/miniconda3/envs/donkey5/lib/python3.9/site-packages/donkeycar/management/kivy_ui.py", line 48, in <module>
     Window.clearcolor = (0.2, 0.2, 0.2, 1)
 AttributeError: 'NoneType' object has no attribute 'clearcolor'
```

Heavy02011, 2023-09-10:
# Raspberry Pi 3B, 1GB 
- OS Lite Bullseye (64-bit) on 32GB sdcard flashed successfully
- install DC5.0dev3 (still running)

Heavy02011, 2023-09-10:
# Pi 3B, 1GB: OS Bullseye (64-bit)
- 32GB sdcard flashed successfully
- DC5.0dev3 successfully tested: install, deeplearning (driving)
- not tested: deep learning (autopilot), calibrate, path follow, computer vision
- problems detected: none

DocGarbanzo, 2023-09-10:
> Failed to run donkey ui on Windows WSL Ubuntu 22.04.
>
> ```
> (donkey5) ed@MSI:~$ donkey ui
> ________             ______                   _________
> ___  __ \_______________  /___________  __    __ [...]

@Ezward - can you please see if `sudo apt install libfmt-dev` fixes the Kivy issue? - we have seen that in the past.

DocGarbanzo, 2023-09-10:
> # Pi 3B, 1GB: OS Bullseye (64-bit)
> * 32GB sdcard flashed successfully
> * DC5.0dev3 successfully tested: install, deeplearning (driving&autopilot), IMU via Robohat MM1
> * not tested: calibrate, path follow, computer vision
> * problems detected:
>   
>   * cv2 missing when trying to drive with a trained model. fix: `sudo apt-get install python3-opencv`

@Heavy02011 - if the above install gets you opencv, then you are running out of a `venv` but not out of a conda env. What do you get if you do the install from the conda env?

TCIII, 2023-09-10:
@DocGarbanzo,
I have successfully installed DC5.0dev3 venv on a Xavier NX 8GB running JP5.1.2, however I am getting this error when running donkey ui:

```
`File "/home/thomas/donkeycar/donkeycar/__init__.py", line 11, in <module>
    f = Figlet(font='speed')
  File "/home/thomas/env/lib/python3.8/site-packages/pyfiglet/__init__.py", line 861, in __init__
    self.setFont()
  File "/home/thomas/env/lib/python3.8/site-packages/pyfiglet/__init__.py", line 868, in setFont
    self.Font = FigletFont(font=self.font)
  File "/home/thomas/env/lib/python3.8/site-packages/pyfiglet/__init__.py", line 128, in __init__
    self.data = self.preloadFont(font)
  File "/home/thomas/env/lib/python3.8/site-packages/pyfiglet/__init__.py", line 141, in preloadFont
    path = importlib.resources.files('pyfiglet.fonts').joinpath(fn)
AttributeError: module 'importlib.resources' has no attribute 'files'`

```
I think that we saw this error when installing DC 4.4.X on JP5.0.2 when using conda?

Heavy02011, 2023-09-10:
try

> @DocGarbanzo, I have successfully installed DC5.0dev3 venv on a Xavier NX 8GB running JP5.1.2, however I am getting this error when running donkey ui:
>
> ```
> `File [...]

try to remove kivy-jetson from install und try to use kivy==2.1.0

Heavy02011, 2023-09-10:
> > # Pi 3B, 1GB: OS Bullseye (64-bit)
> > 
> > * 32GB sdcard flashed successfully
> > * DC5.0dev3 successfully tested: install, deeplearning (driving&autopilot), IMU via Robohat MM1
> > * not tested: calibrate, path follow, computer vision
> > * problems detected:
> >   
> >   * cv2 missing when trying to drive with a trained model. fix: `sudo apt-get install python3-opencv`
> 
> @Heavy02011 - if the above install gets you opencv, then you are running out of a `venv` but not out of a conda env. What do you get if you do the install from the conda env? As described here: https://github.com/autorope/donkeydocs/blob/install_with_pip/docs/guide/robot_sbc/setup_raspberry_pi.md

I followed exactly the docs and did not get an *opencv* installed, thats why I added it and it worked

DocGarbanzo, 2023-09-21:
> I followed exactly the docs and did not get an _opencv_ installed, thats why I added it and it worked

Ok cool, thanks @Heavy02011 - I'll try to reproduce (using RPi 4, but this shouldn't make a difference).

DocGarbanzo, 2023-10-07:
@autorope/contributors - I updated the install instructions above with the recipe of going back to a `venv` install on both RPi and Jetson. This is to address the observed issues, that on Jetson, we are depending on an opencv system install for the camera and on RPi we are depending on the system install for the `picamera2` module.

TCIII, 2023-10-08:
@DocGarbanzo,
I don't recommend using the Bullseye OS on the RPi 3B+ if you plan on using the `pigpio` library for hardware PWM output.
This is based on actual testing experience where running `sudo systemctl enable pigpiod & sudo systemctl start pigpiod` causes the RPi 3B+ green LED to begin flashing rapidly at every bootup.

TCIII

Ezward, 2023-11-12:
@DocGarbanzo I attempted to install on my MacBook Pro with M2-Max, MacOS Ventura 13.4.1 (c).
- I installed Miniconda
- I create the donkey environment and activated it.
- this is what I got:
```
(base) edmurphy@Eds-MacBook-Pro temp % conda activate donkey
(donkey) edmurphy@Eds-MacBook-Pro temp % pip install --extra-index-url https://test.pypi.org/simple/ donkeycar_pypi[pc]
zsh: no matches found: donkeycar_pypi[pc]
```

DocGarbanzo, 2023-11-13:
> @DocGarbanzo I attempted to install on my MacBook Pro with M2-Max, MacOS Ventura 13.4.1 (c).
>
>     * I installed Miniconda
>
>     * I create the donkey environment and activated it.
>
>     * this is what [...]

@Ezward - for zsh you need to backslashify the `[` and `]` afaik. I'm using bash on Mac. Can you try again?

zlite, 2023-11-19:
Good news. All functions tested on WSL and working!

Ezward, 2023-11-29:
@DocGarbanzo I retried with backslashes.  This time it ran but I got a failure:
```
(base) edmurphy@Eds-MacBook-Pro temp % conda activate donkey
(donkey) edmurphy@Eds-MacBook-Pro temp % pip install --extra-index-url https://test.pypi.org/simple/ donkeycar_pypi\[pc\]
Looking in indexes: https://pypi.org/simple, https://test.pypi.org/simple/
Collecting donkeycar_pypi[pc]
  Downloading https://test-files.pythonhosted.org/packages/aa/91/28bb9dc943e977c70c0236c3b77a62bd0f83559ec6d665f401e73791aaea/donkeycar_pypi-5.1.dev15-py3-none-any.whl.metadata (5.4 kB)
Collecting numpy (from donkeycar_pypi[pc])
  Downloading numpy-1.26.2-cp39-cp39-macosx_11_0_arm64.whl.metadata (61 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 61.2/61.2 kB 1.1 MB/s eta 0:00:00
Collecting pillow (from donkeycar_pypi[pc])
  Downloading Pillow-10.1.0-cp39-cp39-macosx_11_0_arm64.whl.metadata (9.5 kB)
Collecting docopt (from donkeycar_pypi[pc])
  Downloading https://test-files.pythonhosted.org/packages/f9/d7/d8f8d3e2d51df48c181d8e3706396c6fc2759b120b6be06721f00d84fe94/docopt-0.6.2-py2.py3-none-any.whl (19 kB)
Collecting tornado (from donkeycar_pypi[pc])
  Downloading tornado-6.4-cp38-abi3-macosx_10_9_universal2.whl.metadata (2.5 kB)
Collecting requests (from donkeycar_pypi[pc])
  Downloading requests-2.31.0-py3-none-any.whl.metadata (4.6 kB)
Collecting PrettyTable (from donkeycar_pypi[pc])
  Downloading prettytable-3.9.0-py3-none-any.whl.metadata (26 kB)
Collecting paho-mqtt (from donkeycar_pypi[pc])
  Downloading paho-mqtt-1.6.1.tar.gz (99 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 99.4/99.4 kB 3.9 MB/s eta 0:00:00
  Preparing metadata (setup.py) ... done
Collecting simple-pid (from donkeycar_pypi[pc])
  Downloading simple_pid-2.0.0-py3-none-any.whl (7.2 kB)
Collecting progress (from donkeycar_pypi[pc])
  Downloading progress-1.6.tar.gz (7.8 kB)
  Preparing metadata (setup.py) ... done
Collecting typing-extensions (from donkeycar_pypi[pc])
  Downloading typing_extensions-4.8.0-py3-none-any.whl.metadata (3.0 kB)
Collecting pyfiglet (from donkeycar_pypi[pc])
  Downloading pyfiglet-1.0.2-py3-none-any.whl.metadata (7.1 kB)
Collecting psutil (from donkeycar_pypi[pc])
  Downloading psutil-5.9.6-cp38-abi3-macosx_11_0_arm64.whl.metadata (21 kB)
Collecting pynmea2 (from donkeycar_pypi[pc])
  Downloading pynmea2-1.19.0-py3-none-any.whl (30 kB)
Collecting pyserial (from donkeycar_pypi[pc])
  Downloading pyserial-3.5-py2.py3-none-any.whl (90 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 90.6/90.6 kB 7.7 MB/s eta 0:00:00
Collecting utm (from donkeycar_pypi[pc])
  Downloading utm-0.7.0.tar.gz (8.7 kB)
  Preparing metadata (setup.py) ... done
Collecting pandas (from donkeycar_pypi[pc])
  Downloading pandas-2.1.3-cp39-cp39-macosx_11_0_arm64.whl.metadata (18 kB)
Collecting pyyaml (from donkeycar_pypi[pc])
  Downloading PyYAML-6.0.1-cp39-cp39-macosx_11_0_arm64.whl.metadata (2.1 kB)
INFO: pip is looking at multiple versions of donkeycar-pypi[pc] to determine which version is compatible with other requirements. This could take a while.
Collecting donkeycar_pypi[pc]
  Downloading https://test-files.pythonhosted.org/packages/51/40/68732f45cc75b2836e2f03deaf8535ec83b26e13a48824d30c61329a1ae3/donkeycar_pypi-5.1.dev14-py3-none-any.whl.metadata (5.3 kB)
  Downloading https://test-files.pythonhosted.org/packages/b3/94/1c0007630fa270925445a7fd52b97a0501d96e42a8272469cbde12dc934c/donkeycar_pypi-5.1.dev13-py3-none-any.whl.metadata (5.6 kB)
  Downloading https://test-files.pythonhosted.org/packages/83/32/d137c80d41b38d87ea173b7ccde981b737b0b4d011a62139144d472157c4/donkeycar_pypi-5.1.dev12-py3-none-any.whl.metadata (5.6 kB)
  Downloading https://test-files.pythonhosted.org/packages/58/b7/84aca1c00d1b55cb42e1345cf7408506e0a524e0eae62b41ecc2cdfd644c/donkeycar_pypi-5.1.dev11-py3-none-any.whl.metadata (5.6 kB)
  Downloading https://test-files.pythonhosted.org/packages/6f/a7/8b32c8f80189db6e8cb3c9dfb4c0219b9d89c8ef9ab67fc36438528da9e0/donkeycar_pypi-5.1.dev10-py3-none-any.whl.metadata (5.6 kB)
  Downloading https://test-files.pythonhosted.org/packages/5e/ba/19788b76bca07e1ceaa9b1782947caaa1b5e5abd2f0b0791773769e9a23e/donkeycar_pypi-5.1.dev9-py3-none-any.whl.metadata (5.4 kB)
  Downloading https://test-files.pythonhosted.org/packages/80/db/8e2c1e6aa15cc45c94f4ecc62cf57d22b1234a51f23c82334b36947d63c6/donkeycar_pypi-5.1.dev8-py3-none-any.whl.metadata (5.4 kB)
INFO: pip is still looking at multiple versions of donkeycar-pypi[pc] to determine which version is compatible with other requirements. This could take a while.
  Downloading https://test-files.pythonhosted.org/packages/b5/c4/6f25574cb3788dd3a3f2cc31c6297abb37389251819332dc6fafab043894/donkeycar_pypi-5.1.dev7-py3-none-any.whl.metadata (5.4 kB)
  Downloading https://test-files.pythonhosted.org/packages/55/df/f7ac6d81726879007e16a0bd1bce7f8ea9723e42bfd8a8498aa10feb171a/donkeycar_pypi-5.1.dev6-py3-none-any.whl.metadata (5.3 kB)
  Downloading https://test-files.pythonhosted.org/packages/c8/c0/94b56d18a8b3a171fae5fdcb9bfe3103f4e7ddc92f38660d9ae6015f5a12/donkeycar_pypi-5.1.dev5-py3-none-any.whl.metadata (5.3 kB)
  Downloading https://test-files.pythonhosted.org/packages/fb/32/121cb475d891c7dcda850d224a59990992f142e11577ace0cf84060c22b9/donkeycar_pypi-5.1.dev4-py3-none-any.whl.metadata (5.3 kB)
Collecting h5py==3.9 (from donkeycar_pypi[pc])
  Downloading h5py-3.9.0-cp39-cp39-macosx_11_0_arm64.whl.metadata (2.5 kB)
Collecting donkeycar_pypi[pc]
[... cut ...]
    donkeycar-pypi[pc] 5.1.dev3 depends on tensorflow==2.9; extra == "pc"
    donkeycar-pypi[pc] 5.1.dev1 depends on tensorflow==2.9; extra == "pc"
    donkeycar-pypi[pc] 5.1.dev0 depends on tensorflow==2.9; extra == "pc"

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
```

DocGarbanzo, 2023-12-10:
> @DocGarbanzo I retried with backslashes. This time it ran but I got a failure:
> 
> ```
>... 
> ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts
> ```

Hi @Ezward 

Please just try again with the new target `macos` that I have created in the pypi package, see install instructions above. This uses `tensorflow-macos` instead of `tensorflow`. Hopefully this works.

Ezward, 2023-12-10:
@DocGarbanzo It failed on Mac M2 with a different message:
```
% pip install --extra-index-url https://test.pypi.org/simple/ donkeycar_pypi[macos]
zsh: no matches found: donkeycar_pypi[macos]
```

DocGarbanzo, 2023-12-11:
> @DocGarbanzo It failed on Mac M2 with a different message:
> 
> ```
> % pip install --extra-index-url https://test.pypi.org/simple/ donkeycar_pypi[macos]
> zsh: no matches found: donkeycar_pypi[macos]
> ```

@Ezward - if you are using `zsh` instead of `bash` you still have to backslashify square brackets, i.e.:
```bash
pip install --extra-index-url https://test.pypi.org/simple/ donkeycar_pypi\[macos\]
```