autorope/donkeycar #1150: tensorflow-2.9.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform

valueerrorx, 2023-11-20:
i try to follow the documentation from here.

i explicitly chose raspian bullseye (legacy) because it seems to be the only version that ist supported

https://docs.donkeycar.com/guide/robot_sbc/setup_raspberry_pi/

everything went fine until i tried to run the following command:

`(env) pi@piracer:~/projects/donkeycar $ pip install -e .[pi]`

it fails with:

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Obtaining file:///home/pi/projects/donkeycar
  Preparing metadata (setup.py) ... done
ERROR: tensorflow-2.9.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.
```

it seems i can't find a way to install tensorflow inside of my virtual env

any ideas ?
thx in advance

TCIII, 2023-11-20:
@valueerrorx,
Are you using a RPi 3B+ or a RPi 4B?
I assume that you are using  these DC installation instructions for Bullseye on your RPi: `https://docs.donkeycar.com/guide/robot_sbc/setup_raspberry_pi/`
TCIII

valueerrorx, 2023-11-20:
RPi 4B .. and yes.. these are the instructions i follow..

TCIII, 2023-11-20:
@valueerrorx,
I believe that you may have installed the 32bit version of Bullseye instead of the 64bit version?
TCIII

valueerrorx, 2023-11-20:
it is aarch64

this is what i've tried:

`wget https://github.com/PINTO0309/Tensorflow-bin/releases/download/v2.9.0/tensorflow-2.9.0-cp39-none-linux_aarch64.whl

pip install tensorflow-2.9.0-cp39-none-linux_aarch64.whl 
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: tensorflow-2.9.0-cp39-none-linux_aarch64.whl is not a supported wheel on this platform.

`

```
python --version
Python 3.9.2

uname -m
aarch64

arch
aarch64

pip install --upgrade pip
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: pip in /home/pi/env/lib/python3.9/site-packages (23.3.1)

```

okay.. this package is for 2.9.0 and the raspi runs 2.9.2.. but it's a .dot release.. shouldn break right?

DocGarbanzo, 2023-11-20:
I've just tried this on my RPi with bullseye and it works without problems. What does `uname -a` give you? I see: 
```bash
(env5) pi@donkeypi:~/projects/donkeycar $ uname -a
Linux donkeypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

valueerrorx, 2023-11-20:
`Linux piracer 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux`

exactly the same... hmmm... i don't get it .. how can this be. wth did i miss

DocGarbanzo, 2023-11-22:
Did you run the usual `sudo apt-get update/upgrade`? Maybe if you comment out tensorflow from setup.py and try installing everything else to see if you get some funny versions. Here is the list of packages that gets installed on my RPi when I run the same command:

```bash
(env5) pi@donkeypi:~ $ pip list
Package                          Version      Editable project location
-------------------------------- ------------ ---------------------------
absl-py                          2.0.0
Adafruit-Blinka                  8.25.0
adafruit-circuitpython-busdevice 5.2.6
adafruit-circuitpython-framebuf  1.6.4
adafruit-circuitpython-register  1.9.17
adafruit-circuitpython-requests  2.0.2
adafruit-circuitpython-rplidar   1.2.13
adafruit-circuitpython-ssd1306   2.12.15
adafruit-circuitpython-typing    1.9.5
Adafruit-GPIO                    1.0.3
Adafruit-PCA9685                 1.0.1
Adafruit-PlatformDetect          3.54.0
Adafruit-PureIO                  1.1.11
arandr                           0.1.10
astroid                          2.5.1
asttokens                        2.0.4
astunparse                       1.6.3
automationhat                    0.2.0
beautifulsoup4                   4.9.3
blinker                          1.4
blinkt                           0.1.2
buttonshim                       0.0.2
cachetools                       5.3.2
Cap1xxx                          0.1.3
certifi                          2020.6.20
chardet                          4.0.0
click                            7.1.2
colorama                         0.4.4
colorzero                        1.1
cryptography                     3.3.2
cupshelpers                      1.0
dbus-python                      1.2.16
distlib                          0.3.6
distro                           1.5.0
docopt                           0.6.2
docutils                         0.16
donkeycar                        5.0.dev3     /home/pi/projects/donkeycar
drumhat                          0.1.0
envirophat                       1.0.0
ExplorerHAT                      0.4.2
filelock                         3.8.0
Flask                            1.1.2
flatbuffers                      1.12
fourletterphat                   0.1.0
gast                             0.4.0
google-auth                      2.23.4
google-auth-oauthlib             0.4.6
google-pasta                     0.2.0
gpiozero                         1.6.2
grpcio                           1.59.3
h5py                             3.10.0
html5lib                         1.1
idna                             2.10
importlib-metadata               6.8.0
isort                            5.6.4
itsdangerous                     1.1.0
jedi                             0.18.0
Jinja2                           2.11.3
keras                            2.9.0
Keras-Preprocessing              1.1.2
lazy-object-proxy                0.0.0
libclang                         16.0.6
logilab-common                   1.8.1
lxml                             4.6.3
[... 82 lines cut ...]
tensorboard-plugin-wit           1.8.1
tensorflow                       2.9.0
tensorflow-estimator             2.9.0
tensorflow-io-gcs-filesystem     0.34.0
termcolor                        2.3.0
thonny                           4.0.1
toml                             0.10.1
tornado                          6.3.3
touchphat                        0.0.1
twython                          3.8.2
typed-ast                        1.4.2
typing_extensions                4.8.0
tzdata                           2023.3
unicornhathd                     0.0.4
urllib3                          1.26.5
utm                              0.7.0
v4l2-python3                     0.3.2
virtualenv                       20.17.0
wcwidth                          0.2.10
webencodings                     0.5.1
Werkzeug                         1.0.1
wheel                            0.41.2
wrapt                            1.12.1
zipp                             3.17.0
```

valueerrorx, 2023-11-23:
yes i did upgrade everything. i did it three times already from scratch and strictly followed the documentation. it makes no sense to me.. that's why i gave up and tried the robocar image.

 everything works on this image.

(except that i have this "waveshare piracer" and they provide their own completely outdated repository for donkey car and with the new version the throttle is not applied to the back motors of the car. 

 but i think i need to open a new issue here for that problem

Ry7en, 2023-11-23:
I've just tried installing donkeycar for the first time, and have run into this issue as well.

kevin23916, 2023-12-07:
I encountered the same problem, and I found a solution. I checked the information and found that RPI 4B+ is the version of Raspberry Pi Bookworm. It only supports versions of TensorFlow 2.14, 2.13, and 2.12. Later, after changing the version, it can be installed and used correctly. Link[:](https://qengineering.eu/install-tensorflow-on-raspberry-64-os.html)

sctse999, 2023-12-07:
> https://github.com/robocarstore/donkeycar-images
> 
> 
> 
> my current workaround is flashing this image.. everything works out of the box. no need to install anything.

I will keep on maintaining this image so pls feel free to report any issue using this image