autorope/donkeycar #1151: JETSON ORIN NANO JETPACK5.1.2 When I use python3 manage.py drive is error OSError: [Errno 16] Device or resource busy

seed0097, 2023-12-07:
I have installed the DONKEYCAR project on the ORIN NANO JETPACK 5.1.2 system using the following URL, but I encountered the following error when running the program. The camera in use is C270.

https://github.com/Heavy02011/donkeycar-test-DC50dev3/blob/main/logs/Try03_test-result.md

It's error
(env) jetson@orin-nano:~/mycar$ python3 manage.py drive
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v5.0.dev3 ...
INFO:donkeycar.config:loading config file: /home/jetson/mycar/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
INFO:__main__:PID: 4018
WARNING:donkeycar.parts.pins:pigpio was not imported.
cfg.CAMERA_TYPE WEBCAM
INFO:__main__:cfg.CAMERA_TYPE WEBCAM
pygame 2.5.2 (SDL 2.28.3, Python 3.8.10)
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO:donkeycar.parts.camera:Opening Webcam...
Xlib:  extension "RANDR" missing on display "localhost:10.0".
INFO:donkeycar.parts.camera:Available cameras ['/dev/video0', '/dev/video1']
INFO:donkeycar.parts.camera:Webcam opened at /dev/video0 ...
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:Webcam ready.
INFO:donkeycar.vehicle:Adding part Webcam.
INFO:donkeycar.parts.web_controller.web:Starting Donkey Server...
INFO:donkeycar.parts.web_controller.web:You can now go to orin-nano.local:8887 to drive your car.
INFO:donkeycar.vehicle:Adding part LocalWebController.
INFO:donkeycar.vehicle:Adding part Pipe.
INFO:donkeycar.vehicle:Adding part ExplodeDict.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part Lambda.
INFO:donkeycar.vehicle:Adding part ThrottleFilter.
INFO:donkeycar.vehicle:Adding part UserPilotCondition.
INFO:donkeycar.vehicle:Adding part RecordTracker.
INFO:donkeycar.vehicle:Adding part AiLaunch.
INFO:donkeycar.vehicle:Adding part DriveMode.
INFO:donkeycar.vehicle:Adding part ToggleRecording.
Traceback (most recent call last):
  File "manage.py", line 1142, in <module>
    drive(cfg, model_path=args['--model'], use_joystick=args['--js'],
  File "manage.py", line 468, in drive
    add_drivetrain(V, cfg)
  File "manage.py", line 953, in add_drivetrain
    pwm_pin=pins.pwm_pin_by_id(dt["PWM_STEERING_PIN"]),
  File "/home/jetson/donkeycar/donkeycar/parts/pins.py", line 266, in pwm_pin_by_id
    return pwm_pin(pin_provider, pin_number, i2c_bus=i2c_bus, i2c_address=i2c_address, frequency_hz=frequency_hz)
  File "/home/jetson/donkeycar/donkeycar/parts/pins.py", line 385, in pwm_pin
    return PwmPinPCA9685(pin_number, pca9685(i2c_bus, i2c_address, frequency_hz))
  File "/home/jetson/donkeycar/donkeycar/parts/pins.py", line 635, in pca9685
    pca = PCA9685(busnum, address, frequency)
  File "/home/jetson/donkeycar/donkeycar/parts/pins.py", line 581, in __init__
    self.pwm = Adafruit_PCA9685.PCA9685(address=address)
  File "/usr/local/lib/python3.8/dist-packages/Adafruit_PCA9685/PCA9685.py", line 75, in __init__
    self.set_all_pwm(0, 0)
  File "/usr/local/lib/python3.8/dist-packages/Adafruit_PCA9685/PCA9685.py", line 111, in set_all_pwm
    self._device.write8(ALL_LED_ON_L, on & 0xFF)
  File "/usr/local/lib/python3.8/dist-packages/Adafruit_GPIO/I2C.py", line 114, in write8
    self._bus.write_byte_data(self._address, register, value)
  File "/usr/local/lib/python3.8/dist-packages/Adafruit_PureIO/smbus.py", line 315, in write_byte_data
    self._select_device(addr)
  File "/usr/local/lib/python3.8/dist-packages/Adafruit_PureIO/smbus.py", line 154, in _select_device
    ioctl(self._device.fileno(), I2C_SLAVE, addr & 0x7F)
OSError: [Errno 16] Device or resource busy

Heavy02011, 2023-12-07:
There are two cameras present on your orin. typically /dev/video0 is the CSI and /dev/video1 should be USB. You seem to try using /dev/video0 as WEBCAM which results in the error, as its a CSI type. 

Please adjust the camera settings in myconfig.py into /dev/video1. Then it should work.

seed0097, 2023-12-08:
Thank you for your reply.
I have changed the video in config.py to 1, but another error occurs.
When using video 0, the LED on the camera is on until an error occurs and then goes out. This indicates that video 0 should be correct.

Is there still a possibility of I2C being occupied? Currently, I am using BUS 7 with the address 0x40 according to the following website. BUS 1 seems to be using 0x40.
https://jetsonhacks.com/nvidia-jetson-orin-nano-gpio-header-pinout/

#9865, over rides only if needed, ie. TX2..
PCA9685_I2C_ADDR = 0x40     #I2C address, use i2cdetect to validate this number
PCA9685_I2C_BUSNUM = 7   #None will auto detect, which is fine on the pi. But other platforms should specify the bus num.

nano ~/projects/donkeycar/donkeycar/parts/camera.py
class Webcam(BaseCamera):
    def __init__(self, image_w=160, image_h=120, image_d=3,
                 framerate=20, camera_index=1):

It's error in video1
(env) jetson@orin-nano:~/mycar$ python3 manage.py drive
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v5.0.dev3 ...
INFO:donkeycar.config:loading config file: /home/jetson/mycar/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
INFO:__main__:PID: 3107
WARNING:donkeycar.parts.pins:pigpio was not imported.
cfg.CAMERA_TYPE WEBCAM
INFO:__main__:cfg.CAMERA_TYPE WEBCAM
pygame 2.5.2 (SDL 2.28.3, Python 3.8.10)
Hello from the pygame community. https://www.pygame.org/contribute.html
INFO:donkeycar.parts.camera:Opening Webcam...
Xlib:  extension "RANDR" missing on display "localhost:10.0".
INFO:donkeycar.parts.camera:Available cameras ['/dev/video0', '/dev/video1']
Traceback (most recent call last):
  File "/home/jetson/donkeycar/donkeycar/parts/camera.py", line 136, in init_camera
    self.cam.start()
SystemError: ioctl(VIDIOC_S_FMT) failure: no supported formats

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "manage.py", line 1142, in <module>
    drive(cfg, model_path=args['--model'], use_joystick=args['--js'],
  File "manage.py", line 104, in drive
    add_camera(V, cfg, camera_type)
  File "manage.py", line 880, in add_camera
    cam = get_camera(cfg)
  File "manage.py", line 802, in get_camera
    cam = Webcam(image_w=cfg.IMAGE_W, image_h=cfg.IMAGE_H, image_d=cfg.IMAGE_DEPTH)
  File "/home/jetson/donkeycar/donkeycar/parts/camera.py", line 106, in __init__
    self.init_camera(image_w, image_h, image_d, camera_index)
  File "/home/jetson/donkeycar/donkeycar/parts/camera.py", line 153, in init_camera
    raise CameraError("Unable to open Webcam.\n"
donkeycar.parts.camera.CameraError: Unable to open Webcam.
If more than one camera is available then make sure your 'CAMERA_INDEX' is correct in myconfig.py

Heavy02011, 2023-12-08:
can you please paste a link to your myconfig.py

seed0097, 2023-12-08:
There is myconfig.py and config.py
https://drive.google.com/drive/folders/1kf6A8L85rkjjGP6ORhDVD0NM491Lr8SE?usp=sharing

Using video 0 open the camera normally.
[image]

use video1
(env) jetson@orin-nano:~/mycar$ python3 test_camera-usb.py
[ WARN:0] global /tmp/pip-req-build-frgo6bcn/opencv/modules/videoio/src/cap_v4l.cpp (890) open VIDEOIO(V4L2:/dev/video1): can't open camera by index
Traceback (most recent call last):
  File "test_camera-usb.py", line 10, in <module>
    cv2.imshow('frame',frame)
cv2.error: OpenCV(4.5.4) /tmp/pip-req-build-frgo6bcn/opencv/modules/highgui/src/window.cpp:1006: error: (-215:Assertion failed) size.width>0 && size.height>0 in function 'imshow'

seed0097, 2023-12-11:
Thank you for your reply.

When I use 
donkey calibrate --channel 0 --bus=7
The wheels can rotate normally, but when executing python3 manage.py drive, those errors occur. Changing it to 1 also displays the same errors.

Because BUS 1 seems to be using 0x40???
https://jetsonhacks.com/nvidia-jetson-orin-nano-gpio-header-pinout/