autorope/donkeycar #1154: Accelerate tensorflow on Mac

cfox570, 2023-12-25:
The Mac install should include tensorflow-metal to leverage the builtin GPU capability and accelerate training.

Ezward, 2024-01-13:
* [Tensorflow Metal plugin](https://developer.apple.com/metal/tensorflow-plugin/)
* [Installing Tensorflow on Mac Silicon Medium Article](https://medium.com/geekculture/installing-tensorflow-on-apple-silicon-84a28050d784)
* [Tensorflow-metal pypi](https://pypi.org/project/tensorflow-metal/)

Ezward, 2024-01-13:
As of DonkeyCar 5.0, we are on Python 3.9 and Tensorflow 2.9.  Tensorflow 2.9 dictates that we install `tensorflow-macos        2.9.0`.  Once we move the Tensorflow >= 2.13 then we would just install vanilla tensorflow, rather than tensorflow-macos.  

In both cases we should install the corresponding tensorflow-metal version to enable the use of Apple Silicon.

DocGarbanzo, 2024-01-13:
> As of DonkeyCar 5.0, we are on Python 3.9 and Tensorflow 2.9. Tensorflow 2.9 dictates that we install `tensorflow-macos 2.9.0`. Once we move the Tensorflow >= 2.13 then we would just install vanilla [...]

I believe we only need to change the following line in `setup.cfg`: 
`tensorflow-macos==2.9` 
to
`tensorflow-metal==0.5`
which should then pull `tensorflow-macos` through pip dependencies automatically. I've dry-run that on my MacBook Intel which of course does not work but at least I'm getting the `pip` error messages & I can see what it tries to do.

cfox570, 2024-01-15:
We should install tensorflow-macos and tensorflow-metal on both Intel and Apple Silicon.  I submitted a pull request with the change to setup.py.  I understand if you don't need to specify tensorflow-macos but make the change to the docs for both Mac architectures.

cfox570, 2024-05-09:
The versions of tensorflow are compiled for all platforms including macOS ARM and macOS Intel.  So there is no need for the unique specifier tensorflow-macos  though that continues to reference the ARM versions of tensorflow.      I have tested both the ARM and INTEL versions of Mac platforms and neither one works properly with tensorflow-metal.   If you look at the pypi.org page, you will notice that version tensorflow-metal 1.1.0 is specified for tensorflow 2.14. When I run a training scripts the errors values explodes. I did find a comment on the internet that there is a bug in tensorflow-metal .   I am watching tensorflow-metal to see a new version is released that will correct the issue. Note that tensorflow 2.16 installs Keras 3.0 which may requires some changes to Keras training.  I welcome your comments.  Maybe I have missed something.