autorope/donkeycar #1158: error: (-215) scn == 3 || scn == 4 in function cvtColor   (JP=4.5.1)(DC=4.2.1)

0jeetparmar0, 2023-12-30:
using donkey v4.2.1 ...
loading config file: /home/jetson/mycar/config.py
loading personal config over-rides from myconfig.py
INFO:root:PID: 8536
INFO:root:cfg.CAMERA_TYPE CSIC
cfg.CAMERA_TYPE CSIC
INFO:donkeycar.vehicle:Adding part CSICamera.
Adding part CSICamera.
Starting Donkey Server...... you can now go to jetson.local:8887 to drive your car.
INFO:donkeycar.vehicle:Adding part LocalWebController.
Adding part LocalWebController.
INFO:donkeycar.vehicle:Adding part ThrottleFilter.
Adding part ThrottleFilter.
INFO:donkeycar.vehicle:Adding part PilotCondition.
Adding part PilotCondition.
INFO:donkeycar.vehicle:Adding part RecordTracker.
Adding part RecordTracker.
INFO:donkeycar.vehicle:Adding part DriveMode.
Adding part DriveMode.
INFO:donkeycar.vehicle:Adding part AiLaunch.
Adding part AiLaunch.
INFO:donkeycar.vehicle:Adding part AiRunCondition.
Adding part AiRunCondition.
PWM Steering created
Init ESC
PWM Throttle created
INFO:donkeycar.vehicle:Adding part PWMSteering.
Adding part PWMSteering.
INFO:donkeycar.vehicle:Adding part PWMThrottle.
Adding part PWMThrottle.
Using catalog /home/jetson/mycar/data/catalog_1.catalog
INFO:donkeycar.vehicle:Adding part TubWriter.
Adding part TubWriter.
You can now go to <your hostname.local>:8887 to drive your car.
INFO:donkeycar.vehicle:Starting vehicle at 20 Hz
Starting vehicle at 20 Hz
OpenCV Error: Assertion failed (scn == 3 || scn == 4) in cvtColor, file /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp, line 9716
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/home/jetson/projects/donkeycar/donkeycar/parts/camera.py", line 174, in update
    self.init_camera()
  File "/home/jetson/projects/donkeycar/donkeycar/parts/camera.py", line 169, in init_camera
    self.poll_camera()
  File "/home/jetson/projects/donkeycar/donkeycar/parts/camera.py", line 181, in poll_camera
    self.frame = cv2.cvtColor(frame, cv2.COLOR_BGR2RGB)
cv2.error: /build/opencv-XDqSFW/opencv-3.2.0+dfsg/modules/imgproc/src/color.cpp:9716: error: (-215) scn == 3 || scn == 4 in function cvtColor

TCIII, 2023-12-30:
@0jeetparmar0
What single board computer and OS are you using?
TCIII

0jeetparmar0, 2023-12-30:
using jetson nano with jetpack 4.5.1 and ubuntu 18.04

0jeetparmar0, 2023-12-30:
> @0jeetparmar0 What single board computer and OS are you using? TCIII

the motors are spinning according  to command but the webserver video output is not working

TCIII, 2023-12-30:
@0jeetparmar0 
Are you using a Waveshare product? If so, did you follow their instructions to install Donkey Car 4.2.1?

0jeetparmar0, 2023-12-30:
> @0jeetparmar0 Are you using a Waveshare product? If so, did you follow their instructions to install Donkey Car 4.2.1?

i am using waveshare product but i installed the robocar donkey car again and  got the same error i installed everything as stated over there

0jeetparmar0, 2023-12-31:
hey can you help me to remove this error?

TCIII, 2023-12-31:
@0jeetparmar0,
I think that it would be best to post in the DC Discord Software forum as there are number of members there that can provide more assistance than I can to help resolve your issue.
TCIII

Ezward, 2024-02-04:
I am closing this as not a bug.  @TCIII thank you for redirecting the user to the discord for support.

closed 2024-02-04 by Ezward