autorope/donkeycar #1160: VESC part hangs on creation

mgagvani, 2024-02-09:
Hello, 

I am trying to use the VESC part with a VESC 6 and Traxxas brushless motor. The VESC is brand-new and has 6.02 firmware. I have tested it to work while connected to my personal computer by running the PyVESC example code: https://github.com/LiamBindle/PyVESC/blob/master/pyvesc/examples/motor_example.py

However, when connected to the Nvidia Jetson Orin Nano on my car, it gets stuck in a spinlock trying to read the firmware version:
[image]

Any ideas why this is happening? The permissions have been set, and I've followed the instructions on the F1-Tenth website to create the udev rules, so if there is anything else I need to do to get it working with DonkeyCar on top of that it would be great to include that in the DonkeyCar docs. 

Pinging @sisaha9 as the creator of this, hopefully you can provide some insight.

sisaha9, 2024-02-09:
I don't have access to a physical VESC anymore to test post graduation. I can ask some folks to test out. But regarding the error itself it looks like it's getting stuck here in the VESC library and not in any Donkeycar code itself

https://github.com/LiamBindle/PyVESC/blob/master/pyvesc/VESC/VESC.py#L89

If you are able to make the example work I would double check the parameters you are inputting in both

mgagvani, 2024-02-09:
That's the weird thing: the example works on my own laptop, but not on the Jetson. So there is some underlying difference in how serial works on x86 laptops and the Jetson which is causing this error. Thanks for your help.

sisaha9, 2024-02-10:
I guess one thing to make sure is if you are in the dialout group. Otherwise you may need to use sudo to read that port