autorope/donkeycar #1165: conda / pip issue with setup tools

fletchrichman, 2024-03-03:
just a heads up for others who may run into it. i ran into issues at the step when trying to do a user or developer install via pip... 

```
ERROR: Could not find a version that satisfies the requirement setuptools (from versions: none)
      ERROR: No matching distribution found for setuptools
```

pip had setuptools installed but conda didnt seem to understand that. i fixed it with `conda install setuptools` and then everything ran fine!

Ezward, 2024-03-03:
@fletchrichman can you give more info; which version where you using and what machine were you installing on?  It sounds like Donkeycar 5, but I want to confirm so when folks from the future read this they will have full context.  Also, it sounds like a bug, so we could use more information in order to duplicate it efficiently.  Thanks.

Ezward, 2024-03-06:
@fletchrichman Thanks for creating this issue; we really do appreciate folks letting us know when something is not working; especially when they can provide a solution.  However, we need more detail to efficiently reproduce this. If you can add more detail so we can efficiently reproduce this issue then we will leave it open an work on it.  But without more detail I will be closing this at the end of the week.  

1. The version of Donkeycar you were installing.
2. The computer and operating system you were installing on.
3. The instructions you were referencing when you encountered the error (especially for install)
4. The steps to recreate the error. 
5. The console output that shows the error and as much before the error as is reasonable.

Thanks very much.

fletchrichman, 2024-03-06:
hey! i was installing donkeycar 5 on a windows machine using windows subsystem for linux. following here https://docs.donkeycar.com/guide/host_pc/setup_ubuntu/ 

i tried `pip install donkeycar[pc]` and a vague error, tried doing the developer install commands and got the error i noted above.