autorope/donkeycar #1171: Recursion error for opening donkey ui

Kim0189826, 2024-03-31:
Why does it happened after i follow all the steps in donkeycar website? Im using sublinux

It said that i had recursionerror ?

TCIII, 2024-03-31:
@Kim0189826,
The [IoT](https://rollbar.com/blog/python-recursionerror/) is your friend.
Please paste the error you are seeing so that we can view it and help you.

Ezward, 2024-03-31:
@Kim0189826 what is sublinux?

DocGarbanzo, 2024-03-31:
I have seen that error too recently. Did you perform the user install (via pip, which would give you DC 5.0.0) or the developer install (using git clone and editable install, which would give you DC 5.1.dev1)? If you did the latter, try to delete all pre-compiled python files (.pyc) and see if that helps. Otherwise, wait a couple of days when we have pushed a new 5.1 release to pypi and then do the (recommended) user install.

Kim0189826, 2024-04-01:
@DocGarbanzo i perform the developer install, so what should i do to address this recursionerror issue for my donkey ui? Additional question, when i train my tubdata for autopilot model, does it still provide h5file ?

DocGarbanzo, 2024-04-03:
> @DocGarbanzo i perform the developer install, so what should i do to address this recursionerror issue for my donkey ui? Additional question, when i train my tubdata for autopilot model, does it still provide h5file ?

@Kim0189826 - I see the same recursion error when running the UI on the RPi. It should not show up when you run with your PC installation (no matter if user or developer install). Models are not saved in the legacy keras h5 format any longer, but rather in tensorflow savedmodel format. However, for all purposes using the tflite format is better. 

I see that the recursion error is gone when using my new version of the refactored UI, so I will create a PR soon which should resolve the issue on RPi.

Kim0189826, 2024-04-16:
> @Kim0189826 - I see the same recursion error when running the UI on the RPi. It should not show up when you run with your PC installation (no matter if user or developer install). Models are not [...]

What’s the coding for using tflite in our pi for autopilot after we trained the tub data? Cause as i know before im using h5 such as command, python manage.py drive - -model ~/mycar/models/pilot 23-05-11 22.h5, and itll start the donkey monitor for running the autopilot.
[image]

Kim0189826, 2024-04-16:
Like this for an example.
[image]

Kim0189826, 2024-04-17:
@DocGarbanzo hopefully you can help to address my issue 🥲

half-freez, 2024-04-17:
> Why does it happened after i follow all the steps in donkeycar website? Im using sublinux
> 
> It said that i had recursionerror ?

Yes, I have the same error, after install using 3 different pc. Using Ubuntu, and also in windows, and also in wsl

half-freez, 2024-04-17:
[image]

Kim0189826, 2024-04-18:
> > Why does it happened after i follow all the steps in donkeycar website? Im using sublinux
> > It said that i had recursionerror ?
> 
> Yes, I have the same error, after install using 3 different pc. Using Ubuntu, and also in windows, and also in wsl 

checkout the latest ui_updated pr. It should resolve the issue.

closed 2024-04-18 by Kim0189826

half-freez, 2024-04-19:
how can i perform the ui_updated pr