autorope/donkeycar #1176: I2C messages are sended too close to each other

Prototyx, 2024-05-02:
Hello,

I made a donkey car with a RPi 4 (CM4) connected to an Arduino UNO via I2C.

[image]

Basically my Arduino UNO is replacing the PCA9685. I have my ESC and my servo attached to some PWM pins on my arduino.

It seems that there is one thread for the steering, and another one for the throttle. And each thread send their command via I2C too close to each other.

Issue : I2C messages too close to each other
[image]

Is there a way to not have both thread sending I2C commands at the same  time ? I think PCA9685 can manage 2 I2C messages very close to each other but on for my Uno it seems  very difficult. I have a lot of I/O  Errors. When I do the calibration for only the steering or throttle  it works well.

Solution :
[image]

Thanks for your help.

Ezward, 2024-05-08:
@Prototyx what type of Arduino are you using?  Is it 5v or 3.3v?  How are you handling the logic-level conversion between the RPi and the Arduino if the Arduino is 5v?

Prototyx, 2024-05-08:
Hello, The arduino is 5V (it's Atmega328p, same as uno board).  I am using a logic level converter between the Arduino and RPi  designed for I2C communications (PCA9306). 
I made some investigations today :
- I2C in Fast mode (400kbit/s) --> A lot of I/O errors
- I2C in Standard mode (100kbit/s)  -->  Something like 1 error per minute

When I put the scope on I2C lines, in fast mode the frequency is not stable at all, I think the RPi has to deal with others tasks with higher priority and decreases the frequency, maybe it is for this reason that it causes a lot of errors, because the arduino is not able to manage this frequency variation.

Anyway, after decreasing the I2C speed, I was able to drive the car without problem, and I did my first autonomous session. I was very impressed !

Ezward, 2024-05-12:
@Prototyx Thank you for detailing the resolution.  I'm closing this issue.

closed 2024-05-12 by Ezward