autorope/donkeycar #1187: Enconter

1328032567, 2024-06-12:
After successfully configuring all the necessary environment dependencies for the donkeycar project, an error originating from the manage.py script. The specific error message reads:

```bash
(env) ➜  mycar python ~/mycar/manage.py drive
________             ______                   _________
___  __ \_______________  /___________  __    __  ____/_____ ________
__  / / /  __ \_  __ \_  //_/  _ \_  / / /    _  /    _  __ `/_  ___/
_  /_/ // /_/ /  / / /  ,<  /  __/  /_/ /     / /___  / /_/ /_  /
/_____/ \____//_/ /_//_/|_| \___/_\__, /      \____/  \__,_/ /_/
                                 /____/

using donkey v5.2.dev0 ...
INFO:donkeycar.config:loading config file: /home/admin/mycar/config.py
INFO:donkeycar.config:loading personal config over-rides from myconfig.py
INFO:donkeycar.templates.complete:cfg.CAMERA_TYPE PICAM
[1:07:46.745294712] [8708]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[1:07:46.771282844] [8714]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:07:46.773854867] [8714]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media1
[1:07:46.773972477] [8714]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
[1:07:46.776431630] [8708]  INFO Camera camera_manager.cpp:284 libcamera v0.2.0+120-eb00c13d
[1:07:46.801027418] [8717]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[1:07:46.803092184] [8717]  INFO RPI vc4.cpp:446 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media1
[1:07:46.803200128] [8717]  INFO RPI pipeline_base.cpp:1102 Using configuration file '/usr/share/libcamera/pipeline/rpi/vc4/rpi_apps.yaml'
INFO:picamera2.picamera2:Initialization successful.
INFO:picamera2.picamera2:Camera now open.
INFO:picamera2.picamera2:Camera configuration has been adjusted!
[1:07:46.810740716] [8708]  INFO Camera camera.cpp:1183 configuring streams: (0) 320x240-BGR888 (1) 640x480-SGBRG10_CSI2P
[1:07:46.811147639] [8717]  INFO RPI vc4.cpp:621 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
INFO:picamera2.picamera2:Configuration successful!
INFO:picamera2.picamera2:Camera started
INFO:donkeycar.parts.camera:PiCamera opened...
INFO:donkeycar.parts.camera:...warming camera
INFO:donkeycar.parts.camera:PiCamera ready.
INFO:donkeycar.vehicle:Adding part PiCamera.
INFO:donkeycar.parts.web_controller.web:Starting Donkey Server...
INFO:donkeycar.parts.web_controller.web:You can now go to donkey.local:8887 to drive your car.
INFO:donkeycar.vehicle:Adding part LocalWebController.
INFO:donkeycar.vehicle:Adding part ExplodeDict.
INFO:donkeycar.vehicle:Adding part UserPilotCondition.
INFO:donkeycar.vehicle:Adding part LineFollower.
INFO:donkeycar.vehicle:Adding part ToggleRecording.
Toggle recording button is option
Decrement PID P button is L2
Increment PID P button is R2
INFO:donkeycar.vehicle:Adding part DriveMode.
INFO:donkeycar.parts.actuator:PWM Steering created
INFO:donkeycar.parts.actuator:Init ESC
INFO:donkeycar.parts.actuator:PWM Throttle created
INFO:donkeycar.vehicle:Adding part PWMSteering.
INFO:donkeycar.vehicle:Adding part PWMThrottle.
Traceback (most recent call last):
  File "/home/admin/mycar/manage.py", line 260, in <module>
    drive(cfg, use_joystick=args['--js'], camera_type=args['--camera'])
  File "/home/admin/mycar/manage.py", line 202, in drive
    tub_writer = TubWriter(tub_path, inputs=inputs, types=types, metadata=meta)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/projects/donkeycar/parts/tub_v2.py", line 125, in __init__
    self.tub = Tub(base_path, inputs, types, metadata, max_catalog_len)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/projects/donkeycar/parts/tub_v2.py", line 30, in __init__
    self.manifest = Manifest(base_path, inputs=inputs, types=types,
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/admin/projects/donkeycar/parts/datastore_v2.py", line 254, in __init__
    self._read_contents()
  File "/home/admin/projects/donkeycar/parts/datastore_v2.py", line 352, in _read_contents
    manifest_inputs = json.loads(self.seekeable.readline())
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/__init__.py", line 346, in loads
    return _default_decoder.decode(s)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 337, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/decoder.py", line 355, in raw_decode
    raise JSONDecodeError("Expecting value", s, err.value) from None
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
INFO:picamera2.picamera2:Camera stopped
INFO:picamera2.picamera2:Camera closed successfully.
```

## OS
```bash
admin@donkey
OS: Debian GNU/Linux 12 (bookworm) aarch64
Host: Raspberry Pi 4 Model B Rev 1.5
Kernel: 6.6.20+rpt-rpi-v8
```

Ezward, 2024-06-12:
I looks like it failed trying to load the manifest file from the /data folder.  Has the /data folder been modified?

1328032567, 2024-06-13:
> @1328032567 It looks like it failed trying to load the manifest file from the mycar/data folder. Has the /data folder been modified? Generally you should not modify anything in the mycar/data folder. So you should either restore it to the previous state or delete it and recreate a new empty mycar/data folder. Please let us know.

thank you!!!!!guys, it does work!!!Our group regenerate the directory "mycar"by the command "donkey createcar --template=cv_control --path ~/mycar" and rerun this command, and successfully solved this problem! Then our group modify the file "myconfig.py" again, and it still works. We think that there is some operates we may ignore.So next time our group would like to try Git to manage this project and track the modifies.
Finally, ty again for you in time replying!!!