autorope/donkeycar #1191: Driving via Gamer Racing Wheel (and pedals)?

generic-beat-detector, 2024-07-06:
Hi!

I'm currently using a Bluetooth PS4 (ripoff) gamepad with the DonkeyCar web-controller interface. It's nice. However, for certain demographics, driving via analog sticks remains quite a challenge (I'm not complaining though!). 

I was wondering whether someone has ever successfully used a gamer racing wheel (e.g. Logitech G920/G20) instead of a gamepad. Is this even possible?

Thanks.

TCIII, 2024-07-06:
@generic-beat-detector,

You would be better off closing this issue and posting your request in the Donkey Car Hardware Forum.

Regards,
TCIII Donkey Car Maintainer.

generic-beat-detector, 2024-07-06:
> @generic-beat-detector,
> 
> You will have better success at determining a solution by posting your request in the Donkey Car Hardware Forum. Once you have a solution, then post the solution on this issue so it can be implemented.
> 
> Regards, TCIII Donkey Car Maintainer.

Yes sir. Thanks for the prompt reply!

generic-beat-detector, 2024-07-06:
@TCIII How do I access this Hardware Forum?

TCIII, 2024-07-06:
@generic-beat-detector,
I assume that you are not a member of the Donkey Car Discord Forum?
Go [here](https://www.donkeycar.com/community.html) and click on the "JOIN OUR DISCORD SERVER" tab.

Regards,
TCIII

generic-beat-detector, 2024-07-06:
> @generic-beat-detector, I assume that you are not a member of the Donkey Car Discord Forum? Go [here](https://www.donkeycar.com/community.html) and click on the "JOIN OUR DISCORD SERVER" tab.
> 
> Regards, TCIII

Thanks! I just joined the discord forum -- well at least I've managed to post.