autorope/donkeycar #1195: donkey ui : RecursionError('maximum recursion depth exceeded') in donkeycar version 5.1

Reply-WRE, 2024-08-30:
Hi guys.

I have the same problem discussed here:

https://github.com/autorope/donkeycar/issues/1171

This problem seems to be fixed in release 5.1, but seems persist.

I have this configuration:

donkeycar==5.1.0
tensorflow==2.15.0
tensorflow-estimator==2.15.0
tensorflow-io-gcs-filesystem==0.37.1

But the problem persist.

`Level 1:tensorflow:Enabling v2 tensorshape
[Level 1           ] Enabling v2 tensorshape
Level 1:tensorflow:Enabling resource variables
[Level 1           ] Enabling resource variables
Level 1:tensorflow:Enabling tensor equality
[Level 1           ] Enabling tensor equality
Level 1:tensorflow:Enabling control flow v2
[Level 1           ] Enabling control flow v2
object address  : 0x7d06258f35e0
object refcount : 2
object type     : 0x874120
object type name: RecursionError
object repr     : RecursionError('maximum recursion depth exceeded')
lost sys.stderr
`

These are the last lines of the error.

DocGarbanzo, 2024-08-30:
We haven't seen this issue since upgrading to 5.1 any longer. Can you please:

1.  Post the full command and shell output when you are running the UI
2. Attach a `pip list` output
3. Tell what OS you are using

Reply-WRE, 2024-09-02:
Ok guys some updates:
The problem is with the user installation, in fact I also tried the dev installation in another conda environment and it works fine.

This morning I tried again with the user installation and something very strange happened: The process seemed to want to start, but then the pc froze and the process was killed. 

I suppose with the dev installation some package was updated and now I don't have the same problem as on Friday, but with the user installation there is still some problem.

pip list command:

> donkey) wre@pop-os:~/Scrivania/donkey$ pip list
> Package                      Version
> ---------------------------- -----------
> absl-py                      2.1.0
> albucore                     0.0.14
> albumentations               1.4.14
> annotated-types              0.7.0
> astunparse                   1.6.3
> cachetools                   5.5.0
> certifi                      2024.8.30
> charset-normalizer           3.3.2
> contourpy                    1.3.0
> cycler                       0.12.1
> docopt                       0.6.2
> docutils                     0.21.2
> donkeycar                    5.1.0
> eval_type_backport           0.2.0
> flatbuffers                  24.3.25
> fonttools                    4.53.1
> gast                         0.6.0
> google-auth                  2.34.0
> google-auth-oauthlib         1.2.1
> google-pasta                 0.2.0
> grpcio                       1.66.1
> h5py                         3.11.0
> idna                         3.8
> imageio                      2.35.1
> keras                        2.15.0
> Kivy                         2.3.0
> Kivy-Garden                  0.1.5
> kiwisolver                   1.4.5
> lazy_loader                  0.4
> libclang                     18.1.1
> Markdown                     3.7
> markdown-it-py               3.0.0
> MarkupSafe                   2.1.5
> matplotlib                   3.9.2
> mdurl                        0.1.2
> ml-dtypes                    0.2.0
> namex                        0.0.8
> networkx                     3.3
> numpy                        1.26.4
> oauthlib                     3.2.2
> opencv-python-headless       4.10.0.84
> opt-einsum                   3.3.0
> optree                       0.12.1
> packaging                    24.1
> paho-mqtt                    2.1.0
> pandas                       2.2.2
> pillow                       10.4.0
> pip                          24.2
> plotly                       5.24.0
> prettytable                  3.11.0
> progress                     1.6
> protobuf                     4.25.4
> psutil                       6.0.0
> pyasn1                       0.6.0
> pyasn1_modules               0.4.0
> pydantic                     2.8.2
> pydantic_core                2.20.1
> pyfiglet                     1.0.2
[... 1427 lines cut ...]
> [Level 1           ] Registering AddN,flops (<function _add_n_flops at 0x73be14013600>) in statistical functions.
> Level 1:tensorflow:Registering MatMul,flops (<function _calc_mat_mul_flops at 0x73be140136a0>) in statistical functions.
> [Level 1           ] Registering MatMul,flops (<function _calc_mat_mul_flops at 0x73be140136a0>) in statistical functions.
> Level 1:tensorflow:Registering BatchMatMulV3,flops (<function _calc_batch_mat_mul_flops at 0x73be14013740>) in statistical functions.
> [Level 1           ] Registering BatchMatMulV3,flops (<function _calc_batch_mat_mul_flops at 0x73be14013740>) in statistical functions.
> Level 1:tensorflow:Registering BatchMatMulV2,flops (<function _calc_batch_mat_mul_flops at 0x73be14013740>) in statistical functions.
> [Level 1           ] Registering BatchMatMulV2,flops (<function _calc_batch_mat_mul_flops at 0x73be14013740>) in statistical functions.
> Level 1:tensorflow:Registering BatchMatMul,flops (<function _calc_batch_mat_mul_flops at 0x73be14013740>) in statistical functions.
> [Level 1           ] Registering BatchMatMul,flops (<function _calc_batch_mat_mul_flops at 0x73be14013740>) in statistical functions.
> Level 1:tensorflow:Enabling eager execution
> [Level 1           ] Enabling eager execution
> Level 1:tensorflow:Enabling v2 tensorshape
> [Level 1           ] Enabling v2 tensorshape
> Level 1:tensorflow:Enabling resource variables
> [Level 1           ] Enabling resource variables
> Level 1:tensorflow:Enabling tensor equality
> [Level 1           ] Enabling tensor equality
> Level 1:tensorflow:Enabling control flow v2
> [Level 1           ] Enabling control flow v2
> kill

My Os is Pop Os 22.04 LTS

> (donkey) wre@pop-os:~/Scrivania/donkey$ uname -a
> Linux pop-os 6.9.3-76060903-generic #202405300957~1721174657~22.04~abb7c06 SMP PREEMPT_DYNAMIC Wed J x86_64 x86_64 x86_64 GNU/Linux

harty1193, 2024-09-12:
Which dev Installaton. I am getting the exact error.

Reply-WRE, 2024-09-25:
> Which dev Installaton. I am getting the exact error.

I just followed the istruction in the documentation for the dev installation. So the code should be the one in the github repo.