autorope/donkeycar #1202: Can't find moblie

1328032567, 2024-10-24:


TCIII, 2024-10-24:
@1328032567 

What SBC are you using: RPi or Nano?

TCIII

1328032567, 2024-10-25:
> @1328032567
> 
> What SBC are you using: RPi or Nano?
> 
> TCIII

The model of my SBC is **RaspberryPi 4B 2GB**, and the os is **Raspberry Pi OS**. Looking forward to your reply.

sctse999, 2024-11-11:
The Android app on play store was suspended by Google because of those developer verification nonsense. I have updated it and the app is back online on play store. Pls check.

wickdynex, 2024-11-13:
> The Android app on play store was suspended by Google because of those developer verification nonsense. I have updated it and the app is back online on play store. Pls check.

Thanks, this [app on Google Play](https://play.google.com/store/apps/details?id=com.robocarLtd.RobocarController) has recovered. Btw, can u have a code review and merge those [PR1](https://github.com/autorope/donkeydocs/pull/59) and [PR2](https://github.com/autorope/donkeydocs/pull/61) from this related project [donkeydocs](https://github.com/autorope/donkeydocs)🥰 if you have admin rights to do so, or that project is seldom maintained?