autorope/donkeycar #1211: adding support for BeagleY-AI and aarch64 with Specifics

silver2row, 2025-07-27:
Hello,

My name is Seth. I tried to port Donkey Car to a TI based am67a chip (aarch64) without success, e.g. as I do not use libraries outside of `gpiod` and some .dts files for PWM access. 

I have PWM access on the board for pushing DC motors forward and in reverse but:

1. I will not be using a PCA9685 for PWM access
2. I will not be using the "RPI_GPIO" or "PIGPIO" libraries
3. If I push some requests with source, can I use that source in Donkey Car

I am seemingly getting wrapped up in changing many pages of source without success. `pins.py` is the main file I have been altering for use with `gpiod` and how beagleboard.org personnel seems to want to promote PWM access.

Anyway, `gpiod` in python3 for input on rising can be found here:

```
import gpiod
from datetime import timedelta
from gpiod.line import Bias, Edge

def watch_rising_edge(chip_path, line_offset):
    """
    Watches for rising edge events on a specified GPIO line.
    """
    with gpiod.request_lines(
        chip_path,
        consumer="rising-edge-detector",
        config={
            line_offset: gpiod.LineSettings(
                edge_detection=Edge.RISING,
                bias=Bias.PULL_DOWN, # Assuming a button connecting to VCC, so pull down
                debounce_period=timedelta(milliseconds=50), # Add debounce for physical buttons
            )
        },
    ) as request:
        print(f"Waiting for rising edge on {chip_path} line {line_offset}...")
        while True:
            # Blocks until at least one event is available
            for event in request.read_edge_events():
                if event.event_type is event.Type.RISING_EDGE:
                    print(f"Rising edge detected on line {event.line_offset} at {event.timestamp_ns} ns")

if __name__ == "__main__":
    # Customize these values for your specific setup
    # For Raspberry Pi 5, external GPIOs are often on gpiochip4
    # For older Raspberry Pis, they are often on gpiochip0
    gpio_chip_path = "/dev/gpiochip0" 
    gpio_line_offset = 21  # Replace with your actual GPIO line number

    try:
        watch_rising_edge(gpio_chip_path, gpio_line_offset)
    except OSError as ex:
        print(f"Error: {ex}\nEnsure the chip path and line offset are correct for your system.")
    except KeyboardInterrupt:
        print("\nExiting program.")
```

At first, I tried altering where the `RPI_GPIO` library is imported and exchanged it out for `import gpiod as GPIO`.

Then, in the source above, I changed `gpiod` for `GPIO`, e.g. `GPIO.request_lines()` instead of `gpiod.request_lines()`.

Also, the `./manage.py drive` command returns error no matter what I do. My error is as follows:

```
/Testing_Donkey/donkeycar-5.1.0/lib/python3.11/site-packages/donkeycar/parts/pins.py", line 262, in output_pin_by_id
    raise ValueError(f"Unknown pin provider {parts[0]}")
ValueError: Unknown pin provider PWM
```

I tried with associating PWM as in the `class PinProvider:` and in the `class PinScheme:` but without success. 

Do you know where I can look to alter the source in your library for handling such tasks? 

Seth

P.S. I think I can make I/O work for GPIO but not PWM so far for some reason (unknown to me thus far).

silver2row, 2025-07-27:
That `gpiod` source was accumulated from AI online via the Google browser. I can alter that source to handle GPIO I/O easily but I am afraid that I do not know where it belongs within our source nor do I understand what keeps me from utilizing it.

TCIII, 2025-07-27:
@silver2row,
Is there a specific reason that you don't want to use the PCA9685 to generate PWM signals as it sounds like its use would negate your issue?
TCIII

DocGarbanzo, 2025-07-28:
@silver2row - I don't think it makes sense for Donkey Car to support the beagle board officially. We used to support the RPi and Jetson Nano, but it has turned out to be very difficult to keep Donkey Car working on both platforms, and only the RPi works with all versions. There is simply not enough people maintaining a wider set of hardware within the Donkey Car group. If you want to add/change the code such that it works with the beagle board, then you will be better off to fork the repo and get everything working on your hardware.

closed 2025-07-28 by DocGarbanzo